probonopd/MiniDexed #10: Support the sustain button on MIDI controllers

probonopd, 2022-02-20:
Support the sustain button on MIDI controllers.

(Once https://github.com/probonopd/MiniDexed/issues/8 is implemented I can say which MIDI message the sustain button sends.)

dcoredump, 2022-02-20:
Will do this with adding MIDI processing support for Synth_Dexed (see also #9, #11)

rsta2, 2022-03-15:
Here is a small patch to implement MIDI CC Sustain (Damper). The behavior is different from the damper pedal of my Yamaha home keyboard. When you release the sustain button, the note does not go off. You have to press the key again, to switch it off. This is managed by Synth_Dexed. @dcoredump I don't know, if it is intended?

[sustain.zip](https://github.com/probonopd/MiniDexed/files/8256855/sustain.zip)

probonopd, 2022-03-17:
On my M-AUDIO Keystation Mini 32 Mk3 a LED goes on when I press the sustain key (sending `B0 40 7F`), and the LED goes off when I press the same key again (sending `B0 40 00`).

So I think it is correct as it is implemented: A sustain button works differently from a sustain pedal.

probonopd, 2022-03-17:
Pressing the sustain key works.

But pressing it again, the already sustained notes keep playing forever. This is not like it should be.

rsta2, 2022-03-17:
The sustain code in Dexed is [here](https://github.com/asb2m10/dexed/blob/master/Source/PluginProcessor.cpp#L386). I think line 386-393 have to be added to `Dexed::setSustain()` in Synth_Dexed. This will trigger the `keyup()` events automatically, when the sustain button/pedal is released.

rsta2, 2022-03-17:
Here is a patch for Synth_Dexed, which enables the full sustain support:

```
diff --git a/src/dexed.cpp b/src/dexed.cpp
index be43a27..a93521a 100644
--- a/src/dexed.cpp
+++ b/src/dexed.cpp
@@ -406,6 +406,15 @@ void Dexed::setSustain(bool s)
     return;
 
   sustain = s;
+
+  if (!sustain) {
+      for (int note = 0; note < max_notes; note++) {
+          if (voices[note].sustained && !voices[note].keydown) {
+              voices[note].dx7_note->keyup();
+              voices[note].sustained = false;
+          }
+      }
+  }
 }
 
 bool Dexed::getSustain(void)
```

It's tested, it works.

probonopd, 2022-03-18:
@dcoredump could you add it please? Thanks.

dcoredump, 2022-03-18:
It's fixed now in `https://codeberg.org/dcoredump/Synth_Dexed`

Thanks @rsta2!

closed 2022-03-18 by probonopd