probonopd/MiniDexed #11: Support selecting patches using MIDI Bank Change and Program Change messages

probonopd, 2022-02-20:
Many USB keyboards can send Bank LSB and Bank MSB Changes, and Program Changes. If we put several `.syx` files on the microSD card, then we might use Bank LSB and Bank MSB Changes to select those `.syx` files, and Program Changes to select patches.

(Once #8 is implemented I can say which MIDI messages we need to react to.)

probonopd, 2022-02-20:
Since our ultimate goal here is to run multiple instances of Dexed (e.g., up to 8 instances akin to the [TX816](https://www.vintagesynth.com/yamaha/tx816.php) and the [TX802](https://www.vintagesynth.com/yamaha/tx802.php), we should also think of a way to define and load "performance" patches (basically a collection of up to 8 patches).

dcoredump, 2022-02-20:
Will do this with adding MIDI processing support for Synth_Dexed (see also #9, #10, #11)

probonopd, 2022-02-25:
@dcoredump in the meantime I've been trying to add an interim solution for loading different default voices but I can load only voice 0. I must be doing a stupid C mistake, can you spot it?

https://github.com/probonopd/MiniDexed/commit/b0038a750a29ae6bfa83815cbe0227638f6f48aa

dcoredump, 2022-02-25:
Currently not much free time for me, sorry.

I think you forgot to decode the 138 bytes of voice data:

uint8_t decoded_voice[156];
dexed.decodeVoice(progmem_bank[b][v],decoded_voice);

probonopd, 2022-02-25:
Decoded them using `sysex2c.py --decode´. Voice 0 works. Must be something else?

dcoredump, 2022-02-25:
> Decoded them using `sysex2c.py --decode´. Voice 0 works. Must be something else?

This is right, but you get 128 bytes per voice and some bytes are having more than one parameter stored. You have to decodem them before (as I wrote above).

probonopd, 2022-02-25:
Thanks. Just to be sure @dcoredump. By using `--decode´  get more than 128 bytes C code. Do I have to decode them _again_ in MiniDexed? Or should I _not_ use  `--decode´ in Python and _just_ do the decoding in MiniDexed?

probonopd, 2022-02-25:
I mean, in the end we may want to load voices from `.syx` files stored on SD anyway, most likely by porting
https://codeberg.org/dcoredump/MicroDexed/src/branch/master/dexed_sd.cpp - maybe @rsta2 could help with this?

dcoredump, 2022-02-25:
> Thanks. Just to be sure @dcoredump. By using `--decode`  get more than 128 bytes C code. Do I have to decode them _again_ in MiniDexed? Or should I _not_ use  `--decode` in Python and _just_ do the decoding in MiniDexed?

sysex2c.py decodes a bank sysex file into a C-Header file (with header and checksum checking).

For using the 128 bytes of a voice you need to decode it again... perhaps the wording is not the best for both steps...

See also https://codeberg.org/dcoredump/MicroDexed/src/branch/master/doc/sysex-format.txt

rsta2, 2022-02-25:
Please apply the following patch and it should work:

```
diff --git a/src/minidexed.cpp b/src/minidexed.cpp
index 1951781..bb9857b 100644
--- a/src/minidexed.cpp
+++ b/src/minidexed.cpp
@@ -15,7 +15,7 @@
 
 CMiniDexed *CMiniDexed::s_pThis = 0;
 
-extern uint8_t voices_bank[1][128][128];
+extern uint8_t voices_bank[1][32][156];
 
 bool CMiniDexed::Initialize (void)
 {
```

probonopd, 2022-02-25:
Works! Now that you are poining it out my mistake feels so stupid. I really wonder why the compiler doesn't catch such stupid mistakes. Thanks for your help.

rsta2, 2022-02-25:
No problem, that happens!

probonopd, 2022-02-25:
Added 

https://github.com/probonopd/MiniDexed/blob/29acc02891badc4d7094c453f5f81ec2b16fe549/src/minidexed.cpp#L141-L144

to print out the patch name (in preparation for my HD44780 LCD that should come in soon) but apparently it is not correct either...

rsta2, 2022-02-25:
Can you try `setName()` instead of `getName()`. I think it's the "reverse" naming logic after looking into the Synth_Dexed code. ;)

Also in such case for a local buffer you should use `char buf_name[11];` instead of `new`. If you use `new`, there must be a `delete` somewhere, otherwise there will be a memory leak. :(

I think it's a question of learning. If you come from Python, it may be a bit hard.

probonopd, 2022-02-26:
Program Change (from the built-in ROM bank) works. Displaying the voice name works.
Next up, loading `.syx`. Let's think about how we want to organize them on the SD card. Something like this?

```
/sysex/voice/
/sysex/voice/1.syx # If there, override the built-in internal ROM voice bank
/sysex/voice/2.syx
/sysex/voice/3.syx
/sysex/performance/
...
```

rsta2, 2022-02-26:
I would prefer the first suggestion. I think it's better to have all banks in one directory. Otherwise you have a lot to do to move banks from one to another directory all the time maybe. I can implement it, if you agree. If we need the other variant some day, this could be changed.

probonopd, 2022-02-26:
Agree. Whatever is simpler. We can always make it more complex if needed later on.

rsta2, 2022-02-26:
OK, I will work on this.

rsta2, 2022-02-26:
Here comes a solution for accessing voices from DX7 `.syx` files from the SD card in the */sysex/voice* directory. It's the first suggested naming for the `.syx` files, but everything as follows should work:

```
1.syx
001.syx
000001.syx
001_ROM1A.syx
000001_ROM2B.syx
```

All files will be loaded on system start into memory. This is necessary, because the MIDI packet handler is an IRQ callback and file access is not possible there, when a program change arrives. The number of possible voice banks is currently limited to 1..128. I hope this is enough for the moment. Perhaps this can be extended later, if required.

The new class `CSysExFileLoader` uses bank and voice numbers of 0..127 internally. Only the file names and user messages use numbers 1..128. Also I guess, that program and bank numbers, which are transferred via MIDI are 0..31 and 0..127. This is implemented differently at the moment (1..32, 1..128). Can you please check this? My USB keyboard cannot send program changes and bank selects, so I used some tricks to test it.

[sysex_file.zip](https://github.com/probonopd/MiniDexed/files/8146958/sysex_file.zip)

probonopd, 2022-02-26:
The current behavior is like this:

* Asking the keyboard to send Bank Change LSB 1 results in MIDI message `B0 20 01` and results in `Select voice bank 1`.
* Then, asking the keyboard to send Program Change 1 results in MIDI message `C0 01` and results in "Loading voice 1" "BRASS 1`. Which in my case is stored in `000001_rom1a.syx`.

Working nicely +1:

The manual of my keyboard is somewhat vague and doesn't say whether the user is expected to send Bank Change LSB 0 or 1 in order to load the first available bank. 

Thanks @rsta2

rsta2, 2022-02-26:
See [this](https://usa.yamaha.com/files/download/other_assets/9/333979/DX7E1.pdf) (page 32, "Program change & Channel number"):

```
0ppppppp    Program number (p=0: INT1 ~ p=31: INT32
            p=32: CRT1 ~ p=63: CRT32)
```

"INT" must be internal ROM, "CRT" cartridge. This DX7 version does not have bank select.

probonopd, 2022-02-26:
Maybe our MIDI message handling should be oriented at (a subset of) TX802 which is more modern?

probonopd, 2022-02-27:
Looks like MIDI messages need to go from 0-127 (0x00-0x7f) (even though this is often displayed to the user as 1-128).

Apparently it is documented misleading even in official MIDI documentation:

https://www.midi.org/forum/3810-midi-tuning-program-change-and-bank-change-messages-really-one-based,-not-zero-based#reply-3811

Someone pointed out that "the spec is not incorrect" but it is _misleading_ to say the least.

> The spec is not incorrect.
The values of "tt" can only be 0-127 because "tt" is a hexidecimal number.
But the spec is not referring to _the value_ of "tt", the spec is referring to what "tt" is _used for_, which is to select Program and Bank numbers, where the first number is "1", and the last number is "128".

I read this as:
When the user sends program change `0x00` (0) then "voice 1" should get loaded (BRASS 1). Likewise, when the user sends program change `0x0A` (10) then "voice 11" should get loaded (E.PIANO 1).

Probably the same is true for the Bank Change and other messages. So we need to change our implementation.

rsta2, 2022-02-27:
Yes, I fully agree. I had a look into the Rosegarden open source sequencer and in its MIDI settings, one can configure 0-127 separately for LSB and MSB for each used bank. For the program number there is a button, with which you can switch the numbering between 1-128 and 0-127. This is a help for the user, but in MIDI transfers the range is always 0-127. I think that's the same as you found out.

probonopd, 2022-02-27:
Does this look approximately right?
https://github.com/probonopd/MiniDexed/pull/32/commits/24cd4bd5601e07de80bf431fe5ee987f87f4f50a

rsta2, 2022-02-27:
Two things to improve: The test in *minidexed.cpp* in `MIDIPacketHandler()` must be `>= 128` and `>= 32`, not `> 128` and `> 32`, otherwise the 128 and 32 are still allowed. And in [sysexfileloader.cpp](https://github.com/probonopd/MiniDexed/blob/main/src/sysexfileloader.cpp#L76) the if-statement must be `if (nBank > MaxVoiceBankID)` now.

probonopd, 2022-02-27:
Currently a MIDI Bank Select message alone does nothing, only after a MIDI Program Change message it has any impact.

So: When receiving a MIDI Bank Change, should we load the same voice number from the new bank automatically? What are real MIDI instruments (e.g., MODX) doing in this case?

probonopd, 2022-03-13:
I didn't find this exact information in DX documentation but the "MX49/MX61 Data List" PDF says:

> Bank Select will be actually executed when the Program Change message is received.

So it seems like we should _not_ load the patch from the new bank immerdiately after a Bank Select not followed by a Program Change message.

> Bank Select and Program Change numbers that are not supported by Yamaha will be ignored.

@rsta2 you are already doing this for the rotary encoder and it seems like we should be doing the same for MIDI Bank Select and Program Change messages.

closed 2022-03-13 by probonopd