probonopd/MiniDexed #12: Run multiple Dexed instances at the same time

probonopd, 2022-02-20:
Our ultimate goal here is to run multiple instances of Dexed (e.g., up to 8 instances akin to the [TX816](https://www.vintagesynth.com/yamaha/tx816.php) and the [TX802](https://www.vintagesynth.com/yamaha/tx802.php).

A very typical use case is to create the characteristic "fat" sound by running multiple instances of the same instrument, each slightly detuned and slightly stereo-panned.

As shown here:
* [Yamaha TX816 - multiple stacked detuned DX7 voices = epic!](https://www.youtube.com/watch?v=HPmQq_2bwXs)
* [TX816 Fulltines](https://www.youtube.com/watch?v=j2emXkQ0HAA)

probonopd, 2022-02-20:
@rsta2] [says](https://github.com/probonopd/MiniDexed/issues/15#issuecomment-1046209785):

> Currently MiniDexed does use only CPU core 0. If you want to change this, you must parallelize things. For example run Dexed::getSamples() on different cores for different DX7 instances, when there are more than one of it.

dcoredump, 2022-02-20:
Just briefly my ideas on this: Processing audio in parallel would be cool, but synchronizing the individual processes for output is not that easy. I would push this far back and first try to reach the maximum on one CPU.

probonopd, 2022-02-20:
Would running sound generation on one core and the effects on a separate core be any easier?
Currently I have no idea about how CPU intensive running multiple Dexed instances with some reverb will be. All I can say is that using REAPER I can run _many_ more than 8 on the same hardware :-)

probonopd, 2022-02-20:
Here are 20 instances of Dexed playing in REAPER on the the Raspberry Pi 4, all slightly detuned, some stereo-shifted, and reverb added to them in groups of 4. When playing this, ~50% of the Raspberry Pi 4 is idle, so I am assuming this type of workload needs ~2 cores.

[20-detuned-dexed-shifted-and-reverb.zip](https://github.com/probonopd/MiniDexed/files/8104089/20-detuned-dexed-shifted-and-reverb.zip)

rsta2, 2022-02-25:
In a quick hack I tried to run four Dexed instances parallel, one on each core of a Raspberry Pi 4. This worked. I used the same method as in MiniSynth Pi to synchronize the cores. I don't know, if I should develop this further or if there are other things, which are more important now?

probonopd, 2022-02-25:
Cool @rsta2. Do you also see a way to run more than one instance on the same core? We should get a rough understanding on how much CPU % all of this is eating, but we also need to keep in mind that we'll want to do mixing and effects.

rsta2, 2022-02-25:
Yes, of course multiple instances on one core should be possible. I could add some measurement, how long the sound rendering in `GetChunk()` takes and how many percent of the available time is currently used. Is this OK?

I will have look at #11.

rsta2, 2022-02-26:
The measurement code is ready. I tried it on a RPi 4B with PWM and one DX7 instance and I got not more then 10% usage of the available time in `GetChunk()`, which is the main workhorse at the moment. So with four cores it should be no problem to provide eight instances there. You find the patch attached. The given percent value is only precise for PWM and I2S at the moment, because HDMI uses a different chunk size. I did not want to change too much here currently.

BTW. The `CHUNK_SIZE [ _HDMI ]` parameter should be tuned and decreased for smaller latency. Maybe it should be configurable for the user later.

[perftimer.zip](https://github.com/probonopd/MiniDexed/files/8146233/perftimer.zip)

probonopd, 2022-02-26:
OK, it seems like we will going to have different functionality depending on Raspberry Pi model.
Intensive playing of 1 instance can bring the CPU on a Raspberry Pi 1 to ~70%.
So maybe we need to distinguish functionality like this:
* Raspberry Pi 1 = DX7 like functionality, maybe with some "light" effects
* ...
* Raspberry Pi 4 = (at least?) TX802 like functionality, with full effects

probonopd, 2022-02-27:
Did some performance testing above. @dcoredump @rsta2 what do you think about how we should best make use of the 4 cores?

rsta2, 2022-02-28:
I would do the following for 8 Dexed instances (2 per CPU core):

* Use an array of synchronization variables (enum, one for each secondary core 1-3) which can have the values *init, idle, busy, exit*.

* A secondary core normally waits in a loop continuously reading its synchronization variable to change from idle to busy. When this happens it calls `Dexed::getSamples()` twice (for 2 Dexed instances per core). The calculated samples will be placed in another data array, which exists for each Dexed instance. When this calculation is finished, the secondary core sets it's synchronization variable back to idle.

* When `GetChunk()` gets called on the primary core 0 from the sound device IRQ handler, it sets the synchronization variables for the secondary cores 1-3 to *busy* and starts doing the same calculation of samples for 2 instances then. When it is ready doing this, it waits in a loop until the synchronization variables for the cores 1-3 have the value *idle* again. This means, the secondary cores have finished their work. After that core 0 sums up the samples from the data arrays of all cores and writes the chunk buffer, which has been handed over to `GetChunk()`.

This is the same technique, which I used in MiniSynth Pi and it works well there.

probonopd, 2022-02-28:
I think some of my earlier tests were done with `sndhdmi` by mistake. Repeated with `sndpwm`, looking better:

* Raspberry Pi 1 up to ~50% [maybe with a little overclocking we can do 2 instances like on the famous DX1... although the PWM output on this model is especially bad]
* Raspberry Pi 2 up to ~26%
* Raspberry Pi 3 up to ~23%
* Raspberry Pi 4 up to ~10%

rsta2, 2022-02-28:
Yes, but the worst case counts. If we want to support HDMI, it must also work there, or we provide a special configuration for HDMI.

I have to note, that we are currently running are normal bare metal speed (600 or 700 MHz) only. See the table [here](https://circle-rpi.readthedocs.io/en/44.3/basic-system-services/cpu-clock-rate-management.html) for the maximum clock rates of the different models. This can be enabled with a small change in MiniDexed, but on RPi 3-4 you can use this only, if you have a heat sink or fan. Otherwise the clock rate may drop occasionally, when the CPU temperature gets too high and this could MiniDexed force to crash at the moment.

probonopd, 2022-03-01:
> Yes, but the worst case counts. If we want to support HDMI, it must also work there.

Sure. I was just wondering whether the measurements for HDMI are currently accurate at all.

rsta2, 2022-03-01:
The measured times for HDMI are correct, but not the percent value. I will fix this with restructuring the code.

probonopd, 2022-03-19:
Thanks @rsta2 for having implemented multiple Dexed instances.

closed 2022-03-19 by probonopd