probonopd/MiniDexed #15: Reverb and other effects

probonopd, 2022-02-20:
Looks like https://codeberg.org/dcoredump/MicroDexed/ has effects like this. Do you think these could be put into Synth_Dexed and hence made usable for MiniDexed @dcoredump?

Let's think about how we should best make use of the 4 cores of the Raspberry Pi  4 for tasks like
* Running multiple Dexed instances (ideally at least 8)
* Running effects like Reverb
* Eventually, running some sort of user interface (in addition to MIDI sysex commands)

Is there a way to periodically output the CPU usage of each core @rsta2? We should find out how CPU intensive the various things are and how to best split them up across CPU cores.

rsta2, 2022-02-20:
You can fetch a microseconds up-counter with:

```
#include <circle/timer.h>

unsigned count = CTimer::GetClockTicks();
```

If you save the count before and after some action, you can determine, how much time it takes.

Because there aren't tasks in MiniDexed, the CPU usage is always 100%. Most time is likely spent in `GetChunk()` and in the main loop in `CKernel::Run()`.

BTW. Currently MiniDexed does use only CPU core 0. If you want to change this, you must parallelize things. For example run `Dexed::getSamples()` on different cores for different DX7 instances, when there are more than one of it.

probonopd, 2022-02-20:
> Currently MiniDexed does use only CPU core 0. If you want to change this, you must parallelize things. For example run Dexed::getSamples() on different cores for different DX7 instances, when there are more than one of it.

Exactly. This is what I mean.

probonopd, 2022-03-21:
@dcoredump now that we have multi-core and multiple Dexed instances (tone generators), how would one best go about adding reverb?

dcoredump, 2022-03-23:
Sorry for answering late...

The Teensy effects can be adapted for using inside MiniDexed. I am doing this currently for a compressor inside Synth_Dexed (dev branch).

The effect_platervb in the third-party folder could be used as reverb effects. I can try this after finishing the compressor - but ist will take a while...

dcoredump, 2022-03-30:
After finishing the compressor code inside Synth_Dexed I will add the effect_platervb (and perhaps later also effect_chorus and effect_delay).

dcoredump, 2022-04-01:
My latest PR includes now also the code for a stereo plate reverb. Currently the values are fixed (see minidexed.cpp).

probonopd, 2022-04-02:
Is the new compressor going to help here?

probonopd, 2022-04-03:
Thank you very much @dcoredump and @rsta2. Working and nicely integrated into the menu. So cool!

closed 2022-04-03 by probonopd

dcoredump, 2022-04-03:
> > I don't know why, but the audio output of MiniDexed (via PWM) is very silent...
> 
> This is caused by diving the summed output of `Dexed::getSamples()` by the number of active TGs to prevent clipping in `CMiniDexed::ProcessSound()`. Maybe there is a better solution, but I did not know one.

I think I should take a look into the mixer code. Perhaps I should try to port the template mixer class of Teensy...

probonopd, 2022-04-03:
Indeed it would be nice if the mixer could be improved in such a way that using e.g., 4 TGs of the same voice is not perceived "less loud" than using just 1 TG.