probonopd/MiniDexed #16: Add a build for Raspberry 3

probonopd, 2022-02-20:
The latest build contains a version for Raspberry 3 in addition to Raspberry Pi 4.
I could not get it to boot on the Raspberry 3 yet, though.

probonopd, 2022-02-20:
@rsta2 am I doing it wrong` Is there more to it than adding `kernel8.img`?

rsta2, 2022-02-20:
I guess, it's the serial MIDI interface, which is on GPIO12/13 for the RPi 4 now. The RPi 3 does not have an UART there, so the serial driver will assert in the early stages, when a message cannot be displayed yet.

If we want to have this serial MIDI interface on the RPi 3 too, then the startup code from circle-stdlib must be adapted, so that it does not use the serial device on GPIO14/15 any more. This can be done, without modifying the circle-stdlib project. I can do this and send a patch for MiniDexed, if you like.

probonopd, 2022-02-20:
That would be great @rsta2. Yes, I do think that serial MIDI is a useful thing to have.

rsta2, 2022-02-20:
Here come the patches. The file *circle_stdlib_app.h* must be placed into *src/* of MiniDexed. I tested playing via USB MIDI on the RPi 3B. Serial MIDI should be on GPIO14/15 on all RPi models now, but I don't think, that it works already.

[serial_patch.zip](https://github.com/probonopd/MiniDexed/files/8104249/serial_patch.zip)

@smuehlst Perhaps there should be the possibility to disable `mSerial` in the class `CStdlibAppScreen` of the circle-stdlib project itself, so that users do not need to adapt *circle_stdlib_app.h*, but currently I do not have an idea, how this could look like.

probonopd, 2022-02-20:
Works for me on the Raspberry Pi 3, thanks @rsta2 :+1:

closed 2022-02-20 by probonopd