probonopd/MiniDexed #18: Do wee need HDMI audio out?

dcoredump, 2022-02-20:
If not, the code should be dropped.

I personally don't need this because my experience is that an external HDMI->Audioconverter adds too much latency.

probonopd, 2022-02-20:
I don't think it is currently implemented in MiniDexed?
Personally I am not too interested in it, but I would not proactively remove code for it in case some users want to add that functionality later. Some TVs have dedicated game modes for minimizing latency; I have never systematically tested this for sound, though.

rsta2, 2022-02-21:
Please see issue #17 for a patch, which enables I2S and HDMI modes. Yes, there is some latency, when using HDMI, but the additional code effort for HDMI is small. On the Raspberry Pi 4 the display must be connected to the HDMI0 connector, so that it works.

probonopd, 2022-02-22:
Implemented in https://github.com/probonopd/MiniDexed/commit/a00bf55eea334e80eb1e9569ddb1221df852a2e5. Thanks @rsta2 :+1:

closed 2022-02-22 by probonopd

probonopd, 2022-02-23:
Successfully tested with a HDMI PC display. Works and the lag is ok here, ymmv.

Makes me wonder whether one of those inexpensive adapters would be usable as low-cost DAC alternatives...

[image]

HDMI to AV Converter Composite RCA CVBS HDMI2AV Adapter (e.g., [here](https://www.ebay.de/itm/224335650793?hash=item343b725fe9:g:LuUAAOSw0G1gF6bj))