probonopd/MiniDexed #20: Support Serial MIDI

probonopd, 2022-02-20:
Support Serial MIDI using a circuit like this:

[image]

Image Source: https://hackaday.com/tag/6n137/

* Raspberry Pi 4: GPIO13. This is a SoC number, not the position on the header (see [this](https://pinout.xyz/pinout/uart)).

probonopd, 2022-02-20:
We need to use `m_Serial (pInterrupt, TRUE, 5)` as discussed [here](https://github.com/rsta2/circle/issues/274#issuecomment-1030851026).

rsta2, 2022-02-20:
Attached the patches for serial MIDI (tested). The UART input (RX) is at GPIO15 now, not 13 any more!!! This has changed, because the RPi 3 does not has an UART at GPIO13.

[serial_midi_patch.zip](https://github.com/probonopd/MiniDexed/files/8104732/serial_midi_patch.zip)

probonopd, 2022-02-20:
Thanks @rsta2.

Getting

```
kernel.cpp: In member function 'virtual bool CKernel::Initialize()':
kernel.cpp:41:15: error: 'class AudioSynthDexed' has no member named 'Initialize'
   41 |  if (!m_Dexed.Initialize ())
      |               ^~~~~~~~~~
make: *** [../circle-stdlib/libs/circle/Rules.mk:143: kernel.o] Error 1
make: *** Waiting for unfinished jobs....
```

Did I do something wrong?

rsta2, 2022-02-20:
@probonopd Can it be, that you haven't updated the Synth_Dexed submodule reference to the new master? It still points to a commit from 5 days ago.

probonopd, 2022-02-20:
D'oh!

probonopd, 2022-02-20:
Build succeeded but I can't get sound via serial. Possibly it is the wiring on my side. What exact circuit are you using?

rsta2, 2022-02-20:
I tested this with a normal USB-to-serial-adapter with PC Linux and a small C program as MIDI sender. For this test I modified the serial baud rate in MiniDexed to 38400, because 31250 cannot be selected on the PC (at least I do not know how). The adapter provides 3.3V signals, so I can connect it directly to GPIO14/15.

The UART GPIO mapping is [here](https://github.com/rsta2/circle/blob/master/include/circle/serial.h#L33). That is the magic "5" in `m_Serial (pInterrupt, TRUE, 5)`, which was there on the RPi 4 to select the device 5 (GPIO12/13) and is now gone, what means that the [default value 0](https://github.com/rsta2/circle/blob/master/include/circle/serial.h#L83) is used. Device 0 is the standard UART available on all RPis via GPIO14/15.

rsta2, 2022-02-21:
@probonopd One hint: The serial MIDI will only work, when USB MIDI is disconnected. You have to remove your USB MIDI keyboard for a test.

If you have a free Raspberry Pi Pico, you could use [this project](https://github.com/rsta2/pico/tree/main/midi_adapter) to connect a sequencer program, running on a PC, to MiniDexed.

probonopd, 2022-02-21:
> The serial MIDI will only work, when USB MIDI is disconnected. You have to remove your USB MIDI keyboard for a test.

Great hint, sure enough I had the USB MIDI connected in addition to Serial MIDI. It is still not working though, but most likely my selfmade MIDI Serial circuit to be the culprit. Will redo it with the help of an oscilloscope and logic analyzer once I find the time for it (might take a while). If anyone is faster and is successful in e.g., using a Serial MIDI keyboard, please let me know!

probonopd, 2022-03-20:
Note to self: Retest this, then close.

probonopd, 2022-03-21:
> Device 0 is the standard UART available on all RPis via GPIO14/15.

@rsta2 are you saying we have bidirectional serial MIDI even?

rsta2, 2022-03-21:
The UART driver is bidirectional, but it is currently not used in TX direction.

probonopd, 2022-04-05:
Tested, working-

closed 2022-04-05 by probonopd