probonopd/MiniDexed #21: Show patch name on HD44780 display

probonopd, 2022-02-20:
Inexpensice HD44780 displays like [this one](https://www.berrybase.de/sensoren-module/displays/alphanumerische-displays/alphanumerisches-lcd-16x2-gr-252-n/gelb?c=123) are [supported](https://github.com/rsta2/circle/blob/master/addon/display/sample/hd44780/README) in Circle.

Looks like the original DX7 uses a similar 16x2 display.

[image]

probonopd, 2022-02-26:
> how to set contrast on this display

A 2.2k resistor between the contrast pin (V0) and GND does the job for me. (Without this I don't see anything, just shorting it out makes the text appear but is too dark.)

rsta2, 2022-02-26:
Patch for the 44780 branch attached. The display library must be build separately and added to *src/Rules.mk*. `make -j` is better than `make -j$(nproc)`. Don't ask me why, but it works quicker here.

[build.zip](https://github.com/probonopd/MiniDexed/files/8147333/build.zip)

rsta2, 2022-02-26:
I would add the `CHD44780Device m_LCD;` member to the class `CMiniDexed`, not to `CKernel`. You also have to move the initializer in the constructor and the call to `m_LCD.Initialize()` to `CMiniDexed`. Then you can call `s_pThis->m_LCD.Write()` in `MIDIPacketHandler()`.

probonopd, 2022-02-26:
Makes sense to me. Will do. Thanks!

rsta2, 2022-02-26:
The method `LCDWrite()` is declared in *minidexed.h*, but not implemented in *minidexed.cpp*:

```
#include <circle/util.h>    // probably to be added to the includes

void CMiniDexed::LCDWrite (const char *pString)
{
	m_LCD.Write (pString, strlen (pString));
}
```

Did I forget to mention this? Sorry.

closed 2022-02-26 by probonopd

probonopd, 2022-02-27:
I noticed that adding a display to the Raspberry Pi 1 adds significant noise to PWM sound output.
It is not as bad on e.g., the Raspberry Pi 2.

Maybe adding a capacitor somewhere could help improve this a bit?

commit b871fa17 (2022-02-26): Show program name on optional HD44780 display

Close #21
Thanks @rsta2