probonopd/MiniDexed #22: Consider support for optional rotary encoder

probonopd, 2022-02-20:
Consider support for a [rotary encoder with button](https://www.berrybase.de/bauelemente/passive-bauelemente/potentiometer/drehimpulsgeber/drehregler/rotary-encoder-mit-breakoutboard-ohne-gewinde-und-mutter).

This could, together with a [display](https://github.com/probonopd/MiniDexed/issues/21), be used for menu based navigation and (simple) parameter editing. But maybe it is not worth the effort to implement menu-based navigation and we should focus on e.g., a a way to edit configuration over the nework instead.

Let's not do anything for now unless this is requested by users. Please comment if you have thoughts regarding this.

probonopd, 2022-03-06:
https://github.com/rsta2/circle/issues/280#issuecomment-1059516929

> Tried to use the KY-040 together with MiniDexed. It works! I can switch between program change and bank select with one click and change the voices or banks by turning the knob. Pressing the switch for 3 seconds reboots the system.

Great! Please do send a PR or patch for MiniDexed for this code. Thanks @rsta2.

rsta2, 2022-03-06:
OK, I will do tomorrow.

closed 2022-03-20 by probonopd