probonopd/MiniDexed #23: Build for Raspberry Pi 2 fails

probonopd, 2022-02-20:
Trying to build a 32-bit version for Raspberry Pi 2 too but running into

```
config.status: executing depfiles commands
make[2]: Leaving directory '/home/runner/work/MiniDexed/MiniDexed/circle-stdlib/build/circle-newlib'
make[1]: Leaving directory '/home/runner/work/MiniDexed/MiniDexed/circle-stdlib/build/circle-newlib'
make[1]: *** [Makefile:879: all] Error 2
make: *** [Makefile:17: newlib] Error 2
Error: Process completed with exit code 2.
```

https://github.com/probonopd/MiniDexed/runs/5265872462?check_suite_focus=true#step:6:1184

It looks like some issue with circle-newlib. @smuehlst can it be built for 32-bit systems such as Raspberry Pi 2?

Maybe I am missing some necessary "cleanup" steps after having built the 64-bit versions?

smuehlst, 2022-02-20:
Sure, circle-stdlib can be built for all platforms that Circle supports. Very likely the cleanup is incomplete. Try running `make mrproper` from the circle-stdlib root directory, this should clean up all build results.

probonopd, 2022-02-20:
Still getting `make: *** [Makefile:17: newlib] Error 2`.

> this should clean up all build results

Does it do this recursively, also for newlib?

smuehlst, 2022-02-20:
> Still getting `make: *** [Makefile:17: newlib] Error 2`.

Can you show the complete sequence of commands that leads to this error? And what command invocation causes make to stop with the error?
 
> > this should clean up all build results
> 
> Should it do this recursively, also for newlib?

Yes, it should. See the circle-stdlib Makefile:

https://github.com/smuehlst/circle-stdlib/blob/47d9deb580f3d91201c2e275d3412abfbafdf156/Makefile#L77-L78

```
	-$(MAKE) -C $(NEWLIB_BUILD_DIR) clean
	-test -n "$(NEWLIB_INSTALL_DIR)" && rm -rf "$(NEWLIB_INSTALL_DIR)"/*
```

probonopd, 2022-02-20:
Can you show the complete sequence of commands that leads to this error?

* https://github.com/probonopd/MiniDexed/blob/main/.github/workflows/build.yml which invokes
* https://github.com/probonopd/MiniDexed/blob/main/build.sh

smuehlst, 2022-02-20:
You are doing the `make mrproper` in the wrong directory. I wrote:

> Try running `make mrproper` from the circle-stdlib root directory ...

You are doing this:

https://github.com/probonopd/MiniDexed/blob/bd5a6c4acd5c3b9e8aee3e70d5f89924cfc3c6c2/build.sh#L18-L20

```
cd circle-stdlib/
make clean || true
( cd build/circle-newlib/ ; make mrproper ) || true
```

What I suggested was this:

```
cd circle-stdlib/
make mrproper
```

probonopd, 2022-02-21:
I can confirm that building _locally_ for the Raspberry Pi 1 succeeds, and the image works. It looks like the Raspberry Pi 1 can handle a single Dexed instance without effects really well, so can do what the DX7 did. Will be interesting to see how many Dexed instances and effects we can squeeze onto the Pi 1.

probonopd, 2022-02-21:
Now the build succeeds but only `kernel7.img` ends up in the image...

probonopd, 2022-02-21:
We now have builds for all Raspberry Pi models, and have 4 kernels:

```
  adding: kernel.img (deflated 61%)
  adding: kernel7.img (deflated 54%)
  adding: kernel8-rpi4.img (deflated 63%)
  adding: kernel8.img (deflated 63%)
```

Thanks for your help @smuehlst :+1:

closed 2022-02-21 by probonopd

probonopd, 2022-02-21:
It looks like `config32.txt` needs to be renamed to `config.txt` in order for this to boot on Raspberry Pi 1 (and probably 2).
Is there a way to merge the `config32.txt` and `config64.txt` files in such a way that the system boots on both kinds of systems (64-bit on 3 and 4, 32-bit otherwise)? Or should we use 32-bit for 3 and 4 as well? cc @rsta2

rsta2, 2022-02-22:
Good question! Both is possible. In MiniSynth Pi I currently use 32-bit mode for all models, but with this *config.txt* you can use 64-bit for RPi 3, 4, 400 and Zero 2, and 32-bit for all other models:

```
#
# config.txt for MiniDexed
#

[pi02]
arm_64bit=1
kernel=kernel8.img

[pi3]
arm_64bit=1
kernel=kernel8.img

[pi3+]
arm_64bit=1
kernel=kernel8.img

[pi4]
arm_64bit=1
armstub=armstub8-rpi4.bin
kernel=kernel8-rpi4.img
```