probonopd/MiniDexed #7: Sound goes bad when 8 keys are pressed, clipping?

probonopd, 2022-02-20:
Sound goes bad when 8 keys are pressed simultaneously.

Is this caused by clipping?
Does this also happen in MicroDexed?

cc @dcoredump

dcoredump, 2022-02-20:
I am very sure that it is clipping. Even when I play the FM-PIano 5-voice very dynamically, it starts clipping. I'll have to have a look at it.

rsta2, 2022-03-04:
@dcoredump are you still working on this? It would be great to have a solution here!

dcoredump, 2022-03-04:
Sorry, I am currently ill and don't have time to code...

There are several solutions for this (sorted by implementation effort):

1. Scale down the calculated values inside `CMiniDexedPWM::GetChunk()`
2. Use `m_pDexed->setGain(0.8);` to scale down inside `Synth_Dexed`
3. The real DX7 has a kind of compressor after adding all voices (someone told me that). But adding a good compressor needs some CPU time, so I used the first suggestions to avoid clipping. For a RPi an additional compression would not be a problem if someone finds code which fits. In fact I added a small and simple compression-like algorithm at https://github.com/probonopd/Synth_Dexed/tree/master/src/synth_dexed.cpp (look for `USE_SIMPLE_COMPRESSOR`)

rsta2, 2022-03-04:
Ah OK. Thank you for explaining this! I tried the suggested solutions with these results:

1. I simply divided the samples by 2 before writing them to the chunk buffer. Unfortunately this did not work. I guess some samples returned by `Dexed::getSamples()` where already greater than 1.0 or smaller than -1.0, before converting them to `int32_t` values.
2. Does work with `setGain(0.4)`, 0.5 is already too much, when hammering to 16 keys at once.
3. I enabled `USE_SIMPLE_COMPRESSOR` and multiplied the samples from `sumbuf[]` with `vuSignal` in `getSamples()`. Unfortunately this did not help and sounds also a bit strange. I do not know enough about this compressor to manipulate the parameters. This is perhaps something for you later.

So for the moment solution 2. is probably the best fitting.

Get well soon!

probonopd, 2022-03-04:
Get well soon @dcoredump.

> The real DX7 has a kind of compressor after adding all voices (someone told me that).

https://www.righto.com/search/label/dx7 by mastermind Ken Shirriff is a fascinating read if you _really_ want to know what is going on in a DX7. Looks like they did some interesting trickery in the output circuitry to work around of the hardware limitations at the time. I don't think we need to do everything in the exact same way as it was done ~40 years ago, but maybe there are some aspects that might still be of interest? I haven't digested all of it yet, but fascinating it is.

rsta2, 2022-03-05:
> https://www.righto.com/search/label/dx7 by mastermind Ken Shirriff is a fascinating read if you really want to know what is going on in a DX7.

Thanks @probonopd for mentioning this! I didn't know, that the DX7 only had a 12 (+2) bit DAC. So we are already a bit (no 2 bits) better with 16 bit. :)

A dynamic range compressor can also be necessary after mixing 8 DX instances together, when we have reached that. I found this interesting [paper](https://www.eecs.qmul.ac.uk/~josh/documents/2012/GiannoulisMassbergReiss-dynamicrangecompression-JAES2012.pdf) on this topic.

dcoredump, 2022-03-05:
> > https://www.righto.com/search/label/dx7 by mastermind Ken Shirriff is a fascinating read if you really want to know what is going on in a DX7.
> 
> Thanks @probonopd for mentioning this! I didn't know, that the DX7 only had a 12 (+2) bit DAC. So we are already a bit (no 2 bits) better with 16 bit (with I2S and HDMI). :)
> 
> A dynamic range compressor can also be necessary after mixing 8 DX instances together, when we have reached that. I found this interesting [paper](https://www.eecs.qmul.ac.uk/~josh/documents/2012/GiannoulisMassbergReiss-dynamicrangecompression-JAES2012.pdf) on this topic.

Maybe [this](https://github.com/chipaudette/OpenAudio_ArduinoLibrary) is a useful implementation for a DRC.

rsta2, 2022-03-05:
> Maybe [this](https://github.com/chipaudette/OpenAudio_ArduinoLibrary) is a useful implementation for a DRC.

Looks like it would require some effort to port this. Unfortunately I'm not familiar with audio processing on Arduino(-like) architectures. From my point of view it would be great to have everything, which does sophisticated audio processing (e.g. the synth, effects) in a class in the Synth_Dexed project to be used from MiniDexed. I don't know, if this is possible.

I tried this simple addition to `Dexed::getSamples()` and it also fixes the clipping problem:

```
diff --git a/src/synth_dexed.cpp b/src/synth_dexed.cpp
index f407898..dea67c0 100644
--- a/src/synth_dexed.cpp
+++ b/src/synth_dexed.cpp
@@ -461,7 +461,14 @@ void Dexed::getSamples(uint16_t n_samples, int16_t* buffer)
   arm_float_to_q15(sumbuf, buffer, AUDIO_BLOCK_SAMPLES);
 #elif defined(__circle__)
   for (i = 0; i < n_samples; i++)
-    buffer[i]=(sumbuf[i]*32768.0)+0.5;
+  {
+    int32_t val = (sumbuf[i]*32768.0)+0.5;
+    if (val < -32768)
+      val = -32768;
+    else if (val > 32767)
+      val = 32767;
+    buffer[i] = val;
+  }
 #endif
 }
 
```

probonopd, 2022-03-06:
Is this effectively a limiter?

rsta2, 2022-03-06:
I don't know the exact terminology of "limiter", but I would think so. The problem seems to be, that some samples exceed the available 16 bit range and without this code, the value "wraps". This means a sample of 40000, which is greater than the maximum of 32767, will be written as -25536 in the end. It's better to use the maximum of 32767 in this case. Testing this, it sounds much better than in the unlimited case. Some distortion is still possible, but I cannot hear it. But you should test it too.

probonopd, 2022-03-06:
@rsta2 this is definitely an improvement. But in some cases one still can hear that clipping is going on, which can probably be resolved by [reducing volume](https://github.com/probonopd/MiniDexed/issues/9).
To reproduce, load the default voice "BRASS 1" and press 8 white keys simultaneously starting from middle C.

closed 2022-03-06 by probonopd

rsta2, 2022-03-06:
Yes, I can hear the distortion now, but not as bad as before. And yes, we need to provide a method to change the volume, via MIDI and/or in the UI.