probonopd/MiniDexed #8: Output MIDI messages for debugging and development

probonopd, 2022-02-20:
Let's output MIDI messages to the screen and/or to serial for debugging and development.

We will also need this if we want MiniDexed to act upon special MIDI messages outside of Synth_Dexed.

Since MIDI is processed inside [Synth_Dexed](https://codeberg.org/dcoredump/Synth_Dexed), would we need a callback of some sort in order to get the MIDI messages? 

@dcoredump wdyt?

dcoredump, 2022-02-21:
> Let's output MIDI messages to the screen and/or to serial for debugging and development.
>
> We will also need this if we want MiniDexed to act upon special MIDI messages outside of Synth_Dexed.
>
> Since [...]

Hm, I'm a little undecided at the moment how best to do this. With MicroDexed I use the MIDI-library, where you can register handlers for various events. I use the following handlers:
```
void handleNoteOn(byte inChannel, byte inNumber, byte inVelocity);
void handleNoteOff(byte inChannel, byte inNumber, byte inVelocity);
void handleControlChange(byte inChannel, byte inData1, byte inData2);
void handleAfterTouch(byte inChannel, byte inPressure);
void handlePitchBend(byte inChannel, int inPitch);
void handleProgramChange(byte inChannel, byte inProgram);
void handleSystemExclusive(byte *data, uint len);
//void handleSystemExclusiveChunk(const byte *data, uint len, bool last);
void handleTimeCodeQuarterFrame(byte data);
void handleAfterTouchPoly(byte inChannel, byte inNumber, byte inVelocity);
void handleSongSelect(byte inSong);
void handleTuneRequest(void);
void handleClock(void);
void handleStart(void);
void handleContinue(void);
void handleStop(void);
void handleActiveSensing(void);
void handleSystemReset(void);
//void handleRealTimeSystem(void);
```

Maybe we should use such a system too?

rsta2, 2022-02-21:
[This logging facility](https://circle-rpi.readthedocs.io/en/44.3/basic-system-services/system-log.html) may be of interest.

probonopd, 2022-02-21:
> With MicroDexed I use the [MIDI-library](https://github.com/FortySevenEffects/arduino_midi_library), where you can register handlers for various events. I use the following handlers

Sounds exactly like what we need :+1:

rsta2, 2022-02-22:
I think, all Raspberry Pi specific things should go to the MiniDexed project, DX7 specific things into Synth_Dexed. This would mean, that the `AudioSynthDexed*` classes for the Raspberry Pi should be moved to the MiniDexed project.

dcoredump, 2022-02-22:
> I think, all Raspberry Pi specific things should go to the MiniDexed project, DX7 specific things into Synth_Dexed. This would mean, that the `AudioSynthDexed*` classes for the Raspberry Pi should be moved to the MiniDexed project.

Yes, that would make things clearer. But `AudioSynthDexed` should be renamed in MiniDexed to avoid duplication of names (`AudioSynthDexed` must be left intact for MicroDexed).

rsta2, 2022-02-22:
> Yes, that would make things clearer. But AudioSynthDexed should be renamed in MiniDexed to avoid duplication of names (AudioSynthDexed must be left intact for MicroDexed).

Yes, to rename it is a good idea.

rsta2, 2022-02-22:
Should I do this update to move the classes `AudioSynthDexed*` out of the Synth_Dexed project and to the MiniDexed project and rename the classes? This will affect a number of source code files.

probonopd, 2022-02-22:
if @dcoredump does not object, I think this would be fantastic.

rsta2, 2022-02-23:
@dcoredump did not object, so here come the patches. It moves the Raspberry Pi specific code from the classes `AudioSynthDexed*` from the Synth_Dexed project to the MiniDexed project. The four new files in the *src/* directory go to *MiniDexed/src/*.

There is an addtiional patch *build.sh.patch* included, which enables the system option `SAVE_VFP_REGS_ON_IRQ` after configure. This is necessary, because `Dexed::getSamples()` is called from an IRQ handler and uses floats.

There is some Circle specific code left in *synth_dexed.h*, which is required there, because it is otherwise provided by *Arduino.h*, how I understand it.

The support for an ASCII (PC) USB keyboard is working now. I copied two files from MiniSynth Pi for that purpose, because the code from the *sample/29-miniorgan* did not fit here perfectly.

I did not pay attention to the headers of the source code files. Perhaps we need a common file header for the MiniDexed source code files, which provides the license info.

[patch.zip](https://github.com/probonopd/MiniDexed/files/8123127/patch.zip)

dcoredump, 2022-02-23:
> if @dcoredump does not object, I think this would be fantastic.

Sorry, I was too busy to answer in time. Removing the _circle specific_ `AudioSynthDexed` class would be good. So I only have to manage one hardware base inside the library again.

I will try the patches soon.

dcoredump, 2022-02-23:
Patches are working for me! Great job. I have removed the circle stuff from `Synth_Dexed`. Currently working on stereo pan.

probonopd, 2022-02-23:
Merged and built! Thanks @rsta2 :+1:

probonopd, 2022-02-23:
ASCII keyboard is working for me, including key-up.

probonopd, 2022-02-24:
So to print MIDI messages, add in`src/minidexed.cpp` `CMiniDexed::MIDIPacketHandler` something like `debug_hexdump ((void *) 0x0, nLength, pPacket)`?

rsta2, 2022-02-24:
It's a little more effort, because `CMiniDexed::MIDIPacketHandler()` is called from the USB MIDI IRQ handler. Displaying a lot of MIDI messages with scrolling the screen would kill the timing. This can be solved with a write buffer, like in the attached patch. You can simply use `printf()` now, if `<stdio.h>` is included.

There is already a MIDI dump implemented in the patch. To disable it, you have to out-comment the `#define MIDI_DUMP` in the first lines of *src/minidexed.cpp*.

BTW. `debug_hexdump()` gets a `void *` to the buffer, a length and maybe a `const char *` with a label for the log lines. But I didn't used it here, because the USB MIDI messages are maximal 3 bytes long.

[midi_dump.zip](https://github.com/probonopd/MiniDexed/files/8135463/midi_dump.zip)

probonopd, 2022-02-24:
Thank you very much @rsta2

commit 1617763a (2022-02-24): Output MIDI messages for debugging and development

https://github.com/probonopd/MiniDexed/issues/8#issuecomment-1050231502
Closes #8
Thanks @rsta2
closed 2022-02-24 by probonopd