probonopd/MiniDexed #9: Support the volume knob on MIDI controllers

probonopd, 2022-02-20:
Support the volume knob on MIDI controllers.

(Once #8 is implemented I can say which MIDI message the volume knob sends.)

dcoredump, 2022-02-20:
Will do this with adding MIDI processing support for Synth_Dexed (see also #10, #11)

probonopd, 2022-03-06:
The volume MIDI messages are `B0 07 00` (knob all the way left) to `B0 07 7F` (knob all the way right).

probonopd, 2022-03-13:
Likewise, `B0 01 ...` for modulation and `E0 00 ...` for Pitch Bend.

rsta2, 2022-03-14:
I was so bold and implemented the volume control (using MIDI CC and via the UI) already in #47. ;) I also tried modulation (using `Dexed::setModWheel()`) and pitch bend (using `setPitchbend()`), but this didn't work. Or I wasn't able to "move the wheel" quick enough, because I can only simulate the MIDI messages by sending hex bytes manually. I also don't know, if it depends on the loaded voice, if these functions have an effect.

BTW. Pitch bend was already implemented, but I think this MIDI message has two parameter bytes (`0xE0 LSB MSB`), and the MSB is not considered yet.

dcoredump, 2022-03-14:
> I was so bold and implemented the volume control (using MIDI CC and via the UI) already in #47. ;) I also tried modulation (using `Dexed::setModWheel()`) and pitch bend (using `setPitchbend()`), but [...]

ModWheel depends on the routing of the voices. Some voices don't have a target for the ModWheel. 

> BTW. Pitch bend was already implemented, but I think this MIDI message has two parameter bytes (`0xE0 LSB MSB`), and the MSB is not considered yet.

Yes, Pitchbend is 14 bit (LSB, MSB)!

probonopd, 2022-03-14:
> E0 00 ... for Pitch Bend.

To be precise, `E0 00 40` (normal pitch wirthout bending) all the way down to `E0 00 00` and all the way up to `E0 7F 7F`.

rsta2, 2022-03-14:
@dcoredump Ah OK, I probably used a voice, which don't use the ModWheel. Thanks for info.

@probonopd This is something, I did wonder about today. When the zero pitch bend level is 0x4000 (I have seen this value too in some doc), then you have to subtract 0x4000 from the value before calling `Dexed::setPitchbend()`. But inside this function, it adds 0x2000 only to get back to the 0..16383 range. I don't understand this.

https://codeberg.org/dcoredump/Synth_Dexed/src/branch/master/src/dexed.cpp#L866

probonopd, 2022-03-14:
Does https://sites.uci.edu/camp2014/2014/04/30/managing-midi-pitchbend-messages/ help?

rsta2, 2022-03-15:
Here is a patch, which should implement Pitch Bend and Modulation CC. Unfortunately I do not have such controls, so I cannot test it.

[pitch-bend-modulation.zip](https://github.com/probonopd/MiniDexed/files/8251099/pitch-bend-modulation.zip)

probonopd, 2022-03-15:
Thanks @rsta2, unfortunately it does not seem to have any audible effect.
@dcoredump can you reproduce this?

rsta2, 2022-03-15:
@probonopd This was also my impression, when trying to fake a Pitch Bend and modulation by sending the MIDI bytes manually with a tool. I tried several voices. But I thought, that you need a real controller to test this.

dcoredump, 2022-03-16:
> @dcoredump can you reproduce this?

Sorry, but I have no equipment which can send MIDI-CC right here - my master keyboard is in the rehearsal room...

rsta2, 2022-03-16:
I was looking around in Synth_Dexed, and now Pitch Bend and Modulation does work (e.g. with the "BRASS 1" voice), when you apply the attached patch to the version from the *pb-mod* branch. There are two major things to do, that it works:

* The controllers must be configured using `Dexed::setPBController (range, step)` and `Dexed::setMWController (range, assign, mode)`. Pitch Bend range and step can be from 0 to 12. Modulation Wheel range can be 0 to 99, assign 0 to 7 and mode 0 to 2. Currently I don't know, if these values should be configurable in `minidexed.ini` or somewhere else?

* Have to call `Dexed::ControllersRefresh()` after `Dexed::setModWheel()`.

I tested it, by manually sending MIDI bytes, and there was definitely some reaction. But we have to find out more about the different configuration parameters.

[pb-mod-2.zip](https://github.com/probonopd/MiniDexed/files/8273925/pb-mod-2.zip)

rsta2, 2022-03-17:
I installed Dexed on Windows. These controllers are switched off by default there. I guess we have to find out a setup by our self here. In the [TX802 Manual](https://usa.yamaha.com/support/manuals/index.html?l=en&c=music_production&k=tx802) on page 22 there is the description for the Pitch Bend parameters, on page 23 for Mod wheel.

probonopd, 2022-03-28:
Volume knob, modulation, and pitch bend on MIDI controllers are working and tested successfully. In case some further tuning of parameters is needed, we can always open separate tickets.

Thanks @rsta2 for implementing this!

closed 2022-03-28 by probonopd