probonopd/MiniDexed #24: Understanding and storing Performances (PCED?)

probonopd, 2022-02-22:
Our objective is to ultimately run multiple Dexed instances at the same time (#12).
Thinking about a good, robust, and simple way to edit and store "patches" for performances (=combinations of multiple Dexed instances).

Sound generators like the TX816 and TX802 use _performance sysex_ formats, but I could not find the data formats in a quick search.

Thus, I am speculating a bit here:
* Dexed understands sysex for Voice Edit Parameters (VCED) which can be converted into a binary format (VMEM)
* Similarly, once we can run more than one instance of Dexed (ideally at least 8), it will be desirable to understand Performance Edit Parameters (PCED) which can be converted into a binary format (PMEM).

Maybe the formats are documented _somewhere_, since there are third-party tools that _seemingly_ can edit performances: 
* http://www.fm-alive.com/pages/DXM.aspx
* https://coffeeshopped.com/patch-base/editor/yamaha/tx802
* https://squest.com/Products/MidiQuest12/Instruments/YamahaTX802/

The publicly available _FM Legacy Converter User Guide_ lists _parameters that are converted_ respectively _not converted_ in the appendix. This gives us an idea about the data fields needed to emulate _performance_ functionality.

---

Formats like VCED and PCED are made to be machine-readable, not so much human-readable. VMEM and PMEM even more so; they were designed around memory constraints of the early 80s.

So the question is, would it be a good idea to introduce human-readable formats (e.g, "VINI" and "PINI")?

Since the `.ini` data format is widely used and natively supported in Dexed, I was thinking along the lines of this:

```
[Performance]
Performance_Name=
...

[INST 1]
Voice_Number=
Detune=
Note_Limit_Low=
Note_Limit_High=
...

[INST 2]
...
```

Also see page 29 of the TX-802 Owners Manual.

Table Add-10 in DX7II_MIDI_DataFormat.pdf mentions "PERFORMANCE Parameters (PCED, PMEM format)". Those are not for the TX-802 but possibly are similar? https://www.manualslib.com/manual/779968/Yamaha-Tx81z.html?page=74#manual mentions "Performance Edit Parameters (PCED)". 

And while we are at it, we might also want to convert those voice sysex (VCED, VMEM) into something more human-readable like

```
[Voice]
Voice_Name=
Algorithm_Selector=
Feedback_Level=
Osc._Sync.=
Transpose=

[OP 1]
...

[OP 2]
...
```

**NOTE:** We may want shorter names for those (ideally standard names), e.g., "ALS" for "Algorithm_Selector" and "FBL" for Feedback_Level. Also see page 30 of the TX-802 Owners Manual and table Add-8 in DX7II_MIDI_DataFormat.pdf.

What do you people think?

probonopd, 2022-02-23:
__UPDATE:__ Looks like MicroDexed already has code to handle performance sysex:
https://codeberg.org/dcoredump/MicroDexed/src/branch/master/dexed_sd.cpp#L576

It even has code for effects sysex. Are those compatible to any Japanese OEM device @dcoredump?

So we "just" need to port `dexed_sd.cpp` over to Circle?

dcoredump, 2022-02-23:
> __UPDATE:__ Looks like MicroDexed already has code to handle **performance** sysex:
> https://codeberg.org/dcoredump/MicroDexed/src/branch/master/dexed_sd.cpp#L576
>
> It even has code for **effects** [...]

Yes, in fact this is what I want to do. But I think it would be better to put the code into Synth_Dexed (without the file handling code) due to it has no dependence to the used OS/uC. So it is easier to maintain errors in the code.

Wdyt?

probonopd, 2022-02-23:
Agree. Same for effects.

probonopd, 2022-02-27:
[TX802_Factory_Performances.syx](https://github.com/weekend-at-bernies/sysex_utils/blob/master/patches/yamaha_TX802/TX802_Factory_Performances.syx) is available on GitHub.

> 	This is the original factory performance data for the
	TX802.  There are 54 performances.  The remander of them
	are the "Init" performance with the name removed for 
	better esthetic value. These were programmed buy hand and
	tripple checked against the original performance lists
	that came with the TX802.

Let's use this file as the test for loading performances.

https://github.com/weekend-at-bernies/sysex_utils/tree/master/patches/yamaha_TX802 has more information.

rsta2, 2022-03-23:
I had a look at the sysex performance file format and extended the class `CSysExFileLoader` respectively. Up to 32 files can be loaded from */sysex/performance*. The filenames have the same format as used for voice banks. There is a method `GetPerformance()`, which returns the data for one performance in the `PMEM` format:

```
struct TPMEMData
{
	struct
	{
		uint8_t RXCH	: 5,
			VCHOFS	: 3;
	}
	VCHOFS_RXCH[8];

	uint8_t	VNUM[8];
	uint8_t	MTTNUM[8];
	uint8_t	OUTVOL[8];

	struct
	{
		uint8_t	OUTCH	: 2,
			KASG	: 1,
			DETUNE	: 4,
			Ignore	: 1;
	}
	DETUNE_KASG_OUTCH[8];

	uint8_t	NLMTL[8];
	uint8_t	NLMTH[8];

	struct
	{
		uint8_t NSHFT	: 6,
			FDAMP	: 1,
			Ignore	: 1;
	}
	FDAMP_NSHFT[8];

	uint8_t PNAM[20];
};
```

Currently the meaning of the parameters VCHOFS (voice channel offset), MTTNUM (tuning) and KASG (key assign group) is not fully clear to me.

I could send a PR for the extended class `CSysExFileLoader`. I tested it by dumping the data of some performances. There is a method `DumpPerformance()` included.

What is not clear at the moment is, how this can be added to the UI? Maybe a menu item "PERFORMANCE", where one can "scroll" though the different performances of the first loaded file, would be a possible solution. The default value can be "NONE", which means, that the parameters have to be configured manually as before.

If we want to use the factory performance file for testing, we also need the factory voice banks, because the performances reference these voices. I don't know, if it is legal to include these voice banks into MiniDexed?

probonopd, 2022-03-24:
Here are first spontaneous thoughts. Will need to think more about this. 

Looking at page 14+ of the TX802 user manual, I am not sure whether Dexed supports everything that the TX802 supports. Also, I don't find a description of "voice channel offset", "tuning", and "key assign group" in the user manual. In this case, shall we just leave the such parameters in performance sysex as "ignored"/unimplemented for now? Maybe we can find a TX802 owner to help us a bit with information at some time to understand those.

But reading the User user manual, I have wonder which of the performance features are really widely understod by users and actually used (I think we have the most important ones covered already in the performance `.ini`) and which ones may be rather esoteric and seldomly used. Maybe we should just stay with performance `.ini` files, add a way to switch between different performance `.ini` files, and maybe optimize them a bit? Wdyt?

As for the GUI, maybe we need a "top level menu" that would one allow to, e.g., "Select Performance", "Edit TG1-TG8" (what we have so far), etc.?

probonopd, 2022-03-24:
> If we want to use the factory performance file for testing, we also need the factory voice banks, because the performances reference these voices.

With "use this file as the test for loading performances" I meant during development and testing. Not to ship those performances by default. So we don't have to ship the matching voices by default.

rsta2, 2022-03-24:
TX802 compatibility is not easy to reach. Perhaps we can go for this in a future version. For now I would agree to use multiple .ini files. We could provide a tool for converting performance .syx files to .ini files.

I think we cannot support all performance parameters of the TX802, but the note low and high limits and note shift are still possible, also in the .ini file. This would allow to split a keyboard for playing different voices.

We should think about the UI menus again from a user perspective. What we have at the moment was only a first approach. I think we need some menu tree and should use the rotary encoder in a way, that it is easy to use, for example:

* Chose a menu item or parameter value by turning the knob
* Click the switch once to enter a menu item
* Double-click the switch to return one level in the menu tree
* Hold for at least one second to go to menu root (as already implemented)

A question is also, if it makes sense to provide all voice parameters for editing? It's probably much more easy to use the original Dexed to edit voices, but to click through many menus in the MiniDexed UI to change some parameter.

probonopd, 2022-03-25:
Fully agree.

I don't think we need TX802 compatibility, unless soemone really has the need for it, wants to do it, and maybe a TX802 owner steps up to help us.
Low and high limits and note shift would be useful indeed, would be great to have them available.

For the UI menus I was thinking of extactly something as you are describing, which is very similar to how typicsl 3D printer menus work.
Maybe we can use graphical arrow symbols (triangles) in the LCD to show what turning a knob changes. Like this:

[image]

Turning the knob here would change Detune to another parameter. Clicking here would change to

[image]

Turning the knob here would change the value of Detune.

Maybe we could use a 1-digit triangle symbol instead of 2-digit `<-`, `->`. Or maybe we should use the blinking cursor for this purpose?

I have never had a chance to use an actual DX7 but it uses the same exact display as we do. Maybe it'd be worthwhile to study which

rsta2, 2022-03-25:
Yes, that UI idea is good. The character set of the LCD is [here](https://cdn-reichelt.de/documents/datenblatt/A500/EA_W162-N3LED.pdf#page=2). Perhaps we should use the arrows 0x7E and 0x7F. It is also possible to define and use eight user characters (5x7 pixel matrix). The blinking block cursor is also available, but I personally would prefer the arrows. I had a look into the DX7 Manual, which gives some info about the available menus, but yes, we have only one knob, so we need a different concept.

I'm currently testing the USB FIQ support for Circle to ensure nothing is broken after some modifications. When I'm ready with this, I would implement the note limits and shift, and when there would be a concept for the (new) UI, I could work on this.

probonopd, 2022-04-05:
Closing for now as we are using `.ini` files for performances now. Please comment below if you think this (or parts of it) should be reopened.

probonopd, 2023-11-04:
Reopening due to renewed interest in being able to parse DX1/DX5/TX7/... performance sysex.

Based on the information from https://github.com/probonopd/MiniDexed/discussions/554 (thanks @BobanSpasic) this is what I have:

```
// Load TX7 performance data from a file
// and show the values of the parameters

#include <iostream>
#include <fstream>
#include <iomanip>
#include <string>
#include <vector>
#include <algorithm>
#include <iterator>
#include <stdexcept>
#include <cstdlib>
#include <cstring>
#include <cmath>
#include <cassert>

using namespace std;

typedef unsigned char byte;
typedef byte PMEM_Dump[64];
typedef byte PCED_Dump[94];

// PCED (Performance Edit)
// PCED contais performance data, including settings for voices, effects, and other parameters

typedef struct {
  byte A_NU0;                      //       Reserved (not used in TX7)
  byte A_NU1;                      //       Reserved (not used in TX7) (MIDI Channel)
  byte A_PolyMono;                 //       0-1
  byte A_PitchBendRange;           //       0-12
  byte A_PitchBendStep;            //       0-12
  byte A_PortamentoTime;           //       0-99
  byte A_PortaGlissando;           //       0-1
  byte A_PortamentoMode;           //       0-1
  byte A_NU8;                      //       Reserved (not used in TX7)
  byte A_ModWheelSens;             //       0-15  translates to 0-99 for DX7
  byte A_ModWheelAssign;           //       0-7
  byte A_FootCtrlSens;             //       0-15  translates to 0-99 for DX7
  byte A_FootCtrlAssign;           //       0-7
  byte A_AfterTouchSens;           //       0-15  translates to 0-99 for DX7
  byte A_AfterTouchAssign;         //       0-7
  byte A_BrthCtrlSens;             //       0-15  translates to 0-99 for DX7
  byte A_BrthCtrlAssign;           //       0-7
  byte A_NU17;                     //       Reserved (not used in TX7)
  byte A_NU18;                     //       Reserved (not used in TX7)
  byte A_NU19;                     //       Reserved (not used in TX7)
  byte A_NU20;                     //       Reserved (not used in TX7)
  byte A_NU21;                     //       Reserved (not used in TX7)
  byte A_NU22;                     //       Reserved (not used in TX7)
  byte A_NU23;                     //       Reserved (not used in TX7)
  byte A_NU24;                     //       Reserved (not used in TX7)
  byte A_NU25;                     //       Reserved (not used in TX7)
  byte A_VoiceAttn;                //       0-7  translates to 0-99 for DX7
  byte A_NU27;                     //       Reserved (not used in TX7)
  byte A_NU28;                     //       Reserved (not used in TX7)
  byte A_NU29;                     //       Reserved (not used in TX7)

  byte B_NU0;                      //       Reserved (not used in TX7)
  byte B_NU1;                      //       Reserved (not used in TX7)
  byte B_PolyMono;                 //       0-1
  byte B_PitchBendRange;           //       0-12
  byte B_PitchBendStep;            //       0-12
  byte B_PortamentoTime;           //       0-99
  byte B_PortaGlissando;           //       0-1
  byte B_PortamentoMode;           //       0-1
[... 554 lines cut ...]
    0x20, 0x4f, 0x52, 0x47, 0x41, 0x4e, 0x2e, 0x20, 0x0d, 0x07, 
    0x00, 0x04, 0x17, 0x2f, 0x18, 0x00, 0x00, 0x00, 0x00, 0x00, 
    0x63, 0x63, 0x1e, 0x0c, 0x3b, 0x07, 0x00, 0x1c, 0x10, 0x20, 
    0x10, 0x00, 0x00, 0x00, 0x00, 0x00, 0x63, 0x63, 0x1f, 0x0c, 
    0x04, 0x27, 0x44, 0x52, 0x4b, 0x2d, 0x20, 0x48, 0x4c, 0x41, 
    0x44, 0x2e, 0x20, 0x20, 0x49, 0x4e, 0x54, 0x52, 0x4f, 0x20, 
    0x2d, 0x20, 0x46, 0x49, 0x52, 0x53, 0x54, 0x2e, 0x20, 0x20, 
    0x20, 0x20,

    // Tail
    0x7f, // Checksum
    0xf7  // End of sysex
};

// Number of bytes in the array above: 4104

// Total size of the sysex dump is thus 4096+6+2 = 4104 bytes
// Header is 6 bytes (start of sysex and manufacturer)
// Tail is 2 bytes (checksum and end of sysex)

// PMEM is 64 bytes
// PCED is 94 bytes

// So how can we parse the sysex dump into PMEM and PCED?
```

BobanSpasic, 2023-11-04:
Performance Bank = Header + 64xPMEM_Dump + Tail
Single Performance= Header + 1xPCED + Tail

probonopd, 2023-11-04:
So far I have figured out that the `example_DX5_PerfData` is sysex header + 64xPMEM + sysex tail.

probonopd, 2023-11-04:
Using this Python script

```
with open("/tmp/user/_home_user_Downloads_DX5 Carts.zip/DX5 Performance.syx", "rb") as file:
    data = file.read()

# Remove the first 6 bytes (the header) from the data, this is the sysex header
data = data[6:]

# Put the data into a list of 64-byte PMEMs, but only the first 64 ones
pmems = [data[i:i+64] for i in range(0, len(data), 64)][:64]

pmem_params = {
    "A_PolyMono": "Poly/Mono for voice A",
    "A_PBSlo_PBR": "Pitch Bend Slope and Pitch Bend Range for voice A",
    "A_PortaTime": "Portamento Time for voice A",
    "A_PM_PGL": "Pitch Modulation and Pitch Glide for voice A",
    "A_MWA_MWS": "Mod Wheel Assign and Mod Wheel Sensitivity for voice A",
    "A_FCA_FCS": "Foot Control Assign and Foot Control Sensitivity for voice A",
    "A_ATA_ATS": "Aftertouch Assign and Aftertouch Sensitivity for voice A",
    "A_BCA_BCS": "Breath Control Assign and Breath Control Sensitivity for voice A",
    "A_NU08": "Unknown",
    "A_NU09": "Unknown",
    "A_NU10": "Unknown",
    "A_NU11": "Unknown",
    "A_NU12": "Unknown",
    "A_NU13": "Unknown",
    "A_ATN": "Aftertouch Noise for voice A",
    "A_PBShi": "Pitch Bend Shift for voice A",
    "B_PolyMono": "Poly/Mono for voice B",
    "B_PBSlo_PBR": "Pitch Bend Slope and Pitch Bend Range for voice B",
    "B_PortaTime": "Portamento Time for voice B",
    "B_PM_PGL": "Pitch Modulation and Pitch Glide for voice B",
    "B_MWA_MWS": "Mod Wheel Assign and Mod Wheel Sensitivity for voice B",
    "B_FCA_FCS": "Foot Control Assign and Foot Control Sensitivity for voice B",
    "B_ATA_ATS": "Aftertouch Assign and Aftertouch Sensitivity for voice B",
    "B_BCA_BCS": "Breath Control Assign and Breath Control Sensitivity for voice B",
    "B_NU08": "Unknown",
    "B_NU09": "Unknown",
    "B_NU10": "Unknown",
    "B_NU11": "Unknown",
    "B_NU12": "Unknown",
    "B_NU13": "Unknown",
    "B_ATN": "Aftertouch Noise for voice B",
    "B_PBShi": "Pitch Bend Shift for voice B",
    "G_VMS_KMOD": "Voice Mode and Keyboard Mode for voice group G",
    "G_NU33": "Unknown"
}

i = 0
for pmem in pmems:
    i += 1
    print(f"PMEM {i}")
    print("Name: %s" % pmem[-30:].decode("ascii"))

    # Print all parameters in 0x00 format,
    j = 0
    for param in pmem_params.keys():
        print(f"{pmem_params[param]}: 0x{pmem[j]:02x}")
        j += 1
    print("")

# Remove the first 64*64 bytes
data = data[64*64:]

# Print the number of bytes remaining
print(f"Bytes remaining: {len(data)}")

# Print the remaining bytes in 0x00 format
print("Remaining bytes:")
for byte in data:
    print(f"0x{byte:02x} ", end="")
print(" - this is the MIDI sysex checksum and end-of-transmission byte.")
```

the data is parsed to

```
PMEM 1
Name: TOM AND DAVE - INTRO FADEBELLS
Poly/Mono for voice A: 0x0a
Pitch Bend Slope and Pitch Bend Range for voice A: 0x02
Portamento Time for voice A: 0x00
Pitch Modulation and Pitch Glide for voice A: 0x06
Mod Wheel Assign and Mod Wheel Sensitivity for voice A: 0x19
Foot Control Assign and Foot Control Sensitivity for voice A: 0x70
Aftertouch Assign and Aftertouch Sensitivity for voice A: 0x2a
Breath Control Assign and Breath Control Sensitivity for voice A: 0x70
Unknown: 0x00
Unknown: 0x00
Unknown: 0x00
Unknown: 0x00
Unknown: 0x63
Unknown: 0x63
Aftertouch Noise for voice A: 0x1d
Pitch Bend Shift for voice A: 0x0c
Poly/Mono for voice B: 0x2a
Pitch Bend Slope and Pitch Bend Range for voice B: 0x02
Portamento Time for voice B: 0x00
Pitch Modulation and Pitch Glide for voice B: 0x06
Mod Wheel Assign and Mod Wheel Sensitivity for voice B: 0x1a
Foot Control Assign and Foot Control Sensitivity for voice B: 0x70
Aftertouch Assign and Aftertouch Sensitivity for voice B: 0x1b
Breath Control Assign and Breath Control Sensitivity for voice B: 0x70
Unknown: 0x00
Unknown: 0x00
Unknown: 0x00
Unknown: 0x00
Unknown: 0x63
Unknown: 0x63
Aftertouch Noise for voice B: 0x5d
Pitch Bend Shift for voice B: 0x0c
Voice Mode and Keyboard Mode for voice group G: 0x2d
Unknown: 0x7e

(...)

PMEM 64
Name: DRK- HLAD.  INTRO - FIRST.    
Poly/Mono for voice A: 0x0d
Pitch Bend Slope and Pitch Bend Range for voice A: 0x07
Portamento Time for voice A: 0x00
Pitch Modulation and Pitch Glide for voice A: 0x04
Mod Wheel Assign and Mod Wheel Sensitivity for voice A: 0x17
Foot Control Assign and Foot Control Sensitivity for voice A: 0x2f
Aftertouch Assign and Aftertouch Sensitivity for voice A: 0x18
Breath Control Assign and Breath Control Sensitivity for voice A: 0x00
Unknown: 0x00
Unknown: 0x00
Unknown: 0x00
Unknown: 0x00
Unknown: 0x63
Unknown: 0x63
Aftertouch Noise for voice A: 0x1e
Pitch Bend Shift for voice A: 0x0c
Poly/Mono for voice B: 0x3b
Pitch Bend Slope and Pitch Bend Range for voice B: 0x07
Portamento Time for voice B: 0x00
Pitch Modulation and Pitch Glide for voice B: 0x1c
Mod Wheel Assign and Mod Wheel Sensitivity for voice B: 0x10
Foot Control Assign and Foot Control Sensitivity for voice B: 0x20
Aftertouch Assign and Aftertouch Sensitivity for voice B: 0x10
Breath Control Assign and Breath Control Sensitivity for voice B: 0x00
Unknown: 0x00
Unknown: 0x00
Unknown: 0x00
Unknown: 0x00
Unknown: 0x63
Unknown: 0x63
Aftertouch Noise for voice B: 0x1f
Pitch Bend Shift for voice B: 0x0c
Voice Mode and Keyboard Mode for voice group G: 0x04
Unknown: 0x27

Bytes remaining: 2
Remaining bytes:
0x7f 0xf7  - this is the MIDI sysex checksum and end-of-transmission byte.
```

This is puzzling me. I was expecting to see, for each performance, which combination of voices is being used, which detune is used, etc.

BobanSpasic, 2023-11-04:
I don't know how DX5 is working, but for TX7 is so - selecting perormance 01 will take A1 for first, and B1 for second voice.

probonopd, 2023-11-04:
> I don't know how DX5 is working, but for TX7 is so - selecting perormance 01 will take A1 for first, and B1 for second voice. P2 = A2+B2 etc

I can confirm that this seems to be the scheme for the DX5 too:

* 1: `DX5A1.SYX` `HORN SEC.A` + `DX5B1.SYX` `HORN SEC.B`
* 2: `DX5A1.SYX` `TRUMPET  A` + `DX5B1.SYX` `TRUMPET  B`
* 3: `DX5A1.SYX` `SYN BRS 1A` + `DX5B1.SYX` `SYN BRS 1B`
* 4: `DX5A1.SYX` `TIGHT BR.A` + `DX5B1.SYX` `TIGHT BR.B`
* 5: `DX5A1.SYX` `SYN BRS 2A` + `DX5B1.SYX` `SYN BRS 2B`
* 6: `DX5A1.SYX` `SYN BRS 3A` + `DX5B1.SYX` `SYN BRS 3B`
* 7: `DX5A1.SYX` `CS80 BRS A` + `DX5B1.SYX` `CS80 BRS B`
* 8: `DX5A1.SYX` `BRIGHT ST.` + `DX5B1.SYX` `BRASS     `
* 9: `DX5A1.SYX` `CELLOS  1A` + `DX5B1.SYX` `CELLOS   B`
* 10: `DX5A1.SYX` `VIOLINS  A` + `DX5B1.SYX` `VIOLINS 1B`
* 11: `DX5A1.SYX` `ST.ENS. 1A` + `DX5B1.SYX` `SOLOVIOLIN`
* 12: `DX5A1.SYX` `MID.STRG A` + `DX5B1.SYX` `MID.STRG B`
* 13: `DX5A1.SYX` `AN.STRG 1A` + `DX5B1.SYX` `AN.STRG 1B`
* 14: `DX5A1.SYX` `CELLOS  2A` + `DX5B1.SYX` `VIOLINS 2B`
* 15: `DX5A1.SYX` `ST.ENS. 2A` + `DX5B1.SYX` `ST.ENS.  B`
* 16: `DX5A1.SYX` `STRINGS  A` + `DX5B1.SYX` `TRUMPET  B`
* 17: `DX5A1.SYX` `PIANO   1A` + `DX5B1.SYX` `PIANO   1B`
* 18: `DX5A1.SYX` `PIANO   2A` + `DX5B1.SYX` `PIANO   2B`
* 19: `DX5A1.SYX` `ELEC GRD A` + `DX5B1.SYX` `ELEC GRD B`
* 20: `DX5A1.SYX` `E.PIANO 1A` + `DX5B1.SYX` `E.PIANO 1B`
* 21: `DX5A1.SYX` `E.PIANO 2A` + `DX5B1.SYX` `E.PIANO 2B`
* 22: `DX5A1.SYX` `E.PIANO 3A` + `DX5B1.SYX` `E.PIANO 3B`
* 23: `DX5A1.SYX` `CLAV.    A` + `DX5B1.SYX` `CLAV.   1B`
* 24: `DX5A1.SYX` `HARPSI. 1A` + `DX5B1.SYX` `HARPSI.  B`
* 25: `DX5A1.SYX` `PIPES    A` + `DX5B1.SYX` `PIPES    B`
* 26: `DX5A1.SYX` `E.ORGAN 1A` + `DX5B1.SYX` `E.ORGAN 1B`
* 27: `DX5A1.SYX` `E.ORGAN 2A` + `DX5B1.SYX` `E.ORGAN 2B`
* 28: `DX5A1.SYX` `E.PIANO 4A` + `DX5B1.SYX` `JAZZ GUITR`
* 29: `DX5A1.SYX` `ELEC.BASS ` + `DX5B1.SYX` `E.PIANO 4B`
* 30: `DX5A1.SYX` `DBL.HARP A` + `DX5B1.SYX` `DBL.HARP B`
* 31: `DX5A1.SYX` `A.MALLET A` + `DX5B1.SYX` `A.MALLET B`
* 32: `DX5A1.SYX` `VIBES    A` + `DX5B1.SYX` `VIBES    B`
* 33: `DX5A2.SYX` `E.P.& BR A` + `DX5B2.SYX` `E.P.& BR B`
* 34: `DX5A2.SYX` `E.GRD&BR A` + `DX5B2.SYX` `E.GRD&BR B`
* 35: `DX5A2.SYX` `E.PIANO 5A` + `DX5B2.SYX` `SAX [BC1] `
* 36: `DX5A2.SYX` `E.PIANO 6A` + `DX5B2.SYX` `CLAV.   2B`
* 37: `DX5A2.SYX` `E.PIANO 7A` + `DX5B2.SYX` `STRINGS 1B`
* 38: `DX5A2.SYX` `HARPSI. 2A` + `DX5B2.SYX` `STRINGS 2B`
* 39: `DX5A2.SYX` `ORCHESTRA ` + `DX5B2.SYX` `ORCH.CHIME`
* 40: `DX5A2.SYX` `FRETLES 1A` + `DX5B2.SYX` `R.CYMBAL  `
* 41: `DX5A2.SYX` `KICK DRUM ` + `DX5B2.SYX` `SNARE     `
* 42: `DX5A2.SYX` `CL.HI-HAT ` + `DX5B2.SYX` `CYMBAL    `
* 43: `DX5A2.SYX` `HAND CLAPS` + `DX5B2.SYX` `TOM TOMS  `
* 44: `DX5A2.SYX` `LOG DRUMS ` + `DX5B2.SYX` `ROTO TOMS `
* 45: `DX5A2.SYX` `TAMBOURINE` + `DX5B2.SYX` `TIMBALI   `
* 46: `DX5A2.SYX` `COW BELL  ` + `DX5B2.SYX` `WOODBLOCK `
* 47: `DX5A2.SYX` `FRETLES 2A` + `DX5B2.SYX` `SAX[BC1]  `
* 48: `DX5A2.SYX` `PIANO   1A` + `DX5B2.SYX` `FLUTE     `
* 49: `DX5A2.SYX` `SYN-RISE A` + `DX5B2.SYX` `SYN-RISE B`
* 50: `DX5A2.SYX` `SAMPLE&HLD` + `DX5B2.SYX` `LEAD LINE `
* 51: `DX5A2.SYX` `POLY SYNTH` + `DX5B2.SYX` `LEAD SYNTH`
* 52: `DX5A2.SYX` `PERC.SYN.A` + `DX5B2.SYX` `PERC.SYN.B`
* 53: `DX5A2.SYX` `MUSIC BX A` + `DX5B2.SYX` `MUSIC BX B`
* 54: `DX5A2.SYX` `FM ENS.  A` + `DX5B2.SYX` `FM ENS.  B`
* 55: `DX5A2.SYX` `PLAN.ICE A` + `DX5B2.SYX` `PLAN.ICE B`
* 56: `DX5A2.SYX` `F. CHOIR  ` + `DX5B2.SYX` `M. CHOIR  `
* 57: `DX5A2.SYX` `BIG BEN   ` + `DX5B2.SYX` `TUNED BELL`
* 58: `DX5A2.SYX` `GLASS WC A` + `DX5B2.SYX` `GLASS WC B`
* 59: `DX5A2.SYX` `GROWL     ` + `DX5B2.SYX` `BIRDS     `
* 60: `DX5A2.SYX` `TWO FOUR  ` + `DX5B2.SYX` `ONE THREE `
* 61: `DX5A2.SYX` `TRAFFIC  A` + `DX5B2.SYX` `TRAFFIC  B`
* 62: `DX5A2.SYX` `FL.CLOUD A` + `DX5B2.SYX` `FL.CLOUD B`
* 63: `DX5A2.SYX` `EXPLOSION ` + `DX5B2.SYX` `MACHINEGUN`
* 64: `DX5A2.SYX` `BOMBDROP A` + `DX5B2.SYX` `BOMBDROP B`

```
[... 35 lines cut ...]
basename_a = filename_a.split("/")[-1]

filename_b = "/tmp/user/_home_user_Downloads_DX5 Carts.zip/DX5B2.SYX"
with open(filename_b, "rb") as file:
    data_b = file.read()
basename_b = filename_b.split("/")[-1]

# Remove the first 6 bytes (the header) from the data, this is the sysex header
data_a = data_a[6:]
data_b = data_b[6:]

# Put the data into a list of 128-byte ones, but only the first 32 ones
voices_a = [data_a[i:i+128] for i in range(0, 128*32, 128)]
voices_b = [data_b[i:i+128] for i in range(0, 128*32, 128)]
voicenames_a = []
voicenames_b = []
i = 0
for voice_a in voices_a:
    voicenames_a.append(voice_a[-10:].decode("ascii"))
for voice_b in voices_b:
    voicenames_b.append(voice_b[-10:].decode("ascii"))

for i in range(0, 32):
    print("* %i: `%s` `%s` + `%s` `%s`" % (i+33, basename_a, voicenames_a[i], basename_b, voicenames_b[i]))
```