probonopd/MiniDexed #28: [Idea] Control Voltage input

probonopd, 2022-02-23:
[Over at Reddit](https://www.reddit.com/r/synthdiy/comments/syup5a/minidexed_dx7_tx816tx802_on_baremetal_raspberry_pi/), someone suggested adding CV control:

> I see the potential for a killer eurorack module !
> 
> Even with a minimal implementation, like CV for pitch and algorithm I see a very interesting thing indeed.

Unlike other minicontrollers or MCUs nowadays, Raspberry Pi does not have a built-in Analog-to-Digital Converter (ADC) Circuit.

So the easiest solution might be to built a CV-to-USB-MIDI (e.g., using a Pro Micro ATmega32U4) or CV-to-3.3V-Serial-MIDI (e.g., using a Nano) circuit using an Arduino and feed the MIDI into the Raspberry Pi.

For this, we would need MiniDexed to listen to more than one MIDI input (USB and Serial and/or 2 USB) at the same time. Is this possible @rsta2?

probonopd, 2022-04-05:
Deprioritized until someone comes around who has a big enough need and wants to implement it.

probonopd, 2022-04-22:
Looking at https://github.com/craigyjp/16bit-MIDI-to-CV-polyphonic-converter it seems lke you know a few things about CV @craigyjp. What do you think about this idea? (Personally I am not interested in CV since I have zero devices that support it, but I know that it'd be interesting for many MiniDexed users.)

craigyjp, 2022-04-22:
A couple of assignable CV inputs might be useful, things like cutoff, resonance, amplitude or LFO speed are quite useful. FM is quite complex unlike subtractive synthesis, so a small change can have a big impact.

probonopd, 2022-04-22:
So a USB device that would convert the CVs to MIDI CCs and/or sysex would probably the most flexible solution? Could be done e.g, on a Raspberry Pi Pico running Python for maximum hackability.

craigyjp, 2022-04-22:
I don't know much about the pi, does it not have ADC inputs?

luisfcorreia, 2022-04-22:
Here's my humble opinion:
the DX7 is a digital synth, it makes little to no sense (for me) to have anything analog controlling it.

Having said that, no Raspberry Pi has Analog Inputs, you would need a secondary microcontroller, like a Raspberry PICO ou any Arduino, extra hardware for allow for standard CV voltages (Eurorack style)
And even then, without extra hardware, you would still be limited to 6 or so inputs.

(too much of a hassle if you ask me)

probonopd, 2022-04-22:
Personally I tend to agree but I don't pretend to know anything about how CV works and what its limitations are.

luisfcorreia, 2022-04-22:
I do know how CV works, for this specific application, it's a bit pointless (but doable)

probonopd, 2022-04-22:
OK, then let's not bother until someone _really_ needs this (and does it) ;-)

craigyjp, 2022-04-22:
Yes, but it was a specific case for integrating into Eurorack. These guys are mental when it comes to tweaking, but you could easily convert CV to midi for most functions. I can't see the point of altering voice parameters in an FM synth on the fly with CV.

luisfcorreia, 2022-04-22:
yes @craigyjp it is a LOT easier to just grab a CV-to-MIDI Eurorack module and connect it to the Raspberry then to try and do it "by ourselves"

diyelectromusic, 2022-08-23:
I think this one should be closed - I agree with the comments above, I don't think it makes sense to try to add CV inputs into MiniDexed.  If someone really wanted it then they could create an add-on board with a microcontroller and get it to generate MIDI but it would be a lot of work for (in my opinion) a sub-optimal result...

And no-one has touched on the monophonic vs polyphonic side of things either...

probonopd, 2022-08-25:
Agree. Closing here, feel free to comment here if you think this ticket should be reopened, or if you would like to share your microcontroller based solution for this sort of task. Thanks!

closed 2022-08-25 by probonopd

zsoltmar, 2022-12-14:
I would 100% be interested in a solution for this! And no, having analog controls for a digital synth makes total sense and it's not pointless at all. There are countless eurorack modules that do this. Mutable Instruments Plaits actually has the engine for DX7 roms so it can play dexed patches - but it would be amazing to have an 8 voice DX7 like minidexed run with CV! I'm playing around with the idea... but would be great to see more inputs from people who are more knowledgeable.

sou7611, 2022-12-14:
My 2 cents on this : CV->Midi adapters exist for a reason

probonopd, 2022-12-15:
Yes, a CV->Midi adapter should do the trick. This is outside the scope of this project.

probonopd, 2023-09-01:
Check this out:

https://github.com/probonopd/MiniDexed/discussions/524#discussioncomment-6881363