probonopd/MiniDexed #33: Optimize CHUNK_SIZE [ _HDMI ] parameter

probonopd, 2022-02-27:
@rsta2 [wrote](https://github.com/probonopd/MiniDexed/issues/12#issuecomment-1051853664)

> The CHUNK_SIZE [ _HDMI ] parameter should be tuned and decreased for smaller latency. Maybe it should be configurable for the user later

Can I help this tuning by testing on various Raspberry Pi models? I have Zero, 1, 2, 3, 4.
How exactly would one tune/test this?

rsta2, 2022-02-28:
Yes, please do. You can just set the chunk size [here](https://github.com/probonopd/MiniDexed/blob/main/src/minidexed.h#L24) and try, if the audio signal is clean with it (no drops). I would try 256 first for `CHUNK_SIZE` and 384 for `CHUNK_SIZE_HDMI`. If there are drops then increase to 512, 1024 and (384x2), (384x4) for HDMI. The `CHUNK_SIZE` must be a multiple of 2, for `CHUNK_SIZE_HDMI` a multiple of 384.

probonopd, 2022-02-28:
These are the smallest values that work for me. Luckily they work across all models tested.

|Model|`CHUNK_SIZE`|`CHUNK_SIZE_HDMI`|
|---|---|---|
|Zero|no headphone jack|384 * 5 (seems to work but seeing `GetChunk:... 524%`)|
|1|2 * 64|384 * 5 (seems to work but seeing `GetChunk:... 524%`)|
|2| |384 * 5 (seems to work but seeing `GetChunk:... 174%`)|
|3|2 * 64|384 * 5 (seems to work but seeing `GetChunk:... 151%`)|
|4|2 * 64|384 * 5|

I never got bad audio, but `usbdev: function is not supported` or `xhcidev: Cannot set device address` when further lowering the values.

probonopd, 2022-02-28:
Possibly this needs to be retested using different voices.

It seems like some voices are fine (e.g., BRASS 1) while others are "destroyed" now (e.g., BRASS 2). Maybe depending on which Alogrithm a voice uses the outcome is different?

rsta2, 2022-02-28:
These values are interesting. Yes, this will probably depend on the used voice and maybe can change with modifications in MiniDexed. So I wouldn't use the absolute minimum to have some margin. Say 256 and 384\*6 for HDMI could be reasonable. Yes, the percent value for HDMI is not valid yet. This is known.

closed 2022-03-13 by probonopd