probonopd/MiniDexed #35: Support more than one MIDI device

probonopd, 2022-02-28:
Currently only one MIDI USB device is being used. `minidexed.cpp` has

```
(CUSBMIDIDevice *) CDeviceNameService::Get ()->GetDevice ("umidi1", FALSE);
```

What would we need to do if we wanted to support more than one USB MIDI device?
Use cases might include
* USB MIDI foot controllers
* USB MIDI sequencers
* Multiple keyboards
etc.

rsta2, 2022-02-28:
Yes, that should be possible, at least for the RPi 4. Because of the USB problems on the RPi 1-3 we had, I'm not sure there. But the whole MIDI input code needs some rework. This enhancement could be implemented together with this. I noticed several problems:

* The serial MIDI handler may be interrupted, when in `keyup()` or `keydown()` by `GetChunk()`. Needs to be synchronized.

* Program change and bank select does not work with serial MIDI.

* MIDI receive must be decoupled from processing, because with the new LCD output, there are programmed delays executed in an IRQ handler in `MIDIPacketHandler()`. This is normally not allowed.

Perhaps we should do some cleanup on the code now. We added some things and I think, the code is not looking that great at the moment. Especially `minidexed.*` is not very clear. If you agree, I would deal with this and add the additional MIDI device support together with this.

@probonopd I also have a question: What are your plans to add a license to MiniDexed (GPLv3?). Strictly speaking nobody is allowed to use MiniDexed, when it does not have a license. When we work together on MiniDexed, of course it's important for every contributor to know the conditions, under which the program and the code can be used. It must be open source. It would also be good to have a template for the file headers.

probonopd, 2022-03-01:
Well, since rsta2/circle and asb2m10/dexed are licensed under the GNU General Public License v3.0 that is our only option, or isn't it? Maybe also the Apache Software License, version 2.0] as the FSF states it as [compatible](https://gplv3.fsf.org/wiki/index.php/Compatible_licenses#GPLv3-compatible_licenses).

probonopd, 2022-03-01:
> Perhaps we should do some cleanup on the code now. We added some things and I think, the code is not looking that great at the moment. Especially minidexed.* is not very clear. If you agree, I would deal with this and add the additional MIDI device support together with this.

Definitely agree, but unfortunately I won't be of much help there. It would be great if you could have a go at restructuring.

rsta2, 2022-03-01:
> Well, since rsta2/circle and asb2m10/dexed are licensed under the GNU General Public License v3.0, this license is our only option, or isn't it? Maybe also the Apache Software License, version 2.0 as the FSF states it as [compatible](https://gplv3.fsf.org/wiki/index.php/Compatible_licenses#GPLv3-compatible_licenses).

Good, I have the same opinion.

> Definitely agree, but unfortunately I won't be of much help there. It would be great if you could have a go at restructuring.

No problem, I will start to work on this.

probonopd, 2022-03-06:
Closed by https://github.com/probonopd/MiniDexed/pull/37:

> Support multiple USB MIDI inputs at once (two on Raspberry Pi 1-3, four on Raspberry Pi 4).

Thanks @rsta2

closed 2022-03-06 by probonopd

probonopd, 2022-03-13:
I have successfully tested using 2 USB MIDI devices at the same time:
* A Raspberry Pi Zero running CircuitPython and a script t that plays MIDI notes
* A real MIDI keyboard
Both are playing nicely together.