probonopd/MiniDexed #38: GPIO pinout for external devices

rsta2, 2022-03-05:
As suggested in https://github.com/rsta2/circle/issues/280 I investigated the GPIO pinout of the (already known) external devices, which can be connected to a Raspberry Pi, when using MiniDexed and generated a list as a suggestion (see attached).

Two comments:

* The default setting for the LCD outputs RS and RW has changed. Would need to be updated in *config.cpp* and *minidexed.ini*.

* We should enable the option `USE_PWM_AUDIO_ON_ZERO` in build.sh. This allows to connect earphones or an amplifier to the Raspberry Pi Zero (2) via GPIO12/13 using a small [external circuit](https://learn.adafruit.com/assets/28754).

[pinout.txt](https://github.com/probonopd/MiniDexed/files/8190119/pinout.txt)

probonopd, 2022-03-06:
> We should enable the system option USE_PWM_AUDIO_ON_ZERO in build.sh.

By uncommenting it like below, or is there a better way?

```
sed -i -e 's|//#define USE_PWM_AUDIO_ON_ZERO|#define USE_PWM_AUDIO_ON_ZERO|g' include/circle/sysconfig.h
```

rsta2, 2022-03-06:
It can be defined in *build.sh* in the same way as `REALTIME` and `SAVE_VFP_REGS_ON_IRQ`. With this the option is globally enabled and the file *sysconfig.h* does not need to be manipulated.

closed 2022-03-06 by probonopd

probonopd, 2022-03-06:
Thanks @rsta2, implemented your suggestions. i2c may prove useful indeed as we may be running low on pins.

probonopd, 2022-03-08:
Looking at the pinout in more detail, I have the following questions:
* Many of the required pins (for rotary encoder, DAC, and LCD) are not on the Raspberry Pi 1
* Can we come up with a configuration that sacrifices less than 3 peripherals on the Pi 1? E.g., by moving LCD RW OUT from GPIO16 to GPIO4 we could have the LCD on Pi 1
* It looks like the pins  GPIO28-31 are on the header P5 on the Raspberry Pi 1, which needs to be soldered to
* Do we really need SPI? If not, we could also have the rotary encoder on the Pi 1
* Can Circle drive the display using i2c? If we could do this, we could save a lot of pins and could also have SPI on the Pi 1 (should we ever need it) (not sure ion this one)

I think we should keep the same pinout for all models of Raspberry Pi to keep things simple and because it will allow the same hat/hardware/connector to be used for all models. This means that if we want to continue support for the Raspberry Pi 1, then we should avoid pins that are not physically present on the Raspberry Pi 1.

rsta2, 2022-03-08:
Good point. I think we can define a pinout, which works on any RPi model, especially when we do not need SPI. Answering your questions:

* Actually the LCD RW pin is not required at all. You can wire the RW pin from the LCD fixed to GND. The Circle LCD driver is not reading from the display, so it is not necessary to switch this pin from software. So the LCD can be used on RPi 1 already.
* The GPIO28-31 pins for DACs for old RPi 1 versions cannot be modified. There are DAC HATs out there, which use this configuration.
* If we do not want to connect SPI displays, we do not need SPI. I would use GPIO9,10,11 for the rotary encoder then (for GPIO5,6,26 in this order).
* Currently there is no I2C LCD display driver in Circle. Unfortunately I don't know, if there is a standard for the I2C slave address and for the exchanged data for such I2C displays. I so, I would support both interfaces as an option. This could be done later for I2C.

I agree, if we continue to support the RPi 1, we should have the same pinout there, as it is on the other models. There is one exception: the DAC is at GPIO28-31 on the RPi 1B, that's a standard. But what, if we have to move to multi-core, because of the USB issues? Perhaps we should wait for this decision, before modifying the pinout again?

rsta2, 2022-03-09:
Another idea: If SPI would be an alternative interface for displays, we could easily use the SPI GPIO pins for the (currently supported) HD44780 LCD display, because both are options for the same purpose and only one display is needed.

rsta2, 2022-03-09:
But if we want to have a CV input using a [MCP3004/3008](https://ww1.microchip.com/downloads/en/DeviceDoc/21295d.pdf) ADC, we also need to have the SPI interface.

There are also differences between Rev 1 and Rev 2 PCBs of the Raspberry Pi 1 Model A and B:

https://elinux.org/RPi_Low-level_peripherals#P1_Header