probonopd/MiniDexed #45: Ignore Bank Select and Program Change numbers that are not supported

probonopd, 2022-03-13:
I didn't find this exact information in DX documentation but the "MX49/MX61 Data List" PDF says:

> Bank Select and Program Change numbers that are not supported by Yamaha will be ignored.

@rsta2 you are already doing this for the rotary encoder and it seems like we should be doing the same for MIDI Bank Select and Program Change messages.

rsta2, 2022-03-13:
We do. Program Change messages are allowed for programs 0..31 (1..32 for the user) and Bank Selects from banks 0..127 (1..128 for the user). All other messages will be ignored. Bank Select MSB will be totally ignored at the moment.

probonopd, 2022-03-13:
My bad. Made a mistake in testing. The "Take off" preset had me tricked - it is so silent in the beginning that I thought I was hearing nothing when Iin fact it was working.

closed 2022-03-13 by probonopd

rsta2, 2022-03-13:
No problem.