probonopd/MiniDexed #58: Record midi and/or wav

probonopd, 2022-03-26:
Would it be possible to record midi and/or wav to microSD and/or USB storage, at least on the more powerful models?

rsta2, 2022-03-26:
First, if you want to record a MiniDexed sound demo, there is already a solution for the Raspberry Pi [here](https://github.com/rsta2/circle/tree/master/sample/42-i2sinput). You need a second Raspberry Pi for this, enable I2S mode in MiniDexed, and connect the I2S input of the second RPi to the I2S output of MiniDexed. The recording RPi will provide a monitor audio signal via PWM on the headphone jack. The resulting file(s) on the SD card are in a raw format, and can be imported into the Audacity program for example, to be converted into any other format.

If you want to include this as a feature into MiniDexed, I would say, maybe. Of course any additional actions in MiniDexed produce more load. Writing to the SD card or an USB flash drive has to be done on CPU core 0, where already some things are processed (sound output, USB, serial MIDI, UI, plug-and-play events, screen and LCD output, rotary encoder). In any case it will increase the latency for serial MIDI, and because there are already problems with the USB (lost MIDI events), I would not suggest to do the recording to USB, at least on the RPi 3. I don't think, it will work on RPi 1-2 and Zero.

Recording MIDI should be possible, because it does not produce that much data, to be saved, but standard MIDI files have a specific structure, which is not that easy to create. Implementing this would take time.

probonopd, 2022-04-05:
OK, this is difficult, so let's not go there now. Unless someone has a big need for this.

> First, if you want to record a MiniDexed sound demo, there is already a solution

Thanks, noted! This will come in handy for e.g., a demo video and/or sound files.

Gustavomurta, 2023-07-22:
Some ideas:
[Reading MIDI input from GO](https://stackoverflow.com/questions/55993819/reading-midi-input-from-go)

Tested by me:

`C:\Users\jgust\go\programas\gomidi\MidiReader>midiReader.exe

MidiInWinMM::initialize: no MIDI input devices currently available.

panic: runtime error: index out of range [0] with length 0

goroutine 1 [running]:
main.main()
        C:/Users/jgust/go/programas/gomidi/MidiReader/midiReader.go:28 +0x2ef

C:\Users\jgust\go\programas\gomidi\MidiReader>midiReader.exe
opening MIDI Port MPKmini2 0
NoteOn at channel 0: key: 3E velocity: 29
NoteOff at channel 0: key: 3E

NoteOn at channel 0: key: 3E velocity: 41
NoteOff at channel 0: key: 3E

NoteOn at channel 0: key: 3E velocity: 4D
NoteOff at channel 0: key: 3E`

Gustavomurta, 2023-07-22:
Reading I2S data, with TinyGO:

https://pkg.go.dev/tinygo.org/x/drivers/microphone