probonopd/MiniDexed #62: UI with a HDMI-Touch-Screen?

dcoredump, 2022-03-31:
What about adding (Lager) a UI for a Touchscreen (connected via HDMI/USB), so voices and setup can be configured with?

dcoredump, 2022-04-01:
Sure, maybe you are right... but it would be a really cool thing if I can add only a touch TFT and program sounds... perhaps later :)

> send MIDI sysex to MiniDexed to modify the voice parameters. If we can get MiniDexed to receive those

Yes, this is in my task list. The code is already in MicroDexed. I only have to put it into Synth_Dexed.

> Also, it would be nice to be able to assign/"learn" Midi CCs

That's also on my task list for MicroDexed. I would try to implement it also in Synth_Dexed later...

rsta2, 2022-04-01:
> >  Also, it would be nice to be able to assign/"learn" Midi CCs
>
> That's also on my task list for MicroDexed. I would try to implement it also in Synth_Dexed later...

I did think about implementing the MIDI CC learn function in MiniDexed too already, but I was busy with another function, that's why I did not answer on this yet.

Because this effects the UI menu, where the parameter must be selected, which should be assigned to the MIDI CC, for which a message is received, there arises the question, should this be implemented in MiniDexed or Synth_Dexed? Personally I don't want to push me in front for this (or other) tasks, but I don't want to produce lots of code, which is replaced afterwards. So maybe this could be the point for me to step back a little bit from working for MiniDexed, I've already done much more than I wanted to do at the beginning.

So please let me know, when you need my help and for what, otherwise I will be more quiet now.

probonopd, 2022-04-05:
> Also, it would be nice to be able to assign/"learn" Midi CCs

>> That's also on my task list for MicroDexed. I would try to implement it also in Synth_Dexed later...

>>> Because this effects the UI menu, where the parameter must be selected, which should be assigned to the MIDI CC, for which a message is received, there arises the question, should this be implemented in MiniDexed or Synth_Dexed?

Moving this discussion to https://github.com/probonopd/MiniDexed/issues/63.

As per "UI with a HDMI-Touch-Screen" I'd like to depriorize this for now, until someone comes along who has a big enough need to implement it.

probonopd, 2022-04-05:
> I've already done much more than I wanted to do at the beginning.

@rsta2 what you have done over the last weeks in this project has been truly outstanding. And without the work of @dcoredump and @smuehlst this project wouldn't have come into existence in the first place.

Let me take this opportunity to thank you very, very much. :1st_place_medal: 

Maybe we can grow the community of users and eventually developers over the coming weeks and months.

rsta2, 2022-04-05:
@probonopd You are welcome. When I was young, the DX7 was one of the things, I wanted to have. Honestly I already tried by myself to extend MiniSynth Pi for FM synthesis, but this didn't work well, and I aborted the project. Now with the support of @dcoredump it was possible to use the existing and further developed FM Synth_Dexed engine, which promised more success. He also provided the first MiniDexed version, which produced sound. And with your continuous valuable feedback and guidance, programming was (mostly ;) fun, as it should be. Many thanks once again to @smuehlst, who's Circle-Stdlib project makes it much easier to use Circle in such projects. Now I have something like a DX7 here. That's great.

probonopd, 2022-04-05:
Actually, you have kind of a shrunken TX802 powerhouse :)

dronus, 2022-05-04:
Actually implementing a whole UI from scratch within the constraints of the realtime audio system is less fun...
But how about using an existing UI? For example, add another RPi, running RaspberryPI-OS sporting a browser serving an UI like this one: 
https://pbpraha.github.io/index.html  
- which fits perfectly on a large FullHD touch screen. 

The two Raspis may be connected by the serial RX/TX pins, assumed there is an working ALSA MIDI driver for this serial interface on the RaspberryPI OS side (receiving MIDI on this pins at the MiniDEXED side should be trivial).

Of course this will have the boot latency of the full OS raspi, but will still deliver the sound with tight timings by the MiniDexed machine.

fedemone, 2023-08-04:
Hi, maybe a UI is less difficult than one thinks. I found this very simple python that is creating a grid of Note buttons / keys, plus additional sticky keys:
https://github.com/touchgadget/buttongrid

The main rows will simply play the note between the selected range that is set in json file. I think it can be added HOLD function too, skipping the note_off if hold global flag is on, and ARP, putting a sleep between two note_on and an arp_gate parameter for the sleep time.
Sticky buttons can be used for basic parameters, a slider or knob can pop at pressing sticky button (I think matplotlib can be used such here https://stackoverflow.com/questions/65024118/how-can-i-use-the-value-of-the-slider-widget) and assing to global value, stored in dictionary. Maybe then saved as parameters.json file. Possibly to load a customized set of parameters can be done calling the pygame.midi.set_instrument API.

This way looks not too bad, but I'm just a beginner in this field.

probonopd, 2023-08-04:
It's not something I am interested to do or have. I am looking to use the minimal and least expensive components possible. Also, the DX... and TX... had no touchscreens either.

closed 2023-08-04 by probonopd

RichyT, 2024-12-12:
While a full UI might be a bit heavyweight, it might be nice to optionally replicate the LCD display in the text mode hdmi output.