probonopd/MiniDexed #63: Assign/"learn" Midi CCs to manipulate voice parameters

probonopd, 2022-03-31:
It would be nice if we could assign/"learn" Midi CCs (knobs and sliders on MIDI hardware controllers) to manipulate Dexed voice parameters.

E.g., these guys:

[image]

probonopd, 2022-03-31:
We'd also need a translation of 0-127 to whatever each voice parameter has as an acceptable range:

[image]
[image]

probonopd, 2022-04-03:
Thanks to the awesome work of @rsta2 most of this is implemented. :1st_place_medal: 

Highlight an aspect of the idea __bold__ in the original ticket above. @rsta2 do you think this would be doable?

rsta2, 2022-04-03:
Thanks! :) I will think about, how this can be done.

probonopd, 2022-04-03:
Similarly, it would be nice if we could manipulate voice parameters using MIDI SYSEX sent over serial/USB.

rsta2, 2022-04-03:
OK, one after another. I will keep this in mind.

dcoredump, 2022-04-05:
I would like to implement both inside Synth_Dexed because both is also needed for MicroDexed. But currently I cannot do everything at the same time 😀.

rsta2, 2022-04-05:
OK, no problem. Let me know, if you need support for an interface to the UI (info about the currently selected parameter).

probonopd, 2022-04-05:
Continuing from https://github.com/probonopd/MiniDexed/issues/62:

> Also, it would be nice to be able to assign/"learn" Midi CCs

>> That's also on my task list for MicroDexed. I would try to implement it also in Synth_Dexed later...

>>> Because this effects the UI menu, where the parameter must be selected, which should be assigned to the MIDI CC, for which a message is received, there arises the question, should this be implemented in MiniDexed or Synth_Dexed?

I think the ideal UX would be: You go to the respective parameter in the UI, and then instead of rotating the knob of the rotary encoder, you just wiggle the control you want to assign on the MIDI controller. MiniDexed then knows which CC message to use (from the messages that came from the MIDI controller) and knows which parameter to edit (the one currently on screen). Learning done! Should be written to a file on the SD card in, e.g., `mapping.ini` so as to survive reboots.

If we agree that we want to do it that way, my question is: Can this even be done in Synth_Dexed (or which portion of it)? or does it need to be done in MiniDexed (because only it has the menu structure and SD writing capabilities)?

probonopd, 2022-04-10:
Worth seeing!
https://www.youtube.com/watch?v=GIE3WfrEEX0

probonopd, 2022-04-17:
For example, a Korg nanoKONTROL 2 could be (ab)used to set the level (volume) of each operator with the slider, and the frequency (ratio) with the knob. Something like that.

maks, 2022-04-18:
Very cool project!

I was wondering if you would consider adding support for something like the Akai Fire:
[image]

For not much more than the price of a small midi keyboard controller plus the existing electronics parts list (lcd,encoder, breadboard,etc) you get a 128x64 OLED, rotary+push button encoder, 4 capacitive touch encoders, multitude of buttons most of which are red/yellow or green leds and full RGB leds pads. And all of it is controllable via Midi Sysex (midi implementation already rev eng'd: https://blog.segger.com/decoding-the-akai-fire-part-1/)  so all a user would need to do is plug it into a RPI along with a DAC.

I think this would need to be a bit more "baked in" than just usual DAW-style midi mapping to get the benefit of using the OLED, LED colors feedback etc.

I've already done midi implementations for the Fire in [Dart](https://github.com/maks/dart_fire_midi) and [TS](https://github.com/maks/ml-1/blob/main/src/firemidi.ts) so there are already existing code bases for reference.

I would be happy to collaborate on this if its something you would be interested in doing in MiniDexed.

probonopd, 2022-04-18:
Hello @maks, interesting device. If this ticket gets implemented, knobs and faders could be easily assigned to MiniDexed parameters. Which imho would already be a _huge_ usability improvement over the original DX7.

We don't want to hardcode MiniDexed for certain MIDI controllers. Besides, without having that piece of hardware the effort is futile. So unless we find a solution purely using configuration files and/or scripts (and someone is willing to implement them) it is unlikely that there will be special support for the Akai Fire.

maks, 2022-04-18:
@probonopd yes, very fair point and I think you are right about avoiding hardcoding support for specific midi controllers. I opened a separate issue for scripting support that I realised after I made the comment here would be the better way to approach this so I'll follow up on that issue.

probonopd, 2022-04-19:
The way I imagine this to work is roughly as follows:

We need a global data structure (a map?) that can hold, for each CC number, (the pointer to) a parameter. Optionally (later), this data structure could be loaded from `cc.ini` and it should be possbile to save it there.

Something like this (just for illustration filled with some values; initially it would be empty):

```
cc_map = {{1, DEXED_OP_FREQ_COARSE,},
  {2, DEXED_OP_FREQ_FINE,},
  {3, DEXED_OP_FREQ_FINE,}};
```

Somewhere around

we should listen for incoming CC messages. Whenever a CC message comes in, we would update the entry in `cc_map` for that CC to contain the parameter that is currently being edited.

For example, if while the display is showing

[image]

we turn a knob that results in a MIDI message with CC 127 coming in, then we would update  `cc_map` to contain `{127, DEXED_OP_FREQ_COARSE,}`.

Learning complete!

Somewhere around

https://github.com/probonopd/MiniDexed/blob/d6025f194319fea7a205621a2aa81bd29ca0fd2f/src/mididevice.cpp#L192-L193

we would go though the map to see if it has an entry for the `cc_number` that has been played. If yes:

We would then use [`scaled_value = constrain(incoming_cc_value, param.Minimum, param.Maximum)`](https://www.arduino.cc/reference/en/language/functions/math/constrain/) to convert the incoming 0-127 to the range allowable by the respective parameter. (`#define constrain(amt,low,high) ((amt)<(low)?(low):((amt)>(high)?(high):(amt)))` is the definition used by Arduino for `constrain()`.)

Finally, we would call `SetParameter (parameter, scaled_value);`.

In the example, whenever we turn the knob, we would change __Freq Coarse__.

Does this make sense?

rsta2, 2022-04-19:
As I understand it, @dcoredump wants to implement most of the functionality in Synth_Dexed. I don't want to interfere with that, but already offered my help, e.g. for determining the current parameter in the UI, when a MIDI CC message comes in.

dcoredump, 2022-04-19:
My thought of implementing this was:

I wanted to give the Dexed class methods for processing SYSEX and MIDI-CC. The MIDI-CC methods should internally get a mapping from the CC number to any function (so also for changing sound parameters) and a scaling (e.g. `CC-7` `0..127` -> `Parameter[42]` `95..43` (so also allow a flip of the values)). The target could be any method to change the sound, but also existing CC values.

The biggest problem is that I can't keep up with the fast development of MiniDexed. In this respect: If someone has desire to implement this: Ideally then in Synth_Dexed, otherwise just directly in MiniDexed. I will definitely implement this in Synth_Dexed, but currently I can't say when.

probonopd, 2022-04-19:
If we want to load and save those assignments, wouldn't we need to do at least parts of the logic in MiniDexed anyways?

I know it's a big ask, but I see this one as the last remaining "killer" feature that I'd _love_ to see implemented while @rsta2 [still has some time](https://github.com/probonopd/MiniDexed/issues/89#issuecomment-1102500841).

rsta2, 2022-04-20:
Yes, there is still some time. Accessing the MIDI CC assignments in an `.ini` file can be done similar to `performance.ini`.

probonopd, 2022-06-05:
Imagine this together with the ROLI Seaboard Block:
https://twitter.com/littlescale/status/1532954508015202304

Would allow us to do things like
https://www.youtube.com/watch?v=6SCug5kUsBs

RK11111111111, 2024-01-04:
> Continuing from #62:
>
>
>
>
> I think the ideal UX would be: You go to the respective parameter in the UI, and then instead of rotating the knob of the rotary encoder, you just wiggle the control you [...]

I really like the idea of adding additional sliders.  However, I think this would be extremely voice dependent, as to which parameters would be modulated.  I think the tone generator section of a performance would be a good section to store the midi mappings.  

For example, if you load Dexed into a Daw like Reaper then learn some mappings.  It's pretty difficult to find a parameter that you would want to tweak live.  As compared to other FM VST synths like Exact or FM8.  I think it is because with Dexed, the emphasis is on automated envelopes, where as VSTs use more of the DAW support.

I also like the idea of a ribbon controller, the other issue with mapping midi controllers is that they often times are not precise enough.  For example, if you map a controller 1...127 to a frequency, unless you do something to map those frequencies to musical notes, its going to be out of tune most the time ratios can also suffer from this.  Even, things like delay times or envelope attack times.  How do you go from a scale of fractions of millisecond to seconds in any precise manner.

The point is it will take some thought to come up with units and mappings that make musical sense.  One that is especially important is the amplitude of the modulating operators, those really need to be tuned precisely.