probonopd/MiniDexed #66: Compressor, Reverb on/off

probonopd, 2022-04-03:
Should we have menu entries to switch effects like Compressor, Reverb on/off?

* Effects
  * Compressor
    * On | Off
   * Reverb
      * On | Off
      * Edit Reverb

rsta2, 2022-04-03:
OK, good suggestion. On my TODO list.

probonopd, 2022-04-03:
Let's also set the default value for Diffusion to 20. The source code comment says:

> Diffusion (lower settings will make the reverb tail more "echoey", optimal value 0.65)

When setting the value to 65 in the GUI, then the echo is removed from the reverb, which I prefer a lot. To reproduce, play the default voice in staccato.

rsta2, 2022-04-03:
OK. But the default Diffusion is already 20:

https://github.com/probonopd/MiniDexed/blob/main/src/minidexed.cpp#L122

probonopd, 2022-04-03:
Sorry I mistyped. I meant: Set the default to 65 rather than 20.

rsta2, 2022-04-03:
OK.

rsta2, 2022-04-04:
Implemented by #69.