probonopd/MiniDexed #71: Revisit final sound processing stages

probonopd, 2022-04-05:
Continuation from https://github.com/probonopd/MiniDexed/pull/69#issuecomment-1087892398:

> Ah yes, I tried the compressor in the VLC media player, and it's getting louder there, when it is switched on. As I read it, @dcoredump wants to re-work the whole final signal processing stage (use floats instead of int16_t, pan, mixer, compressor, reverb), which makes absolutely sense to me. At the moment, each TG has its own compressor, and as I said earlier, my pan processing + mixer is not ideal. There is much room for improvement in it.

This ticket is a reminder that we revisit the way we set up and use limiter/anti-clipping, compressor, pan processing, and mixer once @dcoredump has re-worked the final signal processing stage. No hurries, just so that it won't be forgotten.

dcoredump, 2022-04-07:
The signal chain is now completely in `float32_t` (until it gets converted for `m_pSoundDevice->Write()`) but I have some trouble with noise at the reverb tail. I don't have an idea where it is coming from or if it's (perhaps) only when using PWM as output?

I tried to compare it with @probonopd  current version, but this one always gives me an `assert(reverb)` at startup. Can you try to checkout my current state?

probonopd, 2022-04-07:
Thanks @dcoredump.

Yes, the Raspberry Pi headphone jack PWM is very noisy, especially on the older models. I don't have high hopes that we can do much about it. The only device where it is halfway ok is the Raspberry Pi 4 from my experience.

> I tried to compare it with @probonopd current version, but this one always gives me an assert(reverb) at startup.

Are you saying that the build from https://github.com/probonopd/MiniDexed/releases/tag/continuous is crashing on your machine?

probonopd, 2022-04-07:
Can you try to checkout my current state (https://github.com/dcoredump/MiniDexed)?
I have [opened a PR](https://github.com/probonopd/MiniDexed/pull/73), which normally triggers a build, which I then can test. Unfortunately there is a merge conflict; could you please have a look at it? If you can resolve the conflict in your branch, then it _should_ start to build automagically :)

rsta2, 2022-04-07:
> I tried to compare it with @probonopd current version, but this one always gives me an assert(reverb) at startup.

My mistake, sorry. Put [this line](https://github.com/probonopd/MiniDexed/blob/main/src/minidexed.cpp#L116) behind the creation of the `AudioEffectPlateReverb` object, and the assertion will go away.

PR #73 (2022-04-08): Signal chain in float32_t
Closes #71
closed 2022-04-08 by probonopd