probonopd/MiniDexed #75: Voice initialize function

probonopd, 2022-04-08:
Reading 

> FM Theory & Applications: By Musicians for Musicians 
by Dr. John Chowning and David Bristow
ISBN 4636174828

there are lots of example voices that help one to learn FM synthesis. They start from a "blank" patch, by using the "VOICE INIT" button on the DX7. "This function sets up the basic voice data for creating new voices".

Maybe we should have a similar function, e.g., under

TG X -> Edit Voice -> Initialize

It should set all voice parameters to their initialization default values. I am still trying to find out what "the basic voice data for creating new voices" _exactly_ is.

rsta2, 2022-04-08:
"Edit Voice" and "Initialize" does not fit on one LCD line too. Better "TG#" > "Init Voice" or "TG#" > "Edit Voice" > "Init". Everything else should work.

probonopd, 2022-04-08:
"TG#" > "Edit Voice" > "Init" 
makes most sense for me.

rsta2, 2022-04-09:
Yes, but this does not make sense in my opinion, if we do not have a function to save the edited voice to a voice bank .syx file on the SD card?

probonopd, 2022-04-09:
Indeed I could imagine that sooner or later that feature might be requested by someone.

rsta2, 2022-04-09:
I will have a look at the "Init" function, when I have some break from searching for the USB issue cause.

probonopd, 2022-04-11:
Agree. We should use 2. for the purpose.

And we can remove 1. because we don't need ROM1A hardcoded anymore.

But if the user has no `.syx` files loaded at all, then there should be at least one good-sounding default voice. Would that be 3.?

rsta2, 2022-04-11:
Yes, 2. would be used for the "Init" function in the UI and 3. for all voices in all voice banks, when there isn't any loaded voice bank available. Class `CSysExFileLoader` can be easily changed for this purpose.

probonopd, 2022-04-17:
As a workaround, I found that e.g., http://dxsysex.com/SYSEX_DX7/Guest/ENO.syx contains the init voice on the "empty" slots of the bank.

PR #109 (2022-04-18): Use INIT VOICE as default
Closes #75
commit fe676e94 (2022-04-18): Use INIT VOICE as default (#109)

Closes #75
closed 2022-04-18 by probonopd