probonopd/MiniDexed #76: Mixer object, e.g. for adding reverb send to each TG

dcoredump, 2022-04-08:
Just a placeholder: I am currently creating a mixer object, so a reverb send Mixer (or other sends) can easily be created.

probonopd, 2022-04-08:
> reverb send Mixer (or other sends)

Wow, now we are really getting fancy. I first need to learn what this is all about :-)

dcoredump, 2022-04-09:
> I first need to learn what this is all about :-)

:)

We use the reverb as a master reverb (global). Normaly you decide on a mixer how much amount of each voice is to send to the reverb. Mostly the bass does not get any reverb added and perhaps you also won't add reverb to sounds which have a long decay (strings).

With a mixer you can add a reverb send to each TG:

- TG1: Bass -> 0
- TG2: Strings -> 20
- TG3: PercOrgan -> 75
- TG7: HandClap -> 100

probonopd, 2022-04-09:
I see. Is the result audibly different from having one reverb per TG, or is this a measure to save CPU cycles?

dcoredump, 2022-04-09:
> I see. Is the result audibly different from having one reverb per TG, or is this a measure to save CPU cycles by only needing one reverb?

It's more a question of how you hear sounds. A seperate reverb for each TG would make only sin if you have different reverb settings. But normally we don't hear different types at the same time, so your hearing system might get confused if there are several  reverb types mixed together. Sometimes this might be used as a special effect, but normally you habe only one reverb and you decide how much of each sound part is fed into the reverb.

probonopd, 2022-04-09:
Maybe I am oversimplifying it, but I see reverb as a factor of a simulated "room" in which all instruments are placed. So it makes sense intuitively for me to have one reverb for the whole performance, rather than one per instrument. But I get it that for different instruments (frequences really?) the reverb should be more or less.

rsta2, 2022-04-09:
Good to have a real musician in the team!

dcoredump, 2022-04-09:
> Good to have a real musician in the team!

Please specifiy "real musician"... :) I am more a long term hobby musician.

... and I am a poor programmer. @rsta2 @probonopd: Can you help me with a C++ problem? I have stored the implementation of the current mixer class at [https://github.com/dcoredump/MiniDexed](https://github.com/dcoredump/MiniDexed) I get errors from the linker that methods are not available:
```
  LD    kernel8-rpi4.elf
aarch64-none-elf-ld: minidexed.o: in function `CMiniDexed::CMiniDexed(CConfig*, CInterruptSystem*, CGPIOManager*, CI2CMaster*, FATFS*)':
/home/wirtz/src/MiniDexed-HW/src/minidexed.cpp:118: undefined reference to `AudioStereoMixer<8>::AudioStereoMixer(unsigned short)'
aarch64-none-elf-ld: minidexed.o: in function `CMiniDexed::ProcessSound()':
/home/wirtz/src/MiniDexed-HW/src/minidexed.cpp:764: undefined reference to `AudioStereoMixer<8>::pan(unsigned char, float)'
aarch64-none-elf-ld: /home/wirtz/src/MiniDexed-HW/src/minidexed.cpp:765: undefined reference to `AudioStereoMixer<8>::doAddMix(unsigned char, float*)'
aarch64-none-elf-ld: /home/wirtz/src/MiniDexed-HW/src/minidexed.cpp:779: undefined reference to `AudioStereoMixer<8>::getMix(float*, float*)'
make: *** [../circle-stdlib/libs/circle/Rules.mk:158: kernel8-rpi4.img] Fehler 1
```
But they are defined. No idea where exactly the problem is...

And one more request: Where can I read something about the used naming scheme for the variable names? I would like to change my naming scheme to make the code more readable.

TIA

rsta2, 2022-04-09:
> Please specifiy "real musician"... :) I am more a long term hobby musician.

Someone, who plays in a band and knows, how to setup a room for a concert (e.g. reverb settings).

Can it be, that you have to add the file *mixer.cpp* to the Makefile to satisfy the linker?

I'm afraid, I have to write something down about the naming scheme. It's something like the [Hungarian notation](https://en.wikipedia.org/wiki/Hungarian_notation), but on a minimal basis.

dcoredump, 2022-04-09:
> Can it be, that you have to add the file _mixer.cpp_ to the Makefile to satisfy the linker?

Hm, it's mentioned in Synth_Dexed.mk. Here is the complete output of the run. You can see that effect_mixer.cpp is used and there is an effect_mixer.o. I have no idea why it isn't used for linking...

```
src@buildvm:~/src/MiniDexed-HW/src$ make
  CPP   main.o
  CPP   kernel.o
  CPP   minidexed.o
  CPP   config.o
  CPP   userinterface.o
  CPP   uimenu.o
  CPP   mididevice.o
  CPP   midikeyboard.o
  CPP   serialmididevice.o
  CPP   pckeyboard.o
  CPP   sysexfileloader.o
  CPP   performanceconfig.o
  CPP   perftimer.o
  CPP   effect_mixer.o
  CPP   effect_compressor.o
  CPP   effect_platervbstereo.o
  CPP   ../Synth_Dexed/src/PluginFx.o
  CPP   ../Synth_Dexed/src/dexed.o
  CPP   ../Synth_Dexed/src/dx7note.o
  CPP   ../Synth_Dexed/src/env.o
  CPP   ../Synth_Dexed/src/exp2.o
  CPP   ../Synth_Dexed/src/fm_core.o
  CPP   ../Synth_Dexed/src/fm_op_kernel.o
  CPP   ../Synth_Dexed/src/freqlut.o
  CPP   ../Synth_Dexed/src/lfo.o
  CPP   ../Synth_Dexed/src/pitchenv.o
  CPP   ../Synth_Dexed/src/porta.o
  CPP   ../Synth_Dexed/src/sin.o
  CC    ../CMSIS_5/CMSIS/DSP/Source/SupportFunctions/SupportFunctions.o
  CC    ../CMSIS_5/CMSIS/DSP/Source/BasicMathFunctions/BasicMathFunctions.o
  CC    ../CMSIS_5/CMSIS/DSP/Source/FastMathFunctions/FastMathFunctions.o
  CC    ../CMSIS_5/CMSIS/DSP/Source/FilteringFunctions/FilteringFunctions.o
  CC    ../CMSIS_5/CMSIS/DSP/Source/CommonTables/CommonTables.o
  CC    ../CMSIS_5/CMSIS/DSP/ComputeLibrary/Source/arm_cl_tables.o
  LD    kernel8-rpi4.elf
aarch64-none-elf-ld: minidexed.o: in function `CMiniDexed::CMiniDexed(CConfig*, CInterruptSystem*, CGPIOManager*, CI2CMaster*, FATFS*)':
/home/wirtz/src/MiniDexed-HW/src/minidexed.cpp:118: undefined reference to `AudioStereoMixer<8>::AudioStereoMixer(unsigned short)'
aarch64-none-elf-ld: minidexed.o: in function `CMiniDexed::ProcessSound()':
/home/wirtz/src/MiniDexed-HW/src/minidexed.cpp:764: undefined reference to `AudioStereoMixer<8>::pan(unsigned char, float)'
aarch64-none-elf-ld: /home/wirtz/src/MiniDexed-HW/src/minidexed.cpp:765: undefined reference to `AudioStereoMixer<8>::doAddMix(unsigned char, float*)'
aarch64-none-elf-ld: /home/wirtz/src/MiniDexed-HW/src/minidexed.cpp:779: undefined reference to `AudioStereoMixer<8>::getMix(float*, float*)'
make: *** [../circle-stdlib/libs/circle/Rules.mk:158: kernel8-rpi4.img] Fehler 1
```

rsta2, 2022-04-10:
The effect_mixer is implemented as a C++ template. I'm not a template expert, but I cloned your code and got it building successfully. Please see the attached patch.

Templates are instantiated by string replacements and are not compiled to an object code. That's why the compiler needs the full template code, when it compiles *minidexed.cpp*.

The key is, that for templates normally everything (declaration from .h and implementation from .cpp) is put into the header file. That's why I included the *effect_mixer.cpp* into the header file too. You should put the full code instead there and remove the *effect_mixer.cpp* file.

[effect_mixer.zip](https://github.com/probonopd/MiniDexed/files/8458401/effect_mixer.zip)

rsta2, 2022-04-10:
In short here the naming conventions for Circle:

* Identifiers for types and variables will be written this way: "MyExampleIdentifier" (each word starts with an uppercase letter).
* Class names start with a "C", other types (struct, enum) with a "T".
* Member variable names of a class start with a "m_", static variables in a class definition with a "s_".
* Variable names get a prefix, which specify its (simple) standard type. This prefix follows the "m_" or "s_", if it is used. Variables of complex types do not have such prefix (except a "p" for pointer). Only one of these prefixes is used (i.e. "puch" for a pointer to an unsigned char is not used, use only "p" instead):

```
uch	unsigned char, u8, uint8_t
us	unsigned short u16, uint16_t
ul	unsigned long
ch	signed char, s8, int8_t
s	signed short, s16, int16_t
l	signed long
n	unsigned, int, u32, s32, uint32_t, int32_t, or any other (non-float) numeric variable
f	float, double
b	bool, boolean
p	Pointer to a simple or complex type
```

dcoredump, 2022-04-10:
> The effect_mixer is implemented as a C++ template. I'm not a template expert, but I cloned your code and got it building successfully. Please see the attached patch.

Many thanks @rsta2 !

Got it now. Good we have a real programmer in the team!

I have some small problems to fux, but I think I can manage this the next days.

probonopd, 2022-04-17:
Thanks @dcoredump for the idea and for implementing this. :+1:

closed 2022-04-17 by probonopd