probonopd/MiniDexed #77: Adding Chorus fx

dcoredump, 2022-04-08:
Should I add a chorus instance for each TG?

probonopd, 2022-04-08:
Having a chorus would be great.

I am not sure whether it should be per TG or "global" across everything (like the reverb currently is).
* What did one actually do in the 80s?
* Would it be significantly more CPU load to do it per TG?

rsta2, 2022-04-08:
Every additional stage in the audio signal chain adds some load. It can only be tested, when it is too much, especially on the RPi 2. We could decide to use `fast=true`, but this only works with a heat sink or fan on some RPi models.

probonopd, 2022-04-08:
So right now I am slightly leaning toward having effects global across all TGs (like reverb right now), unless someone thinks that is a really bad idea. After all, aren't effects mainly about simulating a room? All instruments are in the same room.

dcoredump, 2022-04-09:
I am a child of the 80s and I bought in 1987 a DX7II - and still have this nice synth. We used it for several years in our band, but the DX inside its flightcase is very heavy, so I decided to create a lighter version.

> * What did one actually do in the 80s?

I used a seperate FX (Alesis Quadraverb+) for my DX. But I think there is not much change for the FX chain:

Some FX are better used globally (like reverb, master limiter/compressor) and some are better located for each voice (like chorus, flanger, delay). This is not a law of nature but makes sense for most cases.

> Would it be significantly more CPU load to do it per TG?

Yes, sure. But the most CPU intensive FX are currently the compressor and the reverb. A chorus or a delay are more memory consuming but don't eat as much CPU time.

I have started with Synth_Dexed on a Teensy-3.6, which is a 180 MHz ARM Cortex M4 with 256 KB RAM. There was really a problem with CPU cycles and memory. This was solved with the Teensy-4.1 (600 Mhz ARM Cortex M7, 2 * 512 KB RAM). On the T_4.1 I can use 2 * Dexed (16 voices, chorus, delay per instance), 1 * MDA-EPiano (ROMpler, chorus, distortion), 8 Rompler-Drumsounds at a time (from 70 ROM samples) and a global reverb.

So I think: RPi3/4 should manage a delay and chorus for each TG without problems. Perhaps also RPi2. Only a real world test will show if it would do. IF not, it is not very problematic to use the FX as master FX instead of instance FX.

dcoredump, 2022-04-09:
Forgot to say :)

I have added some more NEON optimization for Synth_Dexed - hopefully this will give some more free CPU cycles.

probonopd, 2022-04-09:
@dcoredump so you are obviously the expert here. I'll leave the per-voice vs. global effects decision to you.

probonopd, 2022-08-06:
Hello @dcoredump for reverb we now have reverb-send, maybe the same could be done for chorus? Do you already have a chorus effect available that we could use for MiniDexed?