probonopd/MiniDexed #80: Real-time parameter control

probonopd, 2022-04-10:
It would be useful if changes to voice parameters would have immediate effect on already-playing notes.
This would be especially neat in combination with #63.

For some parameters this is already the case (e.g, volume), but apparently not for all of them. More investigation needed.

Do you perhaps know more about this subject @dcoredump?

probonopd, 2022-04-10:
https://narkive.com/6HNdPD06:3.358.923

> My experience with sending real-time parameter values to the DX7 has
been that if you send the usual sysex midi data, the voice will reset
every time, causing a stutter in the sound if the parameter values get
changed while you are playing a note. **The DX7II didn't have this
problem, and can have it's parameter values changed while being played
without any real problems.** In order to tweak parameters real-time on
the original DX7 without problems, you can use an undocumented
command. This lets you send midi commands to it that will mimic front
panel button presses. Only problem is, to tell it to change a
parameter value by a value of 20 would require telling it to pretend
that the + or - button was pressed... twenty times! At several bytes
per "press", the data adds up fast. Which means you can't update
values very fast. This undocumented command is also on later DX series
instruments, and is documented on them.

dcoredump, 2022-04-10:
> Do you perhaps know more about this subject @dcoredump?

I hope I am getting the Reverb-Send-Mixer ready the next days. After that, I would like to implement parameter changes via SYSEX.

I have an idea how to arrange this, but I have to test it. There is a kind of voice refresh function available, which has to be called after a change.

probonopd, 2022-04-10:
For parameter changes via sysex, I opened a separate [issue](https://github.com/probonopd/MiniDexed/issues/82).

This ticket here is about calling that voice refresh function when voice parameters are changed in the LCD GUI. Would one have to call `doRefreshVoice(void);` after each change? Or should it be sufficient to call `setRefreshMode(true);` once @dcoredump?

dcoredump, 2022-04-16:
I added a `doRefreshVoice();` when changing parameters, so I hope this should be fixed now. Please check out  the latest Synth_Dexed for trying.
(I hope this fixes also #91)

dcoredump, 2022-04-16:
> Or should it be sufficient to call `setRefreshMode(true);` once @dcoredump?

`setRefreshMode()` is for setting the note refresh mode... 

I have added a `doRefreshVoice()` to some of. the moethod of `class Dexed`.

probonopd, 2022-04-16:
Thanks for adding this capability @dcoredump. However, I have build issues for 32-bit RPis with the latest Synth_Dexed. Could you please have a look?

dcoredump, 2022-04-17:
Argh - don't know why this happens just now. But I hope I have fixed it in current Synth_Dexed. Please pull and try again. Compilation works for me with RPI[1234].

probonopd, 2022-04-17:
Works! :+1: 

Thanks @dcoredump and @rsta2

PR #98 (2022-04-17): Enable filter cutoff, resonance, and real-time parameter control
Closes #90, closes #80

Continued from #93 and #94
Thanks @rsta2 and  @dcoredump
closed 2022-04-17 by probonopd