probonopd/MiniDexed #82: Parameter changes via sysex

probonopd, 2022-04-10:
@dcoredump [said](https://github.com/probonopd/MiniDexed/issues/80#issuecomment-1094345167):

> I would like to implement parameter changes via SYSEX

That would be cool!

probonopd, 2022-04-18:
@jrdooley also [requested](https://github.com/probonopd/MiniDexed/discussions/110) this feature with the folliowing use case:

> For example, by connecting a class-compliant USB MIDI interface to the RPi, SYSEX messages could be sent to RPi/MiniDexed from an external device (e.g. a computer with sequencer sending SYSEX messages) updating OP1 Freq Coarse parameter immediately. This would allow for an external interface to program voices.

probonopd, 2022-04-18:
Hello @dcoredump are there changes in [Synth_Dexed](https://codeberg.org/dcoredump/Synth_Dexed/) needed so that this can be implemented in MiniDexed?

dcoredump, 2022-04-19:
Yes, I really want to move the SYSEX code from MicroDexed to Synth_Dexed, but currently I have not much time to manage this. But it's on my list.

jrdooley, 2022-04-19:
@dcoredump Great to hear that it's in the pipeline! This would really open up the creative possibilities for MiniDexed.

probonopd, 2022-04-20:
Note to self: Test this in https://synthmata.github.io/volca-fm/, an editor that runs purely in the browser.

dcoredump, 2022-04-26:
Just started the work on this 😀

dcoredump, 2022-04-28:
Update:

I have moved the SYSEX code from MicroDexed into Synth_Dexed (had also to change SYSEX handling in MicroDexed). Now implementing SYSEX features in MiniDexed. Maybe the next days the first test can start!

BenZonneveld, 2022-05-03:
Maybe an idea to add sysex control over parameters like volume,panning,reverb-send, detune and midi channel for each TG.
I am aware of midi control change for some of those parameters but when you have several TG's set to the same midi channel this is not practical.

probonopd, 2022-05-04:
Let's get the "normal" Dexed sysex working first. Later we can think about additional ones, but should do so in a way compatible with or at least similar to existing instruments.

jrdooley, 2022-05-04:
Fantastic news!

BenZonneveld, 2022-05-04:
After reading through some TX816 documentation I noticed they use the midi channel as the sysex ID so sysex is the way to do it.

probonopd, 2022-05-15:
Turns out that voice loading is __working__ but for it to work **one must not set the tone generator(s) to MIDI channel "Omni"** but to channel 1 (or another channel that is set in Dexed on the PC under PARM -> DX7 Channel).

Is this a bug or is it intentional that voice sysex are ignored on channel "Omni"?

studionebula, 2022-05-16:
FWIW, the original DX7 manual seems to say that voice sysex messages ("bulk data of 1 voice") are channelized. There's definitely a channel byte in the message format, and no mention of reception of this message in Omni mode.

probonopd, 2022-05-16:
Thanks for looking it up @studionebula.

So it looks like it is a feature, not a bug, that TGs set to Omni won't receive voice sysex.

But the other way around, why does one need to set TGs to Omni so that one can receive parameter sysex (for on-the-fly editing)?

studionebula, 2022-05-16:
You're welcome, and thank you for all of your work.

Without knowing anything about the codebase, I'm going to go out on a limb and say that the Omni requirement for parameter sysex is a bug, at least as far as comparison with the original hardware goes. In the section on sysex messages, the manual says "Parameter change - Voice parameter and function parameter request reception with the same format as transmission when MIDI channel numbers are corresponded, system information is available, and Memory protect is off." [grammar and punctuation oddities as per the original text] That's essentially the same verbiage for the voice bulk-data message. On top of that, it doesn't make sense to me that you'd want parameter-change messages to _not_ be channelized, if you consider the use case of e.g. the TX816 where you might have several synths receiving the same data simultaneously.

probonopd, 2022-06-05:
Looks like we can close this as it has been implemented, or are any aspects not working properly yet?

BobanSpasic, 2022-07-11:
How do I save the transferred voice?
After sending a patch/voice from synthmata, on MiniDexed I get the patch name with a question mark at the end. How do I save the changes now?

probonopd, 2025-04-20:
To answer @BobanSpasic: Save a Performance.

Closing, as this has been implemented. Please feel free to comment in case you think there are aspects of this still not implemented. Thanks!

closed 2025-04-20 by probonopd