probonopd/MiniDexed #83: Some voices are not played

probonopd, 2022-04-10:
https://yamahablackboxes.com/patches/dx7/factory/rom3a.syx

* 17=SAX BC
* 24=VOICES BC

TODO: Test with other software instruments

probonopd, 2022-04-10:
One this mystery is solved, we may consider to create nice performances using the TX816  voices.

For example, set TG1...8 to use `Tfi1...8.syx` from `Dexed_cart_1.0/Original Yamaha/TX816/`, each voice 11. (Possibly no additional detune; tbc.) The result is a nice rich piano

probonopd, 2022-04-11:
@dcoredump can you reproduce this issue, also on MicroDexed?

dcoredump, 2022-04-11:
I have to check this tomorrow.

dcoredump, 2022-04-12:
> @dcoredump can you reproduce this issue, also on MicroDexed?

Couldn't try, sorry.

Perhaps those sounds use the breath controller for the AMP envelope?

dcoredump, 2022-04-16:
Hm, have to check this... sorry, currently not much free time for this 😢

frank-que, 2022-06-21:
Hello, I've noticed a strange behaviour, sounds that use algorithm 16,17,18 are muted.

frank-que, 2022-06-21:
I've found where is the problem:
turn left one step "A MOD Sens" on OSC1 and the sound are back.
All setting with "A MOD Sens" on OSC1 at maximum value are muted.
There is some problem there...

dcoredump, 2022-06-21:
> Hello, I've found where is the problem: turn left one step "A MOD Sens" on OSC1 and the sounds are back. In some conditions with "A MOD Sens" on OSC1 at maximum value sounds are muted.

Hi @frank-que!

Thanks for finding THIS issue... I will try to fix it inside Synth_Dexed asap.

Regards, Holger

frank-que, 2022-06-21:
@dcoredump 
A heartfelt thanks to you and to all of the developers for the amazing work that you are doing!

dcoredump, 2022-06-22:
@frank-que 
Can you tell me at which Bank/Voice this problem encounters?

frank-que, 2022-06-22:
The problem seems derived by combination between "Algorithm" and "A MOD Sens"

dcoredump, 2022-06-22:
It looks like a bigger problem... I am 90% sure that it is located inside Synth_Dexed. I think it will take some time (for me) to locate the problem. Stay tuned...

frank-que, 2022-06-22:
@dcoredump 
take your time.
I would like to help you, but I don't have enough knowledge to do it.

dcoredump, 2022-06-23:
Sorry @probonopd - I totally forgot to check if the problems exists on MicroDexed. Now I have done this and the answer is: **NO**.

Band 3A / SAX BC works fine. Also 3A / TRAIN. I used the latest Synth_Dexed (28ef0da952) and everything is fine. So the problem seems to be inside MiniDexed. I tried to check the sysex reading code and  found no problem - but I am pretty sure that the problem must be located when reading sysex data...

arsamus, 2022-06-23:
>So the problem seems to be inside MiniDexed. I tried to check the "sysex reading code" and found no problem - but I am pretty sure that the problem must be located at reading sysex data...

An easy way to test this hypotesis is loading, for example, "SAX BC" data on a PC version of Dexed and send it to MiniDexed vía MIDI. Bank voice data and individual voice data have different sysex structures. Bank voice data needs to be decoded once received in order to extract all parameters, some parameters share the same byte, on the other hand, individual voice data has one byte for each individual parameter. The issue could be on the decoder algorithm.
If I have some time later I'll try to test this.

frank-que, 2022-06-23:
I've already tested send sysex data from dexed, it was when I found the problem, I use dexed as minidexed frontend.

arsamus, 2022-06-23:
> I've already tested send sysex data from dexed, it was when I found the problem, I use dexed as minidexed frontend.

Yes, I've just tested too. In this case, I don't think that it is a SysEx issue...

probonopd, 2022-07-02:
https://yamahablackboxes.com/articles/how-to-program-yamaha-dx7/ says:

> __When the EG bias for a controller is on,__ the controller overrides the controller envelopes for which amplitude sensitivity has been programmed. This will be a complete override if the amplitude sensitivity is set to full on (3), while at lower levels (1 or 2) the internal envelope will still have a reduced but audible effect. The full-on setting is especially useful for wind instrument sounds, because you can control their amplitude from the breath controller, giving realistic articulations. __You can hold down a note on the keyboard and get no sound until you blow into the breath controller.__ When EG bias is off but amplitude or pitch control and sensing are on, a controller becomes an LFO amount control.

So it looks like EG bias is on by default in MiniDexed, whereas it should be off by default (and possibly a way should be added to enable it)?

It seems like you [came to the same conclusion](https://github.com/probonopd/MiniDexed/discussions/298#discussioncomment-3067132) @arsamus, would be great if you could send a PR. Thanks!

arsamus, 2022-07-03:
Hi everybody, you can test the patch to this fix with this build https://github.com/probonopd/MiniDexed/suites/7193919829/artifacts/287422925

You have to set off Modulation Wheel -> EG Bias on the new "Modulation" menu of each TG

frank-que, 2022-07-03:
Hi @arsamus , I've tested it on RPI zero, now seems to work, but with some artifacts (I use PCM1502 module).
Probably this is due by RPI zero performance, but it is very strange, because my modified stereo version for RPI zero, with only one TG  and reverber active, works weel, without artifacts.
I have an RPI4, but I cant'use it for the testing, it is used on my Zynthian.
Thank you for your effort for the project and sorry if my english is not perfect, is not my native language, but I hope you understand what I wrote :)

Frank

arsamus, 2022-07-03:
> Hi @arsamus , I've tested it on RPI zero, now seems to work, but with some artifacts (I use PCM1502 module).
> Probably this is due by RPI zero performance, but it is very strange, because my modified stereo version for RPI zero, with only one TG  and reverber active, works well, without artifacts.
> I have an RPI4, but I cant'use it for the testing, it is used on my Zynthian.
> Thank you for your effort for the project and sorry if my english is not perfect, is not my native language, but I hope you understand what I wrote :)
> 
> Frank

Hi Frank. It is very strange because this PR doesn't add any subprocess to the main Process loop respect the Productive version or on Synth Dexed itself. All changes occurs only when you sulf on some menus and interact with the existing functions/methods of Synth Dexed, but who knows... Let me see if there are some part of code that may be responsible for the issue you've described.

frank-que, 2022-07-03:
Hi @arsamus, I'll try to check (after Silverstone GP) if I've made something wrong.  ;)

frank-que, 2022-07-03:
@arsamus, you're right, I've tested again copying all file to SD and now it works as expected.
Probably the rush made me commit some mistakes.
Have a good evening!

arsamus, 2022-07-03:
That's good! Thanks!

probonopd, 2022-08-06:
Fixed in https://github.com/probonopd/MiniDexed/commit/aa5a7c7450a86b95f4eb80e8122a802eefd4accc

closed 2022-08-06 by probonopd