probonopd/MiniDexed #88: [Idea] Raspberry Pi hat hardware

probonopd, 2022-04-12:
[lonk2905](https://www.reddit.com/user/klonk2905/) wrote over at [Reddit](https://www.reddit.com/r/synthdiy/comments/syup5a/comment/hy2twxk/?utm_source=reddit&utm_medium=web2x&context=3):

> Hey, I'm willing to collaborate on the hardware part.
>
> A "hat" with decent power supply, quality DAC, and dedicated sliders/controls will be top notch at some point, and that is something I can help with.

Let's collect the ideas and requierments here.

D4p0up, 2022-04-14:
(Klonk 2905 Here) Thanks for letting me in. 

Great suggestions, a few philosophical questions first : 
- what is the interaction philosophy you would expect from such hardware? Should it mimic the "frugality" of the original - i.e. one 16x2 display and a couple of buttons" - or do we push further (rotary encoders, ribbons, tactile displays, etc...)?
- What is the form factor expected? Matchbox ultra integrated PIZero Synth, desktop machine, 19' Rack?

Should the direction be settled, I could propose some quick mockups on Openscad to match the interaction/mechanical/PCB/RPi cinematics with the form factor/concept.

probonopd, 2022-04-15:
Hello @D4p0up, welcome.

Yes, I'd say let's mimic the "frugality" of the original. In fact, our UI is even way more simplistic than the original (16x2 display with just 1 rotary encoder and 1 button so far). If we go down the path of making hardware, we should possibly add just a few more buttons to allow for future expanadability and/or UI shortcuts. Possibly we should think about making it compatible with [mt32-pi](https://github.com/dwhinham/mt32-pi) as well.

> What is the form factor expected?

Minimalistic, inexpensive, using jellybean parts. People shold be able to make them at home.

My first idea was to make a Raspberry Pi Hat that sits on top of a Pi with the least extra space needed and in a cost-effective way. 

Actually there is already one for the [mt32-pi](https://github.com/dwhinham/mt32-pi#%EF%B8%8F-project-status). Just missing "our" rotary encoder and button, and a different display.

But that makes for a relatively dull box (like my 3D printed prototype on the [README](https://github.com/probonopd/MiniDexed#minidexed-) page). People might also want to stick them into Euroracks, possibly 19" racks. To allow for those form factors, one would probably need to use a ribbon cable from the Pi to the MiniDexed PCB.

What are others' thoughts?

dcoredump, 2022-04-16:
Hi @D4p0up  and welcome!

a decend hat with audio codec and DIN MIDI I/O would be very nice. The hardware audio outputs should be on two 6.3mm jacks and perhaps also with an automaticly  mono switch when using only the left output. This would make MiniDexed perfectly fitting for a connection to mixing console as well as for a live gig (only connected mono to a monitor box).

I think the minimalistic display and one encoder is enough for fast switching sounds and some parameters. For programming an external MIDI editor can be used. Or someone writes a UI for a HDMI touch display like for [minisynth](https://github.com/rsta2/minisynth).

bengarland, 2022-04-18:
Would ClumsyMIDI work with MiniDexed?

probonopd, 2022-04-19:
The 0.91" OLED display is currently not supported by MiniDexed, only 16x2 alphanumeric displays without i2c at the moment. Although this [might](https://github.com/luisfcorreia/MiniDexed/commit/ada781ee334fdb25d99a04d04cfed13ad60b308d) change in the future if @luisfcorreia succeeds.

diyelectromusic, 2022-04-28:
Re: ClumsyMIDI.

If you leave off the I2C display, Clumsy MIDI works fine with MiniDexed for serial MIDI and the PCM1502 I2S audio output if you configure it for I2S audio.  You have to add the rotary encoded and 1602 LCD as extras as before of course, but GPIO wise it all works out fine.

I have a few notes on this here: https://diyelectromusic.wordpress.com/2022/04/28/bare-metal-raspberry-pi-minidexed-dx7/

Kevin

JohnConnor84, 2022-04-30:
Please a 4x4 Gpio Membrane keypad menu implementation would be so awesome! pleaaase! My idea is indeed a eurorack format or a 2U 19inch rack version.

probonopd, 2022-04-30:
@JohnConnor84 you have now posted the idea "4x4 Gpio Membrane keypad" in 3 different places.
The best place to start a discussion around this would be [here](https://github.com/probonopd/MiniDexed/discussions/new?category=ideas).

probonopd, 2022-05-04:
@diyelectromusic has made one out of perfboard: :+1: 

[image]

https://diyelectromusic.wordpress.com/2022/05/03/bare-metal-raspberry-pi-minidexed-dx7-part-2/

probonopd, 2022-05-07:
https://github.com/probonopd/MiniDexed/issues/163#issuecomment-1120297634

probonopd, 2022-05-13:
Note to self: https://aisler.net/, including parts

stevelittlefish, 2022-06-06:
I'm also interested in this and considering designing a PCB.  Some ideas:

- I'd much rather ditch the rotary encoder module as it makes it more difficult with a PCB.  I would put the encoder straight on the PCB with pull up resistors and a simple RC filter for debouncing (I've done this in the past with a 10k resistor and 10nF capacitor for some eurorack modules).
- I would use SMD parts for the small bits a pieces and get them pre-assembled.  They would be small and not really cost enough to care about.
- Ideally I'd put the DAC chip on the PCB and just replicate one of those modules, but it's probably too time consuming and would increase the cost (those modules are cheap).
- I would like either RCA connectors or dual 6.35mm TS jacks, but that might make it too big.
- I'd also like a DIN midi input.  I could probably get the optocoupler presoldered.  If a full sized DIN wouldn't fit then TRS mini jack could work.
- The LCD screen I've been using for testing has the connector in a different place and it has 2 rows of pins instead of 1.  Rubbish for breadboarding but fine for the PCB.  Ideally I'd make it compatible with both layouts.
- In addition to the encoder I would put a single push button to function as a back button (long press for home) and use the encoder button only to select the current menu item.
- Trimmer potentiometer somewhere for contrast would be nice. Probably with optional additional holes to just add a resistor.

I can do the schematic and PCB layout and would share all of the files.  You could then order PCBs directly from JLCPCB (including pick and place).  I could possibly send a few boards out too.

For an enclosure someone could design a 3d-printable box, or we could do it semi enclosed with a front and back panel (made out of PCB material) and screwed through the corners using machine screws and stand offs.

I think the hardest part would probably be designing it so that all of the user interface components fit in a nice layout and are the correct distance to the front panel.  This would be easier if we forgo the enclosure.

diyelectromusic, 2022-08-23:
> @diyelectromusic has made one out of perfboard: 👍

By the way, I now have a PCB version too... See discussion here: https://github.com/probonopd/MiniDexed/discussions/335

probonopd, 2022-09-04:
[image]

https://diyelectromusic.wordpress.com/2022/08/16/minidexed-raspberry-pi-io-board-part-3/

Great work @diyelectromusic :+1:

closed 2022-09-04 by probonopd