probonopd/MiniDexed #89: Optimize rotary encoder movement

probonopd, 2022-04-15:
Currently, If two clicks are needed for a single movement. `mt32-pi.cfg` has this information:

```
# Set the rotary encoder type (if used by control scheme).
#
# Different rotary encoders may complete different fractions of a Gray-code
# cycle per detent ("click").
#
# If four clicks are needed for a single movement, try "quarter".
# If two clicks are needed for a single movement, try "half".
#
# Values: quarter, half, full*
encoder_type = full
```

Next step: Try `mt32-pi.cfg` with my rotary encoder.

rsta2, 2022-04-16:
The original code is [here](https://web.archive.org/web/20210121145439/https://embdev.net/articles/Rotary_Encoders). I think it is easier to understand. It needs to sample the inputs from the rotary encoder with 1 KHz. mt32-pi uses its own CPU core for the UI and this purpose. We cannot do this, because we need all secondary cores for the audio processing. I could install an 1 KHz user timer on core 0, but with bad luck its IRQs will not be handled quick enough, because an other IRQ handler is running at this time already.

First I will try to extend the Circle KY-040 driver for the "quarter", "half" and "full" cases. If this does not work, I will try the user timer.

rsta2, 2022-04-19:
@probonopd Extending the Circle KY-040 driver did probably not work. Unfortunately I also do only have one KY-040 and I even do not know, if it is "quarter", "half" or "full". I did think about this some time now. Because I already implemented one rotary encoder driver for MiniDexed and Circle, I won't do again.

You will find other developers for sure, which are ready to do this. I'm afraid I have to draw back from your project in one month, to be able to concentrate on my own projects again. This all costs a lot of time.

probonopd, 2022-04-19:
@rsta2 you have been doing pure magic both with Circle and with this project. Who would have thought the project would land on the frontpage of [Hackaday](https://hackaday.com/2022/04/19/bare-metal-gives-this-pi-some-classic-synths/)? All the [credit](https://github.com/probonopd/MiniDexed#acknowledgements) goes to you and to the other giants which this project stands on the shoulders of. You can't be thanked enough :1st_place_medal:

rsta2, 2022-04-19:
Thanks for your understanding! Yes, it was fun. Unfortunately I'm not musician enough to enjoy all the coming features of MiniDexed enough, to continuously work on this project. The (shrunken) TX816, that I have now, is already great enough for me. ;)

Giandj, 2023-02-17:
Is it normal that my rotary encoder miss some pulse ? If I turn clockwise voice change sometimes very well , sometimes I have to rotate again and again to switch from voice1 to voice2 ...3..4...until 32
Instead if i turn it anticlockwise is much reactive voice from 32...4...3...2...1 chance are more sensitive on anticlockwise turns of rotary encoder why ?
The rotary encoder seems to miss some rotation..  what kind of test can I do to encrease encoder sensitivity on pi3b+ ?

craigyjp, 2023-02-17:
Have you installed the 10nf caps on the data pins of the encoder to ground?

It helped with the rotation on both my builds.

Banana71, 2023-02-17:
Hi, 

#322

Banana71, 2023-02-17:
You nerd two 10nF capacitor.
One between CLK and GND 
And one between DT and GND

Giandj, 2023-02-17:
> One between CLK to GND
> And one between DT to GND

Can you give me a schema ? in my case my rotary encoder is wired on rapsberry pi3b+ ,
This is the schema i followed... https://drive.google.com/file/d/1ol9fCtepMS7_lNUFYss7EbOfytvl94Uj/view

Banana71, 2023-02-17:
KY-040 from the back with two capaciators  CLK to GND and  DT to GND.
[image]

[image]

You have a pretty old version there
If I were you, I would download the current version here and read the wiki.
If you have further questions, it is best to open a new discussion.

Giandj, 2023-02-17:
I will try whit 2 capacitors thank you..

Giandj, 2023-02-22:
> KY-040 from the back with two capaciators CLK to GND and DT to GND. [image]
> 
> [image]
> 
> You have a pretty old version there If I were you, I would download the current version here and read the wiki. If you have further questions, it is best to open a new discussion.

Hi I have installed 2 capacitors of 10 Nano Farad and now the rotation of KY-40 encoder is better but not perfect... some rotation are also missing ...
Another question: why you said I have a pretty "old version" of Minidexed? I have the latest version installed but rotary ky-40 encoder still miss some rotations (with 2 capacitors correctly installed like in your precedent schema ... Can I try to modify the code in mt32.cfg file to try if full, half or quarter can adjust better the missing rotations , have you tried ?

Banana71, 2023-02-22:
> This is the schema i followed... https://drive.google.com/file/d/1ol9fCtepMS7_lNUFYss7EbOfytvl94Uj/view

> Another question: why you said I have a pretty "old version" of Minidexed?

In your TX816Pi2.zip file there is a version from 06/05/2022 and I probably drew the wrong conclusions.

> I have installed 2 capacitors of 10 Nano Farad and now the rotation of KY-40 encoder is better but not perfect... some rotation are also missing ...

I've built a few miniDexeds and have always had success with the capacitors. Is your KY-040 ok?

> Can I try to modify the code in mt32-pi.cfg file to try if "full", "half" or "quarter" can adjust better the missing rotations , have you tried ?

I have no experience there. The file "mt32-pi.cfg" is unknown to me.

Giandj, 2023-02-24:
> > This is the schema i followed... https://drive.google.com/file/d/1ol9fCtepMS7_lNUFYss7EbOfytvl94Uj/view
> 
> > Another question: why you said I have a pretty "old version" of Minidexed?
> 
> In your TX816Pi2.zip file there is a version from 06/05/2022 and I probably drew the wrong conclusions.
>yes because after downloading it I have made an upgrade.. 

> > I have installed 2 capacitors of 10 Nano Farad and now the rotation of KY-40 encoder is better but not perfect... some rotation are also missing ...
> 
> I've built a few miniDexeds and have always had success with the capacitors. Is your KY-040 ok?
> 
My Encoder is ok, I have changed 3 encoders with capacitors , but sometimes some rotation are missing...no problem...before 2 capacitors the rotation issue was most frequent... 

> > Can I try to modify the code in mt32-pi.cfg file to try if "full", "half" or "quarter" can adjust better the missing rotations , have you tried ?
> 
> I have no experience there. The file "mt32-pi.cfg" is unknown to me.

In another post (issue) somebody wrote that we can modify quadrature calculation in this text file to make encoder rotation much better...
So In this file you can change quadrature from full, half, or single..
Quadrature is a quantity of encoder movements to obtain a single move...
So in this Minidexed project some .txt or .cfg file specify to the raspberrypi how much movements to do ( if encodet rotation Is clockwise or anticlockwise )..
I Will try to modify quadrature calculation...

Gustavomurta, 2023-07-22:
I have not used Ky-40. I built a circuit with another rotary encoder. And it's working perfectly.

[image]

GusIsanda, 2024-10-12:
I had the same irresponsiveness problem and I figured out something:

[image]

I realized switch button is attached to VCC. In Gustavomurta approach the switch goes to GND (I followed his schematic and everything works perfect). I guess switches to 3V3 interfere with the code at some point generating the laggy behaviour. I did not check the rotary code but It seems easy to fix.

probonopd, 2024-11-15:
Gotta say that with the Dtronics DT-DX it works flawless, so it seems to be a hardware rather than a software issue.

closed 2024-11-15 by probonopd