probonopd/MiniDexed #90: Expose Cutoff and Resonance global Dexed paramters in menu

probonopd, 2022-04-15:
Similar to (De)tune, we should expose the Cutoff and Resonance global Dexed paramters in the menu and save them in `performance.ini`.

[image]

rsta2, 2022-04-16:
I think this can be added. The respective methods are in the `Dexed` class. Will do so.

rsta2, 2022-04-16:
Unfortunately it's not that simple. There is a define `USE_FX` in *PluginFx.cpp*, which probably needs to be enabled, to enable the filter. Also I do not know the parameter range for the following methods:

```
void setFilterCutoff(float cutoff);
void setFilterResonance(float resonance);
```

@dcoredump Can you give some clarification? Does the filter increase the load much?

dcoredump, 2022-04-16:
I am using the filter inside MicroDexed. The `USE_FX` guard is present because I wanted to disable the complete FX stack in MicroDexed to be able to use older processors.

For a RPi enabling the EQ should not be a problem.

dcoredump, 2022-04-16:
Just forgot: the values for the filters are 0.0f .. 1.0f.

rsta2, 2022-04-16:
@dcoredump Thanks!

rsta2, 2022-04-16:
Implemented in #93.

probonopd, 2022-04-17:
Works! :+1: Thank you very much.

PR #98 (2022-04-17): Enable filter cutoff, resonance, and real-time parameter control
Closes #90, closes #80

Continued from #93 and #94
Thanks @rsta2 and  @dcoredump
closed 2022-04-17 by probonopd