probonopd/MiniDexed #91: Osc Mode without function

probonopd, 2022-04-15:
Accroding to the DX7 manual,

> If MODE is set to FIXED FREQ (HZ), COARSE adjustment is possible in four steps-1, 10, 100 and 1000. FINE adjustment is possible from 1 to 9.772 times.

Currently in MiniDexed, setting _Osc Mode_ to _"fixed" does not seem to have any impact on the behavior of _Freq Coarse_ and _Freq Fine_.

rsta2, 2022-04-16:
Can this be caused by the missing refresh function similar to #80?

probonopd, 2022-04-17:
I thought the menu entries would change when one switches __Osc Mode_ to _Fixed_, but __Freq Coarse__ and __Freq Fine__ still have `<-1->`. Do I undertsand it wrong?

rsta2, 2022-04-17:
No, the menu entries do not change. Freq Coarse and Fine are always there, despite of the Osc Mode setting.

probonopd, 2022-04-17:
Doing more experiments, I can confirm that both modes are working.

But what is shown in the display is very different from what a DXx/TXxxx would show, and hence I can't do the examples from the books with MiniDexed at the moment.

probonopd, 2022-04-17:
I think this is how it _should_ be:

* Freq Coarse: If Osc Mode is set to Ratio, the frequency ratio can be set from one half (0.5 times) to 32 times, default is 1 _-  these values should be shown in the display?_ If Osc Mode is set to Fixed, one of the following frequencies can be selected: 1, 10, 100, and 1000 Hz _-  these values should be shown in the display?_
* Freq Fine_ If Osc Mode is set to Ratio, the frequency set by Freq Coarse can be adjusted from 1 to 1.99 times _-  these values should be shown in the display?_ If Osc Mode is set to Fixed, the frequency set by Freq Coarse can be adjusted from 1 to 9.772 times _-  these values should be shown in the display?_

probonopd, 2022-04-18:
Reference:
[For the intended ](https://github.com/probonopd/MiniDexed/wiki/Exercises#exercise-17-fixed-mode)

rsta2, 2022-04-19:
Not that simple to implement this, like it may look like. Currently the UI is designed for the assumption, that there are no dependencies between different parameters, when deciding, how to implement their values. I think about, how this can be changed.

probonopd, 2022-04-19:
Ah, I understand. No hurry! In the meantime, maybe we should have the menus print both values (variable and fixed)?

rsta2, 2022-04-19:
I would like to know, how is the exact mapping from the parameter range 0-31 for coarse to 0.5 to 32 times for ratio and 1, 10, 100, 1000 Hz for fixed? Respectively how is the mapping from 0-99 for fine to 1 to 1.99 times for ratio and 1 to 9.772 times for fixed. At the moment the raw parameter values 0-31 and 0-99 are shown.

probonopd, 2022-04-19:
Ratio:
* **Coarse**: 0.5, 1, 2, ... 31
* **Fine**: 1, 1.01, ... 1.99 times the selected **Course** value (if it is too much work to implement this, then show the unmultiplied values)

Fixed:
* **Coarse**: 1 Hz, 10 Hz, 100 Hz, 1000 Hz (strangely the knob in Dexed repeats these a couple of times, bug?)
* **Fine**: 1 Hz, 1.02329 Hz, 1.04713 Hz, 1.07152 Hz, ... 9.77237 Hz times the selected **Course** value (if it is too much work to implement this, then show the unmultiplied values). I don't know where these odd values are coming from, maybe @asb2m10 can tell us?

rsta2, 2022-04-20:
See #123.

probonopd, 2022-04-20:
Testing asap. Thank you!

probonopd, 2022-04-20:
Coarse in Fixed mode repeats the same values a couple of times, but I think that's more a curiosity than a bug. No need to invest time to change this imho.

probonopd, 2022-04-20:
Closed in https://github.com/probonopd/MiniDexed/commit/43e0c1872c576dcbb217e2c54a51cee12049b112

Thanks @rsta2

closed 2022-04-20 by probonopd

probonopd, 2022-04-20:
Fwiw, I found the official values documented in the TX816 manual:

[image]