probonopd/MiniDexed #101: Reorder menu items in "Edit Voice"

probonopd, 2022-04-17:
I think we should reorder some of the menu items in "Edit voice" (e.g., to bring Edit Voice -> TGx -> Output Level closer to the beginning of the submenu, and to bring "Operator" closer to the beginning of its respective submenu).

Does the current order follow a particular logic we should stay at @rsta2? if not, what would I need to know if I wanted to reorder things a bit?

rsta2, 2022-04-17:
TG1-8 must be the first entries in `s_MainMenu` and OP1-6 must be the first entries in `s_EditVoiceMenu`. This is because of the shortcut functions. Everything else can be changed.

probonopd, 2022-04-17:
Thanks. I can live well with that restriction (didn't want to change that part anyway). Will do some subtle order changing experiments soon.

closed 2022-04-17 by probonopd

probonopd, 2022-04-17:
Also updated the [documentation](https://github.com/probonopd/MiniDexed/wiki/Menu) accordingly.
This arrangement makes it quicker to do examples from the [books](https://github.com/probonopd/MiniDexed/wiki/Recommended-Resources#books).

PR #106 (2022-04-17): Reorder menu items
Closes #101
commit 46e4e864 (2022-04-17): Reorder menu items (#106)

Closes #101