probonopd/MiniDexed #105: "MIDI Through" to Serial MIDI

probonopd, 2022-04-17:
https://github.com/probonopd/MiniDexed/issues/20#issuecomment-1074305908 makes me wonder whether it would be possible to forward the MIDI events coming into MiniDexed via USB to Serial MIDI.

This way, a computer or tablet connected to MiniDexed using a Raspberry Pi Zero (running a Serial MIDI to USB Midi bridge firmware) could be used to run a keyboard learning application like [Synthesia](https://synthesiagame.com/).

rsta2, 2022-04-19:
Probably possible. The question is, if this MIDI forwarding should be done generally or only if switched on in the UI.

probonopd, 2022-04-19:
I think a setting in the `minidexed.ini` file should do.

rsta2, 2022-04-21:
See #129.

closed 2022-04-21 by probonopd