probonopd/MiniDexed #107: feature request: scripting support

maks, 2022-04-18:
I think it would be very nice to be able to have scripting of MiniDexed functions using eg. Lua or JS with a light weight, easily embeddable engine like [QuickJS]( https://bellard.org/quickjs/)

This would also allow the possibility of add lots of functionality that perhaps not all users would want such as step sequencing, control via specialised devices such as [I mentioned here](https://github.com/probonopd/MiniDexed/issues/63#issuecomment-1100989705) , different UI's as mentioned in #62, all without needing to write it completely in low level C/C++ in the core of MiniDexed.

nikitalita, 2022-04-18:
Mt32-pi implemented something like this

maks, 2022-04-18:
@nikitalita thank you that's interesting to know that project is implementing scripting too, though I'm much more interested in this project and using FM synth than soundfonts with fluidsynth.

nikitalita, 2022-04-18:
Sure, I’m just suggesting that the maintainer use that as a reference since they are similar projects and work on the same platform

probonopd, 2022-04-18:
Interesting idea, although I am not sure how this should be integrated technically.

> Mt32-pi implemented something like this

Can you please point to its documentation where the embedded scripting is described? Can't seem to find it. Thanks.