probonopd/MiniDexed #111: Allow enabling and disabling operators

probonopd, 2022-04-18:
Allow enabling and disabling operators from the menu to speed up operation.
Currently, one needs to set the Output Level of an operator to 0.

Reference: [Exercise 8](https://github.com/probonopd/MiniDexed/wiki/Exercises#exercise-8-enabling-and-disabling-operators) in "The Complete DX7"

We could do this similar to **MiniDexed -> Effects -> Enable**:

**MiniDexed -> TGx -> Edit Voice -> OPx -> Enable**.

rsta2, 2022-04-19:
I think there is currently no possibility to disable OPs in Synth_Dexed. They are always on.

probonopd, 2022-04-19:
@dcoredump would you consider adding such functionality? It is in Dexed. These switches: 
[image]

dcoredump, 2022-04-19:
Synth_Dexed has got a method: `void setOPAll(uint8_t ops);`

For enabling/disabling you have to use a bitmask (`uint8_t ops`).

Is this what you are sesrching for?

probonopd, 2022-04-19:
In fact, `void setOP(uint8_t op, bool state);` might be what I was after.

probonopd, 2022-04-19:
I would think that I need to add `	{0,	1,	1,	ToOnOff}		// DEXED_OP_ENABLE` to the end of

https://github.com/probonopd/MiniDexed/blob/74772a689a74b30385018f93a4f3bd7cbf95766b/src/uimenu.cpp#L214-L238

but it is not obvious to me what else is needed, @rsta2 can you give me a hint please?

rsta2, 2022-04-19:
I cannot answer this straight out. Have to think about it.

rsta2, 2022-04-20:
I would implement this as follows:

* Add a `uint8_t m_uchOPMask[CConfig::ToneGenerators]` member variable to `CMiniDexed` to save the current enable state of the different OPs for all TGs. This is necessary, because the current state cannot be read back from `Dexed` (actually not a problem). This member must be initialized to 0x3F (all OPs enabled) for all TGs [somewhere here](https://github.com/probonopd/MiniDexed/blob/main/src/minidexed.cpp#L71).
* Add a `#define DEXED_OP_ENABLE (DEXED_OP_OSC_DETUNE + 1)` [here](https://github.com/probonopd/MiniDexed/blob/main/src/dexedadapter.h#L27). This is not a real Dexed parameter, but necessary for the UI.
* In `CMiniDexed::SetVoiceParameter()` there must be an `if` statement, which checks, if is is called for `nOP < 6 && uchOffset == DEXED_OP_ENABLE`. In this case the bit for the respective operator in `m_uchOPMask[]` must be set or reset, depending on `uchValue`. Afterwards `m_pTG[nTG]->setOPAll(m_uchOPMask[nTG])` gets called to propagate the new OPs state to `Dexed`. The processing for other `uchOffset` values remains unchanged.
* In `CMiniDexed::GetVoiceParameter()` there must be an `if` statement, which checks, if is is called for `nOP < 6 && uchOffset == DEXED_OP_ENABLE`. In this case the bit for the respective operator in `m_uchOPMask[]` must be returned (0 or 1).
* Now the line `{0, 1, 1, ToOnOff} // DEXED_OP_ENABLE` can be added to `CUIMenu::s_OPParameter[]`, as already suggested above.
* Finally an entry `{"Enable", EditOPParameter, 0, DEXED_OP_ENABLE}` has to be added to `CUIMenu::s_OperatorMenu[]`, so that the setting can be manipulated in the UI.

This should be all.

probonopd, 2022-04-20:
Thanks for your detailed instructions @rsta2. I am probably making a fool of myself, but I have trouble with "the bit for the respective operator in `m_uchOPMask[]` must be returned (0 or 1)". Probably this is wrong:

https://github.com/probonopd/MiniDexed/blob/ded7db2d3589f4821e8238b455cb081e26db5662/src/minidexed.cpp#L715

In any case, the additional menu entry appears but MiniDexed crashes upon selecting it.

rsta2, 2022-04-20:
Please change the methods `CMiniDexed::SetVoiceParameter()` and `CMiniDexed::GetVoiceParameter()` as follows:

```
void CMiniDexed::SetVoiceParameter (uint8_t uchOffset, uint8_t uchValue, unsigned nOP, unsigned nTG)
{
	assert (nTG < CConfig::ToneGenerators);
	assert (m_pTG[nTG]);
	assert (nOP <= 6);

	if (nOP < 6)
	{
		if (uchOffset == DEXED_OP_ENABLE)
		{
			if (uchValue)
			{
				m_uchOPMask[nTG] |= 1 << nOP;
			}
			else
			{
				m_uchOPMask[nTG] &= ~(1 << nOP);
			}

			m_pTG[nTG]->setOPAll (m_uchOPMask[nTG]);

			return;
		}

		nOP = 5 - nOP;		// OPs are in reverse order
	}

	uchOffset += nOP * 21;
	assert (uchOffset < 156);

	m_pTG[nTG]->setVoiceDataElement (uchOffset, uchValue);
}

uint8_t CMiniDexed::GetVoiceParameter (uint8_t uchOffset, unsigned nOP, unsigned nTG)
{
	assert (nTG < CConfig::ToneGenerators);
	assert (m_pTG[nTG]);
	assert (nOP <= 6);

	if (nOP < 6)
	{
		if (uchOffset == DEXED_OP_ENABLE)
		{
			return !!(m_uchOPMask[nTG] & (1 << nOP));
		}

		nOP = 5 - nOP;		// OPs are in reverse order
	}

	uchOffset += nOP * 21;
	assert (uchOffset < 156);

	return m_pTG[nTG]->getVoiceDataElement (uchOffset);
}
```

The member variable `m_uchOPMask[]` should be placed in the `private` section of the class `CMiniDexed`, so that it cannot be accessed directly from outside of the class (information hiding).

probonopd, 2022-04-20:
...works!

commit 57470778 (2022-04-20): Allow enabling and disabling operators, closes #111 (#118)

* Allow enabling and disabling operators, closes #111

Thanks @rsta2
closed 2022-04-20 by probonopd