probonopd/MiniDexed #121: Support pan, cutoff frequency/resonance and reverb level changes using standard Midi CCs

ChKK1963, 2022-04-20:
Thanks for already implementing volume, modulation and pitch bend! Next obvious candidates would be pan, cutoff frequency/resonance and reverb level as there are standard Midi CCs defined for them:

- 10 = Pan
- 71 = Resonance
- 74 = Frequency Cutoff
- 91 = Reverb Level

This is independent of the future user-assignable / learnable CCs.

probonopd, 2022-04-20:
Let's also add * 94 = Detune Level

probonopd, 2022-04-20:
@ChKK1963 can you please test the version from https://github.com/probonopd/MiniDexed/suites/6193386464/artifacts/218198239 and let me know whether it works as intended, thank you.

ChKK1963, 2022-04-20:
Hi, just tested your release:

CC10 (Pan), 71 (Resonance) and 74 (Cutoff Frequency) are working fine.

For CC 94 (Detune) I am not sure if this works as intended - this only seems to tune things up. If I have one track running without using the 94 CC and another one with the CC 94 sent, they are only in tune for a CC value 0. Everything other value tunes the track up.

For the 0 value this seems to be in track with the Midi specification which says for CC 94: "0 to 127, with 0 being no celeste effect applied at all.

But isn't detune supposed to be up or down? If you detune two tracks, you will often pitch one down a couple of cents and the other one up accordingly.

Not sure what is a good way to handle this ... I could imagine either using 0 for no detune and 1 - 62 for tuning down 63 in the middle for no detune and 64 to 127 for tuning up. This would allow for "sweeping" from low to high detuning (could be nice with Midi-based LFOs from the sequencer).

Or to let CC 94 as is and use one of the "unused" CCs for tuning things down.

Searched a bit in the Midi specification and the CC charts of a couple of manufacturers but found no "standard" way how this is assumed to behave (except for 0 being "off") ...

Thanks for your work!

From: probonopd
Sent: Wednesday, 20 April 2022 18:57
To: probonopd/MiniDexed
Cc: ChKK1963; Mention
Subject: Re: [probonopd/MiniDexed] Support pan, cutoff frequency/resonance andreverb level changes using standard Midi CCs (Issue #121)

@ChKK1963 can you please test the version from https://github.com/probonopd/MiniDexed/suites/6193386464/artifacts/218198239 and let me know whether it works as intended, thank you.

probonopd, 2022-04-20:
Thanks for testing @ChKK1963.

> But isn't detune supposed to be up or down? 

Most definitely! We need to fix this.

> using 0 for no detune and 1 - 62 for tuning down 63 in the middle for no detune and 64 to 127 for tuning up

I was blindly assuming that this was how it was intended, but thinking about it, the DX7 was made at a time when Yamaha (mis)read the MIDI spec for the first time and thought values should go from 0 to 99.

Note to self, TODO: Hook up Dexed on the PC and have it output to MiniDexed, hooked up to HDMI with MIDI logging enabled. Then play with the global Tune knob and see which values get sent.

probonopd, 2022-04-20:
Fun fact:

> There is no reference tone easily accessible on the DX7to tell us whether or not the machine is in standard concert tuning, that is Middle A = M40, Yamaha tells us that the DX7 will be at this concert tuning if the data entry slider is moved to the halfway point. That's all well and good except that as you've probably noticed there's no indent of any kind to tell us when we are at the halfway point. All we can do is approximate, by eye.

MASSEY, H. (1986). The complete DX7. New York, Amsco Publications. [ISBN 0825610710](https://www.worldcat.org/title/complete-dx7/oclc/869043541)

probonopd, 2022-04-20:
According to the DX7 user manual, page 6:
> MASTER TUNE adjusts the overall tuning of the DY7 to match its pitch with outher instruments. Pitch is variable over a 150 cent range.

According to the DX7 user manual, page 32, Master Tune is 0-127.

ChKK1963, 2022-04-20:
OK, this sounds for me like using “1” for 75 cents down, “63” in the middle as “in-tune”,  and “127” for 75 cents up would make sense. “0” must be “in-tune” according to the Midi spec.

Also just noticed that if you detune a Midi channel and then send a bank/program change to switch to another sound, it stays detuned. Again, something not specified anywhere if a program change should reset this ... but if detune simulates the DX7s master tuning fader, not touching this on a program change is probaly the right thing. You just need to be aware of this and send any CCs you need to reset together with the program change.

probonopd, 2022-04-20:
Sooner or later we are going to need a DX7 or DX7s owner to do some comparisons...

ChKK1963, 2022-04-20:
Sold mine about 20 years ago ...

probonopd, 2022-04-21:
@ChKK1963 tuning works as intended now. I had not scaled some of the CCs properly.

commit fe39d8dc (2022-04-20): Support more MIDI CCs (#122)

Add support for MIDI CCs

10 = Pan
71 = Resonance
74 = Frequency Cutoff
91 = Reverb Send
94 = Detune Level

Closes #121
closed 2022-04-21 by probonopd