probonopd/MiniDexed #125: Feature request - switchable performance.ini files

ebrown3k, 2022-04-20:
I think adding a way to have multiple performance.ini files and swap them through the UI/menu would add a ton of usefulness to this in a live performance setting.

probonopd, 2022-04-20:
Maybe a structure similar to sysex should be used, along the lines of `performances/001_PerformanceName.ini`, `performances/002_PerformanceName2.ini`

craigyjp, 2022-04-23:
The original DX7 only had one performance mode unfortunately, but the MK2 and S models had 32 or 64 performance modes. So it would be nice to have the extra performance patches.

arsamus, 2022-05-04:
> Maybe a structure similar to sysex should be used, along the lines of `performances/001_PerformanceName.ini`, `performances/002_PerformanceName2.ini`

This will be a good enhancement it is necessary for those that use minidexed as a multitumbral synth. Selection of saved performances is the way to change this multitumbral paches, otherwise you have to navigate on the menus and ajust the parameters to set the decired patch. 

Other important enhancement could be load and saving performances and sysex banks on an USB pendrive. this avoid to access to the SD to add new roms or to modify performance files.

probonopd, 2022-05-15:
Do you think you could do it @arsamus?

arsamus, 2022-05-15:
> Do you think you could do it @arsamus?

Let me see if I can do something.

closed 2022-07-20 by probonopd