probonopd/MiniDexed #134: Disable Restart when using the menu Shortcut

miotislucifugis, 2022-04-21:
First, great work everyone!  Im eagerly following the progress!

one suggestion:   A few time i have accidentally triggered a reset while i was using the menu shortcut (pressing the encoder button  and scroll)  to compare settings in the TGs.    A more preferable behavior would be to disable the reset timer when the encoder is being turned, to avoid any "surprises"

probonopd, 2022-04-21:
Happened to me once, too. If I understand @rsta2 correctly, then the functionality you describe is not that easy to implement. Or do you think you could implement it?

miotislucifugis, 2022-04-21:
Sorry, I havent even looked at the code, im sure it is over my head anyway.  If it is difficult to change, no worries.  It is definitely not a make or break issue
best

rsta2, 2022-04-21:
Perhaps this [reboot code](https://github.com/probonopd/MiniDexed/blob/995b25c71a4ef2a5fb8c98e77f00d2088dea36f7/src/userinterface.cpp#L212) should be totally removed? It was intended for development purposes, but 10 seconds is already very long, so switching it off and on again seems to be already quicker now.

probonopd, 2022-04-21:
Well, maybe it's useful as a last resort in some strange circumstances. Setting the timeout to 120 seconds for now.

closed 2022-04-21 by probonopd