probonopd/MiniDexed #136: Sustain CC (64) not handled correctly

ChKK1963, 2022-04-21:
After attaching my Raspi to my master keyboard with a sustain pedal, I observed an issue with sustain pedal handling. Midi sustain is sent as CC 64 - a value of 127 is sent when the pedal is held down, a value of 0 is sent when it comes up again.

While the pedal is pressed all notes should sustain, when the pedal comes up, all sustaining notes must be stopped (well technically not hard stopped but the "release" phase of the envelope starts).

MiniDexed behaves incorrectly:

- When I do not press the pedal all is fine - no sustain on the notes
- When I press and hold the pedal, the notes now played sustain when I release the keys - as expected 
- But when I release the pedal, the sustaining notes are not stopped - they continue playing as if the pedal is still pressed. Only new notes hit after releasing the pedal are played without sustain

This makes "normal" piano playing with the pedal impossible.

probonopd, 2022-04-21:
Can you please test with [Dexed](https://github.com/asb2m10/dexed) on the PC?

ChKK1963, 2022-04-21:
Tested on PC, behaves as expected

probonopd, 2022-04-21:
I guess this would need to be fixed in [Synth_Dexed](https://codeberg.org/dcoredump/Synth_Dexed) @dcoredump?

ChKK1963, 2022-04-21:
One additional note: what is done correctly by MiniDexed is the handling of notes already played where the keys are still held down when the sustain pedal is pressed. These notes started before the sustain pedal will sustain when the pedal is pressed when the keys are released as expected. But their sustain phase is also not ended when the pedal comes up again.

dcoredump, 2022-04-22:
> I guess this would need to be fixed in [Synth_Dexed](https://codeberg.org/dcoredump/Synth_Dexed) @dcoredump?

Thanks for reporting @ChKK1963 !

I will take a look at this asap.

dcoredump, 2022-04-23:
Hi @ChKK1963 @probonopd 

I hope I fixed this problem. Currently I don't have hardware for testing, but I am nearly 100% sure that the problem should be fixed. The code for the fix is taken from MicroDexed where sustain handling was working,

Please fetch the latest master for Synth_Dexed.

Reagards, Holger

probonopd, 2022-04-23:
Thanks @dcoredump. A build is available for testing but I need to somehow trick my MIDI controller into behaving like a pedal to test it. :)

closed 2022-04-23 by probonopd

ChKK1963, 2022-04-23:
Thanks a lot for (likely) fixing this. Unfortunately I can likely not test this before Monday, travelling …

timknab, 2022-04-23:
Thanks for the fix! I can confirm that my sustain pedal is now working correctly.

probonopd, 2022-04-23:
Thanks for letting us know @timknab!

PR #150 (2022-04-23): Update Synth_Dexed to fix Sustain behavior
Closes #136
Thanks @dcoredump and @ChKK1963