probonopd/MiniDexed #139: 3D printable case

paulsijben, 2022-04-22:
Has anyone come up with a 3D printable case that mimics the styling of the original synths?

craigyjp, 2022-04-22:
I built it into this one to make it more professional....

ebrown3k, 2022-04-22:
I am working on a 3D printed case in the form factor of a Korg Volca. The issue is it will take some special parts to get it to fit in the limited height complicating the build. I will also be custom designing it for the specific Pi version and DAC I ordered so I don't know how useful it would really be for a wide range of users.

dcoredump, 2022-04-22:
My case is not ready yet, but here are some pictures.

[image]

[image]

probonopd, 2022-04-22:
For extra authenticity, I recommend a 3D printed knob in this color:

[image]

probonopd, 2022-04-24:
Working on a 3d printable case (here: for the Raspberry Pi Zero) that can be assembled without any tools.

There is something about snap-fit 3d printed cases that I really like. Maybe it's their hackability which makes everything accessible at any time without any tools.

Turn on the sound to hear those satisfying clicks :100: 

https://user-images.githubusercontent.com/2480569/164972750-211dfeb2-eb15-4ba7-a175-af313d03468a.mp4

paulsijben, 2022-04-24:
> Working on a 3d printable case (here: for the Raspberry Pi Zero) that can be assembled without any tools.
> 
> We'd need a whole range of them to accomodoate for the various RPi and DAC models.
> 
> There is something about snap-fit 3d printed cases that I really like. Maybe it's their hackability which makes everything accessible at any time without any tools.
> 
> Turn on the sound to hear those satisfying clicks 💯
> 
>  3dprintedcase.mp4

nice! 
I was still in doubt whether to use the zero 2 or the pi 3 leveraging the latter's USB ports

probonopd, 2022-04-24:
The connectors on the Zero suck. A Pi 3 Model A might be the sweet spot, unless we start using Networking one day.

paulsijben, 2022-04-24:
> The connectors on the Zero suck. A Pi 3 Model A might be the sweet spot, unless we start using Networking one day.

true but I have never seen a 3A in the flesh so to speak. I do have a 3b laying around ;-)

probonopd, 2022-04-24:
It's alive!

[image]

craigyjp, 2022-04-24:
Nicely done