probonopd/MiniDexed #143: Midi led indicator

craigyjp, 2022-04-22:
Just wondering if it's possible to use a couple of gpio pins or maybe just one to indicate midi activity? I can easily add it on the UART input and output by buffering these lines to drive LEDs, but it would be nice if any USB or DIN midi was indicated.

probonopd, 2022-04-22:
I was actually wondering whether we should use the Raspberry Pi built-in LED to blink when there are MIDI messages. Would it interfere with sound performance @rsta2?

craigyjp, 2022-04-22:
I don't know how that would work on a pi400 as I don't have access to these LEDs to bring to a front panel.

probonopd, 2022-04-22:
Wait, you are "wasting" a percious 400 for MiniDexed? ;-)

craigyjp, 2022-04-22:
Like I said, it's currently all I can get hold of here in the Ukraine, there is a bad man called Putin making things hard for me, I will swap it out with a Pi 3 when I get the chance.

dcoredump, 2022-04-22:
> Like I said, it's currently all I can get hold of here in the Ukraine, there is a bad man called Putin making things hard for me, I will swap it out with a Pi 3 when I get the chance.

Please send me an email. I will try to send you a RPi3.

craigyjp, 2022-04-22:
I appreciate your offer, but please you don't have to. We are probably going to try to relocate to Bulgaria in the next few weeks depending on how it goes, currently we are safe in Dnipro and not in any danger (yet). The reason we have not left is we have an 18 year old son and we can't cross the border with him as he of fighting age, but he is slightly disabled so we are trying to get docs to allow him to cross the border. We should have these soon. Any postage might take 3-4 weeks to arrive and by that time we will be gone or dead, not sure which.

rsta2, 2022-04-23:
At the moment the Act LED is dedicated for the SD card activity. I think `CActLED::Blink()` cannot be used for that purpose, because it pauses the execution for the given blink time. Instead you have to use `CActLED::On()` and `CActLED::Off()` and use a kernel timer, which is started, when a MIDI event comes in, and canceled and restarted, when another event arrives. When the kernel timer expires after say 20ms, it switches the LED off.

probonopd, 2022-04-23:
Exceeds my C++ skills for the moment. If anyone wants to implement it, a PR would be welcome.

craigyjp, 2022-04-23:
Managed to get a Pi 4 today, mounted it in my case so I have access to the SD card for future releases.[image]

probonopd, 2022-04-23:
Not so sure I like what I am seeing with regards to mains power... it could be very dangerous. At the **very** least use some shrink tubing! But I am not an eletrician, so I strongly advise anyone to use an unmodified USB power supply.

craigyjp, 2022-04-23:
Yeah I understand, bit it's in a 3mm thick plastic box so well insulated. I have some shrink tubing I should stick over those terminals and the power switch, I was just being lazy as I used shrink on the led connections.

dcoredump, 2022-04-24:
> Yeah I understand, but it's in a 3mm thick plastic box so well insulated. I have some shrink tubing I should stick over those terminals and the power switch, I was just being lazy as I used shrink on the led connections.

I am an electrocian and I would highly recommend to insulate all mains voltage contacts somehow! At least with insulating tape but heat shrink tubing is usually better because no adhesive that gets old is involved.

craigyjp, 2022-04-24:
Next time I pop the lid off I will add some, I have been into electronics for 48 years now, pretty sure it's safe, no one but me will ever open it. Well maybe the Russians will if they loot it from my house with the rest of the Ukrainian peoples property.

dcoredump, 2022-04-24:
My old D110 (the successor of the MT-32) had a kind of level-meter on the display for every part.
I made the same for the two instances of Dexed for MicroDexed, so no LEDs and GPIOs are involved. And the Raspi has much more CPU power than a Teensy.
The update of the LCD does not need to be done at audio rate, it is totaly ok when updating at control rate (maybe 10 times a second).

craigyjp, 2022-04-24:
That's a nice idea, I have an SC55 which does the same thing, shows you
which TG is getting activity. Would be good to see based on velocity levels
0-127 I guess, or just blocks for each TG.

dcoredump, 2022-04-24:
> That's a nice idea, I have an SC55 which does the same thing, shows you
> which TG is getting activity. Would be good to see based on velocity levels
> 0-127 I guess, or just blocks for each TG.
> 
> 
> > My old D110 (the successor of the MT-32) had a kind of level-meter on the
> > display for every part.
> > I made the same for the two instances of Dexed for MicroDexed, so no LEDs
> > and GPIOs are involved. And the Raspi has much more CPU power than a Teensy.
> > The update of the LCD does not need to be done at audio rate, it is totaly
> > ok when updating at control rate (maybe 10 times a second).

I created a full block when a keydown MIDI event arrives on a channel. If a keyup occurs a "blank icon" is displayed. Between those events (inside control rate) I slowly count down and show icons with lower "black" parts until only one black line at the bottom is left (you only have 5*8 pixels, so 8-1 states are shown). This line is also shown for sustained notes and disapear after sustain is released.

craigyjp, 2022-04-24:
> #143 (comment)

I did something similar with a Roland A110 emulator that I built recently.