probonopd/MiniDexed #144: Midi thru switchable from menu

craigyjp, 2022-04-22:
I love the fact that midi thru is an option, but it would be nice if it could be configured globally, maybe a MIDI menu with the out/thru selection and the destination options.

probonopd, 2022-04-22:
What is your use case for it? I thought this is something one doesn't want to change very often, and hence thought an option in `minidexed.ini` would be sufficient.

craigyjp, 2022-04-22:
I thought it would be a useful feature if midi with an editor to have the option of thru or out. Also could be used as a midi/usb converter. Just seemed a more logical way of using it, most soft synths I have are switchable from the menu for thru or out. Even some 30 year old effects I have can be switched between thru and out from a menu.

I can add a thru easily enough, I built a MIDI interface to the minidexed with an 8 way buffer and have spare outputs to drive a thru.

probonopd, 2022-04-22:
How are your C++ skills? Do you think you could try it?

craigyjp, 2022-04-22:
Clumsy at best, you have seen my code. But I can take a look tomorrow. Too much beer tonight hence throwing out ideas

miotislucifugis, 2022-04-25:
After using the new midi thru feature for a bit,  I have agree that having a switchable midi thru option in the menu would be nice, especially since discovering that the minidexed can also be used as a stand alone usb host device.   very useful.  

I can imagine that one would use serial->serial  to connect to a sequencer  or something with "classic" DIN midi and THRU  being used to daisychain other gear....other times the mini dexed could be used as a utility box to connect usb devices to DIN Gear without needing a computer.    Not a priority but definitely a nice feature.

penfold42, 2025-11-25:
I have been looking at the Thru code as I'd like to have bi-directional forwarding.
In my case, I'm fitting a Pi inside an old Clavinova as an alternative (not replacement) for the synth in it and would like to throw midi data over the network to it _and_ catch midi data from the keyboard.
I've got this working as a simple duplication so I can configure:
```
MIDIThru=umidi1,udp
MIDIThru2=udp,umidi1
```
and it appears to work well.
If we add a menu, should we implement a midi router that can support forwarding from each input to any output ?
For default configuration, I'm tempted to honour the existing MIDIThru but also add
```
MIDIForward=umidid1:udp,udp:umidi1,another_intput:another_output
```

probonopd, 2025-12-04:
While I personally have no need for this, I am open to merging a PR if it doesn't break anything.