probonopd/MiniDexed #147: PiSound as DAC

lazzarello, 2022-04-23:
This project came up in [a topic on the PiSound forum](https://community.blokas.io/t/bare-metal-pisound/2966/7?u=lazzarello) regarding working with a DAC outside of a Linux kernel. I installed the latest SD card release but the DAC on the PiSound doesn't work by default. Could be interesting to add support for it's audio out + 5 pin DIN MIDI connectors.

probonopd, 2022-04-23:
Hello @lazzarello.

Currently PCM5102A and PCM5122 based DACs are supported, and WM8960 is in the works.

From looking at https://blokas.io/pisound/ I can't even figure out which chipset it uses. It is not even mentioned on https://blokas.io/pisound/docs/specs/. Do you know?

craigyjp, 2022-04-23:
I heard from a guy on Facebook that is building 8 of these onto Zero devices that the UDA1334 24/96 DAC works perfectly well.

probonopd, 2022-04-23:
Acording to https://github.com/dwhinham/mt32-pi/wiki/I²S-DACs the UDA1334 might be 1:1 compatible with the PCM5xxx and might hence work on MiniDexed as is.

According to the same source, the  Blokas Pisound uses a PCM5102A so it _should_ work, too. But I don't have one to test.