probonopd/MiniDexed #151: When saving the performance the sounds is garbled afterwards.

craigyjp, 2022-04-23:
I setup a performance, just midi channel changes and voice/pan/volume assignments and straight away the sound was garbled after double clicking the performance save.

Powering off and on fixed the issue, this is the second time it's happened to my build. First with a Pi 400 and now with a Pi 4 rev B.

miotislucifugis, 2022-04-23:
Ive also experienced this,  using a Pi 3.

rcarmo, 2022-05-15:
Reproduced this on a 3A+ as well. Happens pretty much every time I save.

probonopd, 2022-05-16:
I can confirm this on a Raspberry Pi 2.
Just saving without having changed something doesn't cause the bug. But changing, e.g., some voices, then saving does.

Rebooting makes the newly saved performance work correctly.

Do we have to load the performance file after saving it?

izash, 2022-05-16:
I can confirm this, almost every time I save a performance on my pi2. 
Rebooting clears the problem.

probonopd, 2022-05-17:
@rsta2 do you have any idea why this might be?

https://github.com/probonopd/MiniDexed/pull/72

Do you think one would need to call `m_PerformanceConfig.Load ()` at the end of `m_PerformanceConfig.Save ()` in `src/performanceconfig.cpp`? Or is the culprint something else entirely?

rsta2, 2022-05-17:
Unfortunately I did a mistake, when implementing the "save performance" function. The file functions are called directly from the interrupt handler of the rotary encoder at `IRQ_LEVEL`, which is not allowed. One has to set a flag only in the interrupt handler instead, and call `CMiniDexed::SavePerformance()` from `Process()` at `TASK_LEVEL`, when this flag has been set. Afterwards the flag has to be reset, of course.

probonopd, 2022-05-17:
Thanks for the hint @rsta2. Perhaps someone well-versed in C++ can pick it up from here.

rsta2, 2022-05-18:
Because it was my mistake, here is a fix for this issue:

[fix-save-performance.zip](https://github.com/probonopd/MiniDexed/files/8715726/fix-save-performance.zip)

PR #238 (2022-05-21): Fix Save Performance
Closes #151
Thanks @rsta2
commit 2f044b6f (2022-05-21): Fix Save Performance (#238)

Closes #151
Thanks @rsta2
closed 2022-05-21 by probonopd