probonopd/MiniDexed #156: Configurable to ignore MIDI program change messages

fp64lib, 2022-04-23:
I have a setup with several synths, all connected via MIDI and also to central MIDI master keyboard. Usually, I want most of the synths to keep their current voice and ignore any MIDI Program Change messages. Most synth have a parameter for that like "Enable RX CC change". It would be great to have something similar on the MiniDexed as a changeable global parameter.

I already implemented a quick solution with a static parameter in minidexed.ini
`MIDIRXProgramChange=1`

As there is currently another pull request opened by me, git would create a new pull request including both features. As I did not want to create another branch just for this quick hack, take a look at my current commit https://github.com/fp64lib/MiniDexed/commit/3af6cb533167927bfd98ff2d2f745254cf1ed4c6 . This works fine for the time being. An editable parameter would of course be better.

probonopd, 2022-04-24:
Thanks @fp64lib. Out of curiosity, why don't you run your different synths on different MIDI channels, so that the program changes would be received only by the synth you want to change the program on?

How do you switch between patches with MIDI Program Change messages disabled? Always using the menu?

probonopd, 2022-04-24:
Please test
https://github.com/probonopd/MiniDexed/suites/6235882911/artifacts/221385951

Thanks!

probonopd, 2022-04-24:
> As there is currently another pull request opened by me, git would create a new pull request including both features. 

Not if you do your new changes on a different branch, and then create a pull request from that new branch. Hint for the next time ;-)

fp64lib, 2022-04-24:
> Thanks @fp64lib. Out of curiosity, why don't you run your different synths on different MIDI channels, so that the program changes would be received only by the synth you want to change the program on?

All my synth run on different MIDI channels. Problem is that there are only 16 MIDI channels,  some devices support multiple voices in one track / song. So some devices have to share channels. So I usually use set most of the synths to ignore program change, and change the voice via menu.

For the MiniDexed, I am not yet decided on whether I like the setup of one MIDI channel per voice (which blocks 8 channels) or whether it would be better to have one channel for all voices. The first give a you the possibility to have a mini band, the second to use it as a dedicated multi-voice synth e.g. like my Yamaha Modx. Maybe another parameter... ;-)

luisfcorreia, 2022-04-24:
as I do for my other FM Synth, I've reserved a specific sub-midi adapter just for it

(adapters are cheap)

fp64lib, 2022-04-24:
> Please test https://github.com/probonopd/MiniDexed/suites/6235882911/artifacts/221385951
> 
> Thanks!

Works fine as expected, no problems. Thanks!

probonopd, 2022-04-24:
> Problem is that there are only 16 MIDI channels

Ah, now I understand.

closed 2022-04-24 by probonopd