probonopd/MiniDexed #160: Small SD cards do not work correctly

fp64lib, 2022-04-24:
To speed-up testing, I use several Micro SD-cards. One of them was a 128MB card - which is absolutely fine: even with 26 sound banks, the entire MiniDexed only takes up about 16MB. The small SD card is properly detected during boot, however neither the sound card settings nor the additional voices are read from the card. So the synth always uses PWM mode, does not load the LCD, does not use the rotary encoder and always falls back to the built-in default voices. So basically only what is available via MIDI can be controlled.

I don't know how many small Micro SDcards are there still in (re-)use for the MiniDexed. For me it would also work out, if is clearly mentioned in the installation procedure that the minimum size needed for an SD card is xx GB ;-)

Small (128MB) card boot-up log on HDMI screen:
```
logger: Circle on Raspberry Pi 2 Model B
00:00:00.66 timer: SpeedFactor is 1.51
00:00:01.18 emmc: Found a valid **version 1.0 or 1.01** SD card
00:00:02.05 usbdev0-1: Device ven424-9514, dev9-0-0 found
00:00:02.05 usbdev0-1: Interface int9-0-0 found
00:00:02.05 usbdevé-1: Using device/interface int9-0-0
00:00:02.81 usbdev0-1: Device ven424-ec00 found
00:00:02.81 usbdevl-1: Using device interface ven424-ec00
00:00:62.88 SmSc951x: MAC address is B8:27: EB:FB:76: DE
00:00:02.90 usbhub: Port 1: Device configured
00:00:02.92 duroot: Device conf igured
00:06:02.92 minidexed: Compile time: Apr 24 2022 06:47:46
00:00:02.93 minidexed: **PWM mode**
**00:00:02.93 syxf ile: Directory /sysex/voice not found**
00:00:02.94 minidexed : Serial MIDI interface enabled
00:00:02.94 mcore: CPU core 1 started
00:00:02.94 mcore: CPU core 2 started
00:00:02.94 mcore: CPU core 2 started
```

Normal (16GB) card boot-up log on HDMI screen (differences are **highlighted** with **):
```
logger: Circle 44.4.1 started on Raspberry Pi 2 Model B
00:00:00.66 timer: SpeedFactor is 1.51
OG:00:01.19 emmc: Found a valid **version 3.0x** SD card
00:00:02.06 usbdevo-1: Device ven424-9514, dev9-0-0 found
00:00:02.06 Lisbdev8-1: Interface int9-0-0 found
00:00:02.06 usbdev8-1: Using device/interface int9-6-0
00:00:02.82 usbdevo-1: Device ven424-ec00 found
00:00:02.82 usbdev8-1: Using device/interface ven424-ec00
00:00:02.89 SmSc951x: MAC address is B8:27:EB:FB:70: DE
OG:00:02.91 usbhub: Port 1: Device configured
00:00:02.93 dwroot: Device configured
00:00:02.93 minidexed: Compile time: Apr 24 2022 06:47:46
00:00:02.94 minidexed : **I2S mode**
**00:00:02.95 ui: LCD initialized**
**00:00:02.95 ui: Rotary encoder initialized**
**00:00:02.96 syxfile: Bank #1 successfully loaded**
**00:00:02.97 syxfile: Bank #2 successfully loaded**
00:00:02.97 minidexed : Serial MIDI interface enabled
00:00:02.98 mcore: CPU core 1 started
00:00:02.98 mcore: CPU core 2 started
00:00:02.98 mcore: CPU core 3 started
```

minidexed.ini on small card:
```#
# minidexed.ini
#

# Sound device
SoundDevice=i2s
#SoundDevice=pwm
#SoundDevice=hdmi
SampleRate=48000
#ChunkSize=256
DACI2CAddress=0
ChannelsSwapped=0

# MIDI
MIDIBaudRate=31250
#MIDIThru=umidi1,ttyS1
MIDIRXProgramChange=0

# HD44780 LCD
LCDEnabled=1
LCDPinEnable=17
LCDPinRegisterSelect=27
LCDPinReadWrite=16
LCDPinData4=22
LCDPinData5=23
LCDPinData6=24
LCDPinData7=25

# KY-040 Rotary Encoder
EncoderEnabled=1
EncoderPinClock=5
EncoderPinData=6
EncoderPinSwitch=26

# Debug
MIDIDumpEnabled=0
ProfileEnabled=0
```

Content of small SD-card (identical copy of bigger SD card):
```
sdcard-small:
total 16048
-rw-r--r-- 1 uwe uwe    1594 Apr 24 06:49  LICENCE.broadcom
drwxr-xr-x 1 uwe uwe    4096 Apr 24 11:35 'System Volume Information'
-rw-r--r-- 1 uwe uwe     336 Apr 24 06:49  armstub8-rpi4.bin
-rw-r--r-- 1 uwe uwe   51543 Apr 24 06:49  bcm2711-rpi-4-b.dtb
-rw-r--r-- 1 uwe uwe   51675 Apr 24 06:49  bcm2711-rpi-400.dtb
-rw-r--r-- 1 uwe uwe   52184 Apr 24 06:49  bcm2711-rpi-cm4.dtb
-rw-r--r-- 1 uwe uwe   52460 Apr 24 06:49  bootcode.bin
-rw-r--r-- 1 uwe uwe      14 Apr 24 06:49  cmdline.txt
-rw-r--r-- 1 uwe uwe     361 Apr 24 06:49  config.txt
-rw-r--r-- 1 uwe uwe    7223 Apr 24 06:49  fixup.dat
-rw-r--r-- 1 uwe uwe    5354 Apr 24 06:49  fixup4.dat
-rw-r--r-- 1 uwe uwe    3145 Apr 24 06:49  fixup4cd.dat
-rw-r--r-- 1 uwe uwe    3145 Apr 24 06:49  fixup_cd.dat
-rw-r--r-- 1 uwe uwe    2165 Apr 24 06:49  getsysex.sh
-rw-r--r-- 1 uwe uwe 2228940 Apr 24 06:49  kernel.img
-rw-r--r-- 1 uwe uwe 2074028 Apr 24 06:49  kernel7.img
-rw-r--r-- 1 uwe uwe 2531976 Apr 24 06:49  kernel8-rpi4.img
-rw-r--r-- 1 uwe uwe 2515120 Apr 24 06:49  kernel8.img
-rw-r--r-- 1 uwe uwe     513 Apr 24 11:36  minidexed.ini
-rw-r--r-- 1 uwe uwe    2129 Apr 24 06:49  performance.ini
-rw-r--r-- 1 uwe uwe 2964864 Apr 24 06:49  start.elf
-rw-r--r-- 1 uwe uwe 2240608 Apr 24 06:49  start4.elf
-rw-r--r-- 1 uwe uwe  800028 Apr 24 06:49  start4cd.elf
-rw-r--r-- 1 uwe uwe  800028 Apr 24 06:49  start_cd.elf
drwxr-xr-x 1 uwe uwe    4096 Apr 22 15:52  sysex

'sdcard-small/System Volume Information':
total 0
-rw-r--r-- 1 uwe uwe 76 Apr 24 11:35 IndexerVolumeGuid
-rw-r--r-- 1 uwe uwe 12 Apr 24 11:35 WPSettings.dat

sdcard-small/sysex:
total 0
drwxr-xr-x 1 uwe uwe 4096 Apr 22 15:52 voice

sdcard-small/sysex/voice:
total 16
-rw-r--r-- 1 uwe uwe 4104 Dec  4 22:18 000001_rom3a.syx
-rw-r--r-- 1 uwe uwe 4104 Dec  4 22:18 000002_rom3b.syx
```

probonopd, 2022-04-24:
Please reformat your small SD card with the FAT32 filesystem and try again.

fp64lib, 2022-04-24:
Sorry for not mentioning it: I did that already before raising the issue. But just re-checked it. No difference, it still does not work.

probonopd, 2022-04-24:
Can you reproduce the same issue with https://github.com/dwhinham/mt32-pi/?

fp64lib, 2022-04-24:
Yes, there the situation is even worse: An identical image does not work at all on the small Micro SD card, you can clearly see the harddrive LED that the roms are not loaded. Seems to be a problem of the low-level FAT or directory routines.

probonopd, 2022-04-24:
In this case you might want to report it over at https://github.com/rsta2/circle/issues/. Once it is fixed there, it will be fixed in MiniDexed.

fp64lib, 2022-04-24:
Card has definitely FAT32 file system on it. I raised issue #288 to https://github.com/rsta2/circle/issues/, see link above. Hope they find something.

pegli, 2022-12-28:
I also ran into this issue.  I'm on a Mac so I naturally used Disk Utility to format the card.  Disk Utility will happily put a FAT32 filesystem onto an SD card, but it will only partition it for 2 Gb.  The workaround is to use the command line to format.  Use `diskutil list` to find the device file associated with the SD card, then run:

```
% diskutil eraseDisk "FAT32" MINIDEXED MBRFormat /dev/disk3
```

Where "MINIDEXED" is the name of the new volume and "/dev/disk3" is the device file for the SD card.  Don't mess that second one up... :)  This gave me a card with a FAT32 filesystem with a 16 Gb partition on a 16 gig card.

Gustavomurta, 2023-07-22:
Following Raspberry Pi recommendations, slow SD cards are not recommended!
The vast majority of old cards (2,4,8,16GB) have class 4, which is very slow.

**Reccomendations for SD Cards**
https://forums.raspberrypi.com/viewtopic.php?t=350898

For better performance with miniDexed, I recommend SD card with class 10 or higher.