probonopd/MiniDexed #163: Change default pinout

probonopd, 2022-04-24:
For maximum device compatibliity (including Raspberry Pi 1), we want to have a pinout that needs none of the extra pins introduced with the Raspberry Pi 2.

Since more and more people are building their hardware now, I'd like to propose the new default pinout rather sooner than later.

[image]

Should we ever use i2c for the display (easier cabling with fewer wires, potentiometer onboard but ~3 EUR more expensive), then we could even have room for SPI:

[image]

What do you think? Feedback appreciated.

craigyjp, 2022-04-24:
You still have 4 pins assigned to the LCD on your second diagram, surely those will not be needed if your are using I2c?

probonopd, 2022-04-24:
Thanks @craigyjp, updated.

dcoredump, 2022-04-24:
I am using I2C displays for my [MicroDexed](https://codeberg.org/dcoredump/MicroDexed) project. In fact, cabling is easier and the I2C versions have a small pot for setting the contrast on the back side.

On the other side you only can define up to 8 or 10 self defined icons. I don't know if this is a library or  a hardware limitation, but sometimes more than 8/10 self defined icons are necessary (e.g. for displaying VU-Meter like bars for each TG).

Also you must have in mind that the I2C displays are slower than direct connected ones. This was a problem for me when changing sounds by fast turning an encoder. The display was not fast enough for displaying the sound names. I "fixed" it by using an encoder library which handles fast turns by sending bigger values instead every single step.

Just my two cents...

probonopd, 2022-04-24:
Successfully built a unit with the first pinout above (no i2c).

```
#
# minidexed.ini
#

# Sound device
SoundDevice=i2s
SampleRate=48000
#ChunkSize=256
DACI2CAddress=0
ChannelsSwapped=0

# MIDI
MIDIBaudRate=31250
#MIDIThru=umidi1,ttyS1

# HD44780 LCD
LCDEnabled=1
LCDPinEnable=17
LCDPinRegisterSelect=27
LCDPinReadWrite=20
# LCDPinReadWrite should be NONE. How to specify that?
LCDPinData4=22
LCDPinData5=23
LCDPinData6=24
LCDPinData7=25

# KY-040 Rotary Encoder
EncoderEnabled=1
EncoderPinClock=9
EncoderPinData=10
EncoderPinSwitch=11

# Debug
MIDIDumpEnabled=0
ProfileEnabled=0
```

If there are no objections, this will become the default soon-ish.

probonopd, 2022-04-24:
@rsta2 what should one write in `LCDPinReadWrite` if LCD pin 5 (RW) is attached to a GND pin directly?

rsta2, 2022-04-25:
Set `LCDPinReadWrite=0`, if RW is directly connected to GND.

luisfcorreia, 2022-04-26:
I'm trying this new pinout config, to be able to test it with both RPI1 and RPI2

LCD works well in RPI1, does not with RPI2

(same minidexed.ini configuration on both cards)

dcoredump, 2022-04-28:
[...]
> If there are no objections, this will become the default soon-ish.

If the pinout is fixed, perhaps some can create a nice Fritzing oucture with the wiring?

luisfcorreia, 2022-04-28:
Can you please generate a release for testing?

probonopd, 2022-04-28:
According to https://raspberry-projects.com/pi/pi-hardware/raspberry-pi-pcb-versions, I seem to have a Raspberry Pi Model B Revision 2.0 (512MB) because it says `Raspberry Pi (c)2011.12`. ON this one I tested the new pinout successfully, including display and rotary encoder. Did not tests DAC there.

luisfcorreia, 2022-04-28:
> According to https://raspberry-projects.com/pi/pi-hardware/raspberry-pi-pcb-versions, I seem to have a Raspberry Pi Model B Revision 2.0 (512MB) because it says `Raspberry Pi (c)2011.12`. ON this [...]

I have the same board here, will connect additional pins for DAC testing "soon"(TM)

probonopd, 2022-04-28:
> Can you please generate a release for testing?

https://github.com/probonopd/MiniDexed/suites/6304896074/artifacts/226117903

diyelectromusic, 2022-04-28:
I believe that last release has the encoder CLK/DAT pins swapped compared to your original diagram and description - is that correct?

probonopd, 2022-04-28:
Don't think so, these are the changes:
https://github.com/probonopd/MiniDexed/pull/189/files

Anyway, will be moving LCDPinRegisterSelect from GPIO 27 to GPIO 4 as per your advice now. Then there will be a new round of testing.

diyelectromusic, 2022-04-28:
I might be missing something but doesn't that say Clock = 10; Data = 9 compared to the original diagram above where it says Clock = 9; Data = 10?  To be honest, C=10;D=9 means the pins are in the same order as my encoder's pins, so that is what I had used on my V1 Pi anyway, but I was just checking that it was correct - that you'd changed them with respect to your original diagram...?

BTW - Would you expect this to work on an I2S DAC connected to the Pi V1 Model B Rev 2's P5 header?  That should be supported in the Circle environment, but I don't know if that means it should also work here...?

luisfcorreia, 2022-04-28:
these encoders are IMHO wrongly labelled

there is no Clock and Data, just A and B

diyelectromusic, 2022-04-28:
> BTW - Would you expect this to work on an I2S DAC connected to the Pi V1 Model B Rev 2's P5 header? That should be supported in the Circle environment, but I don't know if that means it should also work here...?

Ignore that - I can tell you it does :)  I'm using a V1 Model B Revision 2.0 with a PCM5102 DAC off the P5 header and all works fine with USB MIDI.

probonopd, 2022-04-28:
If the encoder pins are reversed, the encoder goes in the wrong direction. Did I mix it up?

probonopd, 2022-04-28:
Gota LOVE snap-fit project boxes. I'd be screwing around with screws otherwise all the time ;-)

probonopd, 2022-04-28:
```
#
# minidexed.ini
#

# Sound device
SoundDevice=i2s
SampleRate=48000
#ChunkSize=256
DACI2CAddress=0
ChannelsSwapped=0

# MIDI
MIDIBaudRate=31250
#MIDIThru=umidi1,ttyS1

# HD44780 LCD
LCDEnabled=1
LCDPinEnable=17
LCDPinRegisterSelect=4
LCDPinReadWrite=0
LCDPinData4=22
LCDPinData5=23
LCDPinData6=24
LCDPinData7=25

# KY-040 Rotary Encoder
EncoderEnabled=1
EncoderPinClock=10
EncoderPinData=9
EncoderPinSwitch=11

# Debug
MIDIDumpEnabled=0
ProfileEnabled=0

```

probonopd, 2022-04-28:
Please test:
https://github.com/probonopd/MiniDexed/suites/6306050512/artifacts/226182252

probonopd, 2022-04-28:
> these encoders are IMHO wrongly labelled
> there is no Clock and Data, just A and B

Yes, but we use what is printed on the PCB so that people know where to hook it up:

[image]

diyelectromusic, 2022-04-29:
> Please test: https://github.com/probonopd/MiniDexed/suites/6306050512/artifacts/226182252

All works fine on my V1 Model B Rev 2 with USB MIDI, LCD, encoder (both directions/select) with internal PWM and external (via P5) PCM5102.

Kevin.

diyelectromusic, 2022-04-29:
> Please test: https://github.com/probonopd/MiniDexed/suites/6306050512/artifacts/226182252

Also fine on my RPi V4 with external PCM5102 and serial MIDI (both on my Clumsy MIDI interface that has no I2C display).  LCD, encoder, audio seem fine.

diyelectromusic, 2022-04-29:
> Hi @diyelectromusic. Thanks for testing and for your article. Would you mind making a Fritzing diagram of [this pinout](https://github.com/probonopd/MiniDexed/issues/163#issuecomment-1112696014) without a cobbler, just with the bare Raspberry Pi 1, for use in our Wiki?

Yes I can do that.  With or without solderless breadboard?  i.e. do you just want to show the connections or do you want a practical thing someone could construct (which really needs things pinned into something rather than just "hanging on" by wires)?  So it can be more of a "getting started" diagram than an "informational" diagram?

luisfcorreia, 2022-04-29:
> Yes I can do that. With or without solderless breadboard? i.e. do you just want to show the connections or do you want a practical thing someone could construct (which really needs things pinned [...]

I would say, direct connections without any breadboard in between elements

probonopd, 2022-04-29:
Yes.

diyelectromusic, 2022-04-29:
Ok will do.  The only issue that there are three 5V connections required if you include something to control the brightness of the LCD, so some soldering of wires would be required.  Adding a solderless breadboard as a breakout means the only soldering is headers, but I'll have a go and see what it looks like!

probonopd, 2022-04-29:
Who needs brightness control ;-)

diyelectromusic, 2022-04-29:
OK, well there is something like this...?
[image]

Braincell1973, 2022-04-29:
Just my 2p worth but I built a mt32pi with serial based midi ports and a dac and encoder along with an i2c display........ Wouldn't the best idea to work towards would be an identical pinout so a change of sdcard can swap between mt32pi and minidexed ??

hizzlekizzle, 2022-04-30:
I just messed around with this new pinout and found it much, much easier to work with, so cheers for that.

probonopd, 2022-04-30:
Thanks @diyelectromusic.

Are you sure the LCD backlight is driven correctly (with a resistor) when attached this way? 

Also, the additional resistor or potentiometer for the contrast should be attached to the LCD between V0 and GND.

probonopd, 2022-04-30:
@Braincell1973 mt32pi is using i2c for the display and uses a pinout that doesn't work with the original Raspberry Pi ("1").
If someone would like to use mt32pi hardware with MiniDexed, some manual editing of the pin configuration will be needed as we want to support all Raspberry Pi models by default.

diyelectromusic, 2022-04-30:
> Are you sure the LCD backlight is driven correctly (with a resistor) when attached this way?

No, so if someone wants to say what they want to use, happy to do that - I had a pot wired in in mine on the breadboard, but took you are your word on "who needs brightness control" and assumed you literally just wanted the connections showing rather than an actual circuit.

One thing though, if lots of components are added it stops being a "wiring diagram" pretty quickly and then in my view ought to go one of two ways: either a) show a physical circuit on a breadboard that someone can actually make (e.g. as a "beginner" getting up and running quickly) or b) show a schematic rather than a Fritzing diagram and let people make their own (as a more experienced person being given guidelines).

I'm not sure "floating components" on a Fritzing diagram really helps people very much as it implies something that can be built like that, but in reality can't...?  And the more components are added the more it gets away from a "beginner having a go" - lots of circuitry around an encoder, whilst better in the long run, doesn't get someone up and running quickly...

Happy to do whatever people want, but I think we need to decide who is the audience for the diagram and tailor it accordingly?

probonopd, 2022-04-30:
Like this:

[image]

luisfcorreia, 2022-04-30:
to make things _clear_:

Contrast is LCD's pin 3 (should be adjustable)
Backlight on pins 15 and 16 are Brightness

probonopd, 2022-04-30:
> Are you sure the LCD backlight is driven correctly (with a resistor) when attached this way?

This is how I was driving my backlight using the resistor that is already on the PCB. Pins 15 and 16 stay unpopulated:

[image]

Turns out I was doing it overly complicated: Pin 15 seems to be connected to the anode through the 100 Ohm resistor, too. So **your picture is OK**, I was just saving 2 cables to the raspberry Pi ;-)

As for contrast, I find that this is needed:

[image]

> I'm not sure "floating components" on a Fritzing diagram really helps people very much as it implies something that can be built like that, but in reality can't...?

In fact, soldering to the modules is exactly what I am doing. The only exception being the Raspberry Pi for which I am soldering to such a header so that I don't have to solder to the Raspberry Pi directly:

[image]

diyelectromusic, 2022-04-30:
> In fact, soldering to the modules is exactly what I am doing.

I know - that's what I mean about who is the audience for these - it isn't really someone like you! :)

Ideally I'd add the PCM link and MIDI too, but then that is getting quite a long way away from who I'd imagine the audience would be - someone just wanted to get started and "have a go" with MiniDexed.  With a LCD with pre-soldered headers and a KY-040 encoder module and a wiring diagram, almost anyone could try it without any soldering which would be a nice thing to be able to offer.

Anyone capable of soldering extra components probably won't need a Fritzing diagram - they mostly just need the GPIO table, but a recommended schematic would be nice.

> If you share your Fritzing files, we can make both schematics and "flying component" wiring diagrams.

I'll see what I can do :)

diyelectromusic, 2022-04-30:
Does this one seem any better?  I've not added the capacitors, I just can't get it looking clear enough with them in, so they might need to be added as text.  It would probably make more sense to solder them directly onto the encoder pcb anyway?

I dropped back to just a single 5V and GND to the LCD (and changed the GND pin used on the PI in the process). It just seems neater that way?  What do we think about connections crossing the LCD?  I try not to do that, but there are far less crossings doing it this way than not doing it.  Alternatively I could move the LCD to the right and drop it down so the connections are all above it, but that is starting to stretch the diagram out a bit and some of the links will be getting quite long...

The main issue is that the wiring makes a lot of sense when wired across the RPi (as if the LCD and encoder are sitting on top like a Hat/Shield) but that isn't so clear in a diagram as everything has to cross the RPi...

Thoughts?

[image]

probonopd, 2022-04-30:
Yes, I think this is great. Except for the blue connection point on top of the yellow line - this could be misintepreted. I'd do the yellow line more like the orange one, have it to to the left, then to the top, then to the display. Btw, the SCK pin on my i2s module can be left ungrounded and it still works.

diyelectromusic, 2022-04-30:
> Would you be willing to share the Fritzing file?

Oh yes, sure :)  Where do you want it!?

[image]

probonopd, 2022-04-30:
Cool! If you zip the Fritzing file, then you can attach it to your post above this one.

diyelectromusic, 2022-04-30:
> Cool! If you zip the Fritzing file, then you can attach it to your post above this one. Thank you very much.

Try that

diyelectromusic, 2022-05-05:
Here is an updated Fritzing file, this time with a possibly workable schematic too (seeing as someone was asking elsewhere). There was also two buggy connections in the first one that are now hopefully fixed!

Kevin

[image]
[image]
[RPi-miniDexed-DefaultPins-Wires.zip](https://github.com/probonopd/MiniDexed/files/8629202/RPi-miniDexed-DefaultPins-Wires.zip)

closed 2022-05-05 by probonopd

probonopd, 2022-05-07:
Using the Fritzing file, it should be possible to create Gerber files for a service like JLCPCB...
https://support.jlcpcb.com/article/171-how-to-generate-gerber-and-drill-files-in-fritzing

BobanSpasic, 2025-01-07:
Hi guys,

I can't find this pinout from here https://github.com/probonopd/MiniDexed/issues/163#issuecomment-1112696014 in the Wiki anymore. Is it deleted?
Please, include it again. It was such a useful reference for me so many times.

probonopd, 2025-01-07:
It's right on the homepage https://github.com/probonopd/MiniDexed:

https://github.com/probonopd/MiniDexed?tab=readme-ov-file#pinout

BobanSpasic, 2025-01-07:
<---- needs glasses, immediately

Thanks, I was looking for it in Wiki and totally forgot about the main page.

PR #189 (2022-05-05): Change default pinout
Closes #163